indigo-astronomy/indigo #391: INDIGO driver for RaspberryPi HQ camera

astrorafael, 2020-12-17:
It is now really feasible to perform low cost astrophotograpy using the [new Raspberry Pi HQ camera](https://www.raspberrypi.org/products/raspberry-pi-high-quality-camera/)- You can see [astonishing examples](https://terramex.neocities.org/astro) and [various setups](https://www.hackster.io/news/hubble-pi-is-a-raspberry-pi-based-astrophotography-camera-63f7c3b03a17).

For scientific purposes, the Bayer RAW mode should be supported. Other modes such as YUV unfortunately perform gamma scaling which spoils linearity (done in the RPi VCore firmware). Raspistill supports this raw bayer mode writting an EXIF header like an ordinary JPEG image but substituting the JPEG encoded pixel data with the Bayer pixel data. Beyond astrophotograpy, the possibilities of such low cost setup are wide:remote  sky pollution monitoring, fireball monitoring, transits & occultation monitoring, etc.

However, this CMOS camera + raspistill cannot be integrated and managed properly as an astronomical device, just like other devices in the generic INDI model. It would be very nice if this new CMOS camera could have its own INDIGO driver, probably based on the [raspistill C source code](https://github.com/raspberrypi/userland/blob/master/host_applications/linux/apps/raspicam/RaspiStill.c), but simplified and suited to astrophotography.

An interesting tutorial on how these CMOS images can be found in the [picamera Python library](https://picamera.readthedocs.io/en/release-1.13/fov.html)

polakovic, 2020-12-17:
I'll check raspistill to learn how to use the camera. I always wanted to make a driver for it, this is what I needed :)

Thank you! Peter

astrorafael, 2020-12-18:
Excelent news!
For scientific purposes, it is convenient to disable automatic gains and automatic white balance modes, as discussed in [this note](https://picamera.readthedocs.io/en/release-1.13/recipes1.html#capturing-consistent-images).

polakovic, 2020-12-18:
Definitely, we always capture raw image without any "automatic" correction whenever possible.

johnwebbcole, 2021-06-20:
Any updates on the RPi HQ driver?

polakovic, 2021-06-20:
Not yet, but it is a good time to look at it..

callahan69, 2022-02-06:
Since debian bullseye, the camera subsystem is now libcamera. I tried to understand, how libcamera works and tried to build the classes of libcamera-apps on indigosky. Unforunally it cannot compile an easy indogo_driver based upon ccd_simulator, the c++includes on indigosky are not found, and redefining the include pathes opens other problems, 
It seems that using the class structure by libcamera-apps does not match the architecture of indigo, is this correct?

rumengb, 2022-02-06:
indigosky is not meant to be a build system. It is supposed to be a
lightweight system, optimized to run INDIGO. So you need to install the
whole build system on your own. First try to build indigo without any
modifications according to the README and then install the additional
dependencies needed by libcamera. however if you install libcamera
development package it should install its dependencies too. And only then
you should try to build your driver.

rumengb, 2022-02-07:
Sorry I do not understand. Is it OK now?

callahan69, 2022-02-07:
In principle yes, but there are some questions about the architecture, and the setup of driver development. Is there a documentation about the buildprocess? If I need to link a library (libcamera), how is this to be added to the makefile.inc? Or is linking of external libraries not allowed?

rumengb, 2022-02-08:
First you should add your driver to linux_drivers libcmera is Linux specific driver. You should name it indigo_ccd_libcamera or something like that indigo_ccd_ is mandatory prefix. Since libcamera is standard library you can simply use it you can look at
https://github.com/indigo-astronomy/indigo/blob/master/indigo_linux_drivers/ccd_gphoto2/Makefile.inc
It shows how to install dependency that you provide (in this case libraw) and system library libgphoto2.
Example how to make a driver can be found everywhere bit the simulator is not the best example. it overly complex with many  devices being exported.  For that matter I would suggest  you to look at:
https://github.com/indigo-astronomy/indigo/blob/master/indigo_docs/DRIVER_DEVELOPMENT_BASICS.md
it is also important to be familiar with the driver concepts:
https://github.com/indigo-astronomy/indigo/blob/master/indigo_docs/INDIGO_SERVER_AND_DRIVERS_GUIDE.md
and some simpler camera driver like:
indigo_ccd_mi or indigo_ccd_dsi (do not pay attention to the firmware loader)

sajmons, 2023-05-07:
Is there any progress on this? Being able to use HQ Camera from ASCOM would be really great.

polakovic, 2023-05-08:
Sorry, nothing new yet.

rohan5sep, 2025-04-05:
Hi, I would like to use Raspberry Pi camera as well. Is there any plan to add this driver? If not, I have used libcamera and think I can fill in the gap add add the driver. Let me know if I can submit the code here.

polakovic, 2026-10-02:
A libcamera-based driver is available as indigo_optional_drivers/ccd_rpi. It is not part of the default build because of the size of its dependencies (see [indigo-astronomy/indigo#732](https://github.com/indigo-astronomy/indigo/issues/732)) and has to be built explicitly.

closed 2026-10-02 by polakovic