indigo-astronomy/indigo #392: Add support for celestron focuser connected via AUX port

ckuethe, 2020-12-20:
Short version: Indigo doesn't detect a Celestron focuser connected to the AUX port on a NexStar mount. Focuser driver expects serial interface.

More details: Using IndigoSky (Indigo 2.0-136) connected to a Celestron CPC1100 with Celestron focuser attached via the AUX connection. Topology is roughly like this:

[image]

The hand controller can align and drive the mount, can move the focuser, and picks up time/location from the internal GPS. `indigo_control_panel` and the indigosky web server both report connection to via `indigo_mount_nexstar`, detecting a Celestron CPC with firmware version 5.31, with internal GPS, and my current location.

Alas, I don't see the the focuser anywhere on that bus. I've enabled the `indigo_focuser_efa` driver but it wants a serial port - my focuser is attached to the mount via one of the AUX ports.

Just for fun, I plugged in my wifi accessory and to check for firmware updates - everything appears to be updated already.
[image]

polakovic, 2020-12-20:
Yes, that's right. It is not implemented yet, but it is on TODO list. For now you have to connect focuser to USB hub instead of AUX port and to use "Celestron / PlaneWave EFA focuser driver".

ckuethe, 2020-12-20:
OK, thanks for confirming. I've made note of of the focuser positions for various configurations - cameras/eyepieces/reducers - so it's not a huge issue. Happy to test diffs though :)

chrsoo, 2024-02-28:
+1 - would love to see support for this!

chrsoo, 2024-04-29:
The Celestron Focus Motor has a [blocking hardware/firmware issue](https://forums.raspberrypi.com/viewtopic.php?p=2204466&hilit=celestron) for using it on a Raspberry PI 5. 

Unless there a firmware fix is possible and made available from Celestron, a workaround is to use a [USB Hat](https://www.waveshare.com/pcie-to-usb-3.2-gen1-hat-plus.htm) that circumvents the USB incompatibility between the RPI 5 and the Celestron Focus Motor.

Given that I need the HAT headers for SSD, the USB workaround is not an option in my case.

Adding AUX support for the Focus Motor would avoid the RPI 5 problem entirely.

It would also mean that the Focus Motor could work with the Hand Controller and/or the HBG3 nunchuck, without the need for switching between an USB and AUX configuration focuser. This improves the workflow if/when manual focus adjustments is needed or simply preferred in addition to using INDIGO.

Just wanted to add some additional arguments in the hope of boosting the priority of this issue. 😉