indigo-astronomy/indigo #397: Svbony SV305 or SV305 pro camera

tlkw79, 2021-01-07:
Hi,

Is there a plan to support this camera soon?

Thanks
Tom

rumengb, 2021-01-07:
Not really :(
I tried to get the SDK for that camera but it is not available for linux and mac. We do not have the hardware so we need to write the driver blindly and hope that somebody will test and report back. this is a long process and it can not happen in the next several weeks or month. So I will add it in the TODO list, contact Svbony about the sdk and we will see. This is all I can promise now.

polakovic, 2021-01-07:
I'm in contact with Svbony already. They promised linux/macOS SDK couple of times during last year or two, but I don't expect they will do it.

tlkw79, 2021-01-07:
Here you are.....

https://www.mediafire.com/file/ugd0j3m7i8t9z0f/macos_SV305Pro_SVBCameraSDK_v1.2.5.tar.gz/file

https://www.mediafire.com/file/ehsov9wzsvlrlqj/SDK1.2.5-1119.zip/file

polakovic, 2021-01-07:
Tom, do you know where is that code from? There is no license, no documentation and no technical or business contact inside.

@rumengb the header file is identical to header file of ZWO ASI SDK. If the library has also identical behaviour, it may be easy. What puzzles me is, that INDI has completely different driver for it.

tlkw79, 2021-01-07:
All these SDK files are from Svbony.  

For INDI, someone re-write the whole driver for SV305 but not for SV305 pro.

polakovic, 2021-01-07:
I'll try to clone ZWO ASI driver and we'll see.

polakovic, 2021-01-07:
Hm, ARM6 libraries are missing. SDK used by INDI-3rdparty contains only dynamic libraries and we need static :(

polakovic, 2021-01-07:
So it is not so easy :( It is a clone of some ZWO ASI SDK, but of some very old one. Different content of data structures and different function signatures. This can't be debugged without an access to the camera...

tlkw79, 2021-01-17:
ok. Will close this as well since INDI already successfully supports it well. Thanks.

closed 2021-01-17 by tlkw79