indigo-astronomy/indigo #398: Image was resolved fine with astrometry.net but didn't come back in the interface of AstroDSLR4 on first try

jurgenfd, 2021-01-12:
Hi there,

This image https://drive.google.com/file/d/1DSe-fpyZ7PH0WFkjA-6NKW6ZrmxAX133/view?usp=sharing
[WILL BE REMOVED shortly] was resolved fine with astrometry.net but didn't come back in the interface. The status was still "Image uploaded".

AstroDSLR4 actually hung and needed to be quit forcefully. The log is: [log.txt](https://github.com/indigo-astronomy/indigo/files/5804766/log.txt)

When repeating, the process did come back but the buttons Slew, Sync&Center etc. were greyed out so the image could still not
be used for slewing the mount.
[image]

[image]

https://drive.google.com/open?id=1DSe-fpyZ7PH0WFkjA-6NKW6ZrmxAX133

Cheers,
Jurgen

JohanBakk, 2021-01-12:
Hi Jurgen, Fellow user here; 

The usual issue for the buttons Slew, Sync and Sync&Center being graded out, is that the related agents are not set. In the left-bottom corner of AstroDSLR, you find two selection boxes. The top one is to select the Mount Agent of AstroTelescope. When selected,  AstroDSLR and AstroTelescope can communicate, which is required for Slew, Sync and Sync&Center. 

In order to enable the full functionality of all the apps, you should set the related agents in AstroDSLR, AstroTelescope and AstroImager to the proper agents, so all the apps can communicate with each other. AstroGuider (Guider Agent) does not need to know the related agents of the other apps, since it only serves as a server to the other apps.

Regards, Johan

jurgenfd, 2021-01-12:
That fixes this issue. Human error here. I was able to follow your suggestions and it's working fine now.
Thanks for the tips!

Please mark as INVALID/FIXED  or so. Oh wait, I can do that right ;-) ?

closed 2021-01-12 by jurgenfd