indigo-astronomy/indigo #399: Pentax DSLR not supported anymore?

tlkw79, 2021-01-17:
Hi guys,

Pentax DSLR is not supported anymore? Last time I still connect to my Pentax K5IIs. Gphoto2 is disappeared? 

Advice pls.

REgards,
Tom

polakovic, 2021-01-17:
Hi Tom, well, Pentax is not supported by cross-platform PTP-over-USB driver yet and Gphoto2 driver is not developed anymore because it has no owner :(

But if you're a programmer, have that camera and can pick the responsibility for making Pentax driver, you're welcome :)

tlkw79, 2021-01-17:
Hi,

If you put back at least the latest Gphoto2 (12-10-2020) then for older Pentax DSLR maybe can survive. What do you think?

tlkw79, 2021-01-17:
Marcus Meissner no more maintain Gphoto2? I can see his Github still active.  Weird....

polakovic, 2021-01-17:
Gphoto2 is certainly still active, I mean Gphoto2 INDIGO driver :)

I don't thing the driver will even compile and from the semantic point of view it is not compatible with INDIGO standard for DSLR anymore.

tlkw79, 2021-01-17:
ok. Then no point in using INDIGO for DSLR imaging anymore. Especially for older DSLR.  :( Thanks for the advice.

closed 2021-01-17 by tlkw79