indigo-astronomy/indigo #401: Android app for INDIGO

marcocipriani01, 2021-02-14:
_(off topic, this is not an issue, nor a feature request. I just hope someone finds this useful! 😃)_

Hello,
I don't personally use INDIGO, because I'm fine with INDI and Astroberry, but I would like to share my work with this community. I created an Android app that can control INDI devices, and some users reported me that it works with INDIGO, too. If someone is interested, it is available on [Google Play](https://play.google.com/store/apps/details?id=io.github.marcocipriani01.telescopetouch) and on [GitHub](https://github.com/marcocipriani01/Telescope.Touch). It can move the telescope, perform Go-To from a database of objects, control the focuser and receive photos from a CCD. It also integrates an interactive sky map which shows the constellations and the position of the telescope. I'm open to improvements, and I would be happy if you could test the app with more devices than the ones I own!

Kind regards,
Marco Cipriani

polakovic, 2021-02-14:
Hi Marko, thank you very much, I'll share that info on our users forum.
Peter

marcocipriani01, 2021-02-14:
Great!

ckuethe, 2021-02-14:
At the this very moment, I can't do any optical astronomy from my location, but telescope-touch does communicate with my indigo-sky. I look forward to trying it when the weather gets better.

closed 2021-11-25 by rumengb