indigo-astronomy/indigo #416: Can't set gain in web imager

ckuethe, 2021-11-17:
Kind of related to indigo-astronomy/indigo_imager#6 - I can enable streaming preview in the web ui but I can't change camera gain.  

Trace log: https://gist.github.com/6e0bb6b55f8e0e343fc1b7b5c73622df
[image]

rumengb, 2021-11-17:
Web Imager is supposed to provide only basic functionality and not a complete imaging solution. However you can change every single property from the web GUI, just go to "Control panel" tab then Imager Agent-> Camera -> Gain:
[image]

closed 2021-11-17 by rumengb

ckuethe, 2021-11-17:
Yes, I use the control panel tab a lot - I just thought it would be helpful to have that that relatively basic control in the imager tab.