indigo-astronomy/indigo #417: Sky watcher - Synscan mount error -  ERROR GETTING FIRMWARE

duckchip, 2021-11-20:
Hi guys,

I'm really new in the world of telescopes so forgive me for my noobish questions :) 

I am trying to setup a Sky-watcher telescope with AstroTelescope and Indi control panel on Mac OS Monterey. 
I use the Synscan Mount and received the following error message.

16:40:58.967 'Mount SynScan' selected
16:41:04.026 indigo_mount_synscan[synscan_read_response:153, 0x70000586d000]: SYNSCAN_TIMEOUT
16:41:04.029 indigo_mount_synscan[synscan_read_response:164, 0x70000586d000]: Reading response failed
16:41:09.041 indigo_mount_synscan[synscan_read_response:153, 0x70000586d000]: SYNSCAN_TIMEOUT
16:41:09.041 indigo_mount_synscan[synscan_read_response:164, 0x70000586d000]: Reading response failed
16:41:09.041 indigo_mount_synscan[synscan_configure:109, 0x70000586d000]: ERROR GETTING FIRMWARE
16:41:09.068 Trying to change baudrate
16:41:14.112 indigo_mount_synscan[synscan_read_response:153, 0x70000586d000]: SYNSCAN_TIMEOUT
16:41:14.112 indigo_mount_synscan[synscan_read_response:164, 0x70000586d000]: Reading response failed
16:41:19.117 indigo_mount_synscan[synscan_read_response:153, 0x70000586d000]: SYNSCAN_TIMEOUT
16:41:19.117 indigo_mount_synscan[synscan_read_response:164, 0x70000586d000]: Reading response failed
16:41:19.117 indigo_mount_synscan[synscan_configure:109, 0x70000586d000]: ERROR GETTING FIRMWARE
16:41:19.146 Failed to connect to mount
16:41:19.162 Failed to connect

If I disconnect the telescope completely from my computer, I don't get these errors so it seems there is some sort of connection.  
I tried connecting it via the controller and directly into to telescope which ended in the same result. 

Any idea what might be wrong? 

Thx in advance!!

polakovic, 2021-11-20:
What serial port did you use? Check it in AstroTelescope > Preferences > INDIGO Ports.

SynScan driver is intended for the direct control (EQDir or similar devices). So if you used hand controller as the interface, you ma need to use NexStar driver.

duckchip, 2021-11-21:
Hi @polakovic, thx for your fast reply! 

Port I'm using is /dev/cu.usbserial-141220

Selecting Mount Nexstar results in the following error

11:49:48.382 indigo_mount_nexstar[position_timer_callback:176, 0x70000fb33000]: tc_get_tracking_mode(31) = -5 (Invalid argument)
11:49:49.032 indigo_mount_nexstar[position_timer_callback:160, 0x70000fb33000]: tc_get_location(31) = -5 (Unknown error: 316)
11:49:49.032 indigo_mount_nexstar[position_timer_callback:176, 0x70000fb33000]: tc_get_tracking_mode(31) = -5 (Invalid argument)
11:49:49.681 indigo_mount_nexstar[position_timer_callback:160, 0x70000fb33000]: tc_get_location(31) = -5 (Unknown error: 316)
11:49:49.695 indigo_mount_nexstar[position_timer_callback:176, 0x70000fb33000]: tc_get_tracking_mode(31) = -5 (Invalid argument)

So this is a step closer 👍 but we're not there yet unfortunately

polakovic, 2021-11-21:
OK, first of all, how do you connect the mount (port on the hand controller, port on the mount, EQDir, ...) and what mount model is it?

duckchip, 2021-11-21:
The controller is running firmware version SynScan TM Ver. 03.09.

I fear that the outdated version might be the reason.. What do you think? 

We're are using a serial connector with RJ-11 (going in the controller) and a USB-B connector that is plugged in into a usb dongle, connected to my MacBook.

polakovic, 2021-11-21:
In this case NexStar driver is definitely the right one. SynScan version is outdated, but should be OK, most of the features is supported since 3.03.

This error looks like the controller doesn't respond on the giver serial port. Can you make a trace log? We need to see the communication with the mount from the very beginning.

Check this: https://bb.cloudmakers.eu/viewtopic.php?f=5&t=632

duckchip, 2021-11-21:
Complete log: 

```
12:04:26.026 'No mount' selected
12:08:32.465 ACL: set master_token = 0x0
12:08:33.131 Server started on 51563
12:08:33.169 INDIGO Service AstroTelescope is ready to publish
12:08:33.333 INDIGO Service AstroTelescope was removed
12:08:36.752 Can't connect to socket 127.0.0.1:7624 (Connection refused)
12:08:39.816 Worker thread started socket = 15
12:08:39.849 Protocol switched to XML
12:08:40.273 'Mount Nexstar' selected
12:08:40.497 libnexstar: write 56
12:08:40.527 libnexstar: read 02 02 23
12:08:40.527 libnexstar: write 56
12:08:40.527 libnexstar: read 02 02 23
12:08:40.527 libnexstar: write 56
12:08:40.527 libnexstar: read 02 02 23
12:08:40.527 libnexstar: write 76
12:08:41.754 Can't connect to socket 127.0.0.1:7624 (Connection refused)
12:08:45.521 libnexstar: read FAILED
12:08:45.556 libnexstar: write 6d
12:08:45.557 libnexstar: read 01 23
12:08:45.557 libnexstar: write 6d
12:08:45.557 libnexstar: read 01 23
12:08:45.557 libnexstar: write 56
12:08:45.557 libnexstar: read 02 02 23
12:08:45.557 libnexstar: write 76
12:08:46.756 Can't connect to socket 127.0.0.1:7624 (Connection refused)
12:08:50.545 libnexstar: read FAILED
12:08:50.576 indigo_mount_nexstar[mount_handle_connect:355, 0x70000fbb6000]: tc_get_tracking_mode(31) = -5 (Undefined error: 0)
12:08:50.577 indigo_mount_nexstar: 'Mount Nexstar (gps)' attached
12:08:50.695 Can't open /Users/duckchip/Library/Containers/eu.cloudmakers.AstroTelescope/Data/.indigo/Mount_Nexstar_(gps).config (No such file or directory)
12:08:50.787 libnexstar: write 4c
12:08:50.787 libnexstar: read 30 23
12:08:50.787 libnexstar: write 65
12:08:50.913 libnexstar: read 37 39 41 34 35 42 30 30 2c 31 42 37 46 46 42 30 30 23
12:08:50.914 indigo_mount_nexstar[position_timer_callback:160, 0x70000fb33000]: tc_get_location(31) = -5 (Invalid argument)
12:08:50.914 indigo_mount_nexstar[position_timer_callback:176, 0x70000fb33000]: tc_get_tracking_mode(31) = -5 (Invalid argument)
12:08:51.289 Config file is locked, property 'Mount Nexstar.DEVICE_PORT' not saved
12:08:51.364 libnexstar: write 54 00
12:08:51.364 libnexstar: read 23
12:08:51.411 indigo_mount_nexstar[mount_handle_geo_coordinates:489, 0x700010051000]: tc_set_location(31) = -5 (Invalid argument)
12:08:51.411 indigo_mount_nexstar[mount_handle_st4_guiding_rate:620, 0x70000fdc2000]: tc_set_autoguide_rate(31) = -5 (Invalid argument)
12:08:51.411 indigo_mount_nexstar[mount_handle_st4_guiding_rate:632, 0x70000fdc2000]: tc_set_autoguide_rate(31) = -5 (Invalid argument)
12:08:51.412 indigo_mount_nexstar[mount_handle_st4_guiding_rate:643, 0x70000fdc2000]: tc_get_autoguide_rate(31) = -5 (Invalid argument)
12:08:51.412 indigo_mount_nexstar[mount_handle_st4_guiding_rate:651, 0x70000fdc2000]: tc_get_autoguide_rate(31) = -5 (Invalid argument)
12:08:51.523 Config file is locked, property 'Mount Nexstar.DEVICE_PORT' not saved
12:08:51.762 Can't connect to socket 127.0.0.1:7624 (Connection refused)
12:08:51.800 libnexstar: write 4c
12:08:51.800 libnexstar: read 30 23
12:08:51.800 libnexstar: write 65
12:08:51.919 libnexstar: read 37 39 41 35 32 30 30 30 2c 31 42 37 46 46 42 30 30 23
12:08:51.920 indigo_mount_nexstar[position_timer_callback:160, 0x70000fb33000]: tc_get_location(31) = -5 (Unknown error: 316)
12:08:51.920 indigo_mount_nexstar[position_timer_callback:176, 0x70000fb33000]: tc_get_tracking_mode(31) = -5 (Invalid argument)
12:08:52.585 libnexstar: write 4c
12:08:52.611 libnexstar: read 30 23
12:08:52.611 libnexstar: write 65
12:08:52.723 libnexstar: read 37 39 41 35 45 30 30 30 2c 31 42 37 46 46 42 30 30 23
12:08:52.724 indigo_mount_nexstar[position_timer_callback:160, 0x70000fb33000]: tc_get_location(31) = -5 (Unknown error: 316)
12:08:52.724 indigo_mount_nexstar[position_timer_callback:176, 0x70000fb33000]: tc_get_tracking_mode(31) = -5 (Invalid argument)
12:08:52.724 indigo_mount_nexstar[mount_handle_geo_coordinates:489, 0x7000100d4000]: tc_set_location(31) = -5 (Unknown error: 316)
12:08:52.818 indigo_mount_nexstar[mount_handle_set_utc_from_host:517]: tc_set_time: '21/11/2021 12:08:52 +0'
12:08:52.818 indigo_mount_nexstar[mount_handle_set_utc_from_host:524, 0x70000fdc2000]: tc_set_time(31) = -5 (Invalid argument)
12:08:52.818 ACL: Token found '' = 0x0
12:08:53.410 libnexstar: write 4c
12:08:53.435 libnexstar: read 30 23
12:08:53.435 libnexstar: write 65
12:08:53.545 libnexstar: read 37 39 41 36 41 33 30 30 2c 31 42 37 46 46 42 30 30 23
12:08:53.546 indigo_mount_nexstar[position_timer_callback:160, 0x70000fb33000]: tc_get_location(31) = -5 (Unknown error: 316)
[... 39 lines cut ...]
12:09:02.768 libnexstar: write 4c
12:09:02.804 libnexstar: read 30 23
12:09:02.805 libnexstar: write 65
12:09:02.916 libnexstar: read 37 39 41 44 37 44 30 30 2c 31 42 37 46 46 42 30 30 23
12:09:02.955 indigo_mount_nexstar[position_timer_callback:160, 0x70000fb33000]: tc_get_location(31) = -5 (Unknown error: 316)
12:09:02.985 indigo_mount_nexstar[position_timer_callback:176, 0x70000fb33000]: tc_get_tracking_mode(31) = -5 (Invalid argument)
12:09:03.593 libnexstar: write 4c
12:09:03.624 libnexstar: read 30 23
12:09:03.661 libnexstar: write 65
12:09:03.789 libnexstar: read 37 39 41 45 33 44 30 30 2c 31 42 37 46 46 42 30 30 23
12:09:03.809 indigo_mount_nexstar[position_timer_callback:160, 0x70000fb33000]: tc_get_location(31) = -5 (Unknown error: 316)
12:09:03.821 indigo_mount_nexstar[position_timer_callback:176, 0x70000fb33000]: tc_get_tracking_mode(31) = -5 (Invalid argument)
12:09:04.611 libnexstar: write 4c
[... cut ...]
12:09:04.854 indigo_mount_nexstar[position_timer_callback:176, 0x70000fb33000]: tc_get_tracking_mode(31) = -5 (Invalid argument)
12:09:05.479 libnexstar: write 4c
12:09:05.517 libnexstar: read 30 23
12:09:05.555 libnexstar: write 65
12:09:05.641 libnexstar: read 37 39 41 46 43 35 30 30 2c 31 42 37 46 46 42 30 30 23
12:09:05.642 indigo_mount_nexstar[position_timer_callback:160, 0x70000fb33000]: tc_get_location(31) = -5 (Unknown error: 316)
12:09:05.677 indigo_mount_nexstar[position_timer_callback:176, 0x70000fb33000]: tc_get_tracking_mode(31) = -5 (Invalid argument)
```

duckchip, 2021-11-21:
Log with status backtrace:

```
12:14:14.942 'Mount Nexstar' selected
12:14:15.206 INDIGO Bus: property definition
12:14:15.206 'Server'.'DRIVERS' SWITCH rw Ok 2.0 0 AnyOfMany { // Drivers
12:14:15.206   'indigo_aux_cloudwatcher' = Off // AAG CloudWatcher
12:14:15.208   'indigo_aux_arteskyflat' = Off // Artesky Flat Box USB
12:14:15.208   'indigo_aux_astromechanics' = Off // ASTROMECHANICS LPM
12:14:15.208   'indigo_aux_mgbox' = Off // Astromi.ch MGBox
12:14:15.209   'indigo_dome_baader' = Off // Baader Classic Dome
12:14:15.209   'indigo_dome_simulator' = Off // Dome Simulator
12:14:15.209   'indigo_gps_nmea' = Off // Generic NMEA 0183 GPS
12:14:15.209   'indigo_aux_geoptikflat' = Off // Geoptik Flat Generator
12:14:15.210   'indigo_gps_gpsd' = Off // GPS Sevice Daemon (GPSD) Client
12:14:15.210   'indigo_gps_simulator' = Off // GPS Simulator
12:14:15.210   'indigo_aux_joystick' = Off // HID Joystick
12:14:15.210   'indigo_aux_skyalert' = Off // Interactive Astronomy SkyAlert
12:14:15.210   'indigo_dome_skyroof' = Off // Interactive Astronomy SkyRoof
12:14:15.210   'indigo_mount_ioptron' = Off // iOptron Mount
12:14:15.211   'indigo_aux_fbc' = Off // Lacerta FBC
12:14:15.211   'indigo_dome_dragonfly' = Off // Lunatico Dragonfly Dome
12:14:15.211   'indigo_aux_dragonfly' = Off // Lunatico Dragonfly Relay Controller
12:14:15.211   'indigo_mount_lx200' = Off // LX200 Mount
12:14:15.211   'indigo_mount_simulator' = Off // Mount Simulator
12:14:15.211   'indigo_dome_nexdome' = Off // NexDome
12:14:15.211   'indigo_dome_beaver' = Off // Nexdome Beaver Dome
12:14:15.211   'indigo_dome_nexdome3' = Off // NexDome3
12:14:15.211   'indigo_mount_nexstaraux' = Off // NexStar AUX Mount
12:14:15.211   'indigo_mount_nexstar' = On // Nexstar Mount
12:14:15.211   'indigo_aux_flipflat' = Off // Optec Flip-Flat
12:14:15.211   'indigo_aux_flatmaster' = Off // Pegasus Astro FlatMaster
12:14:15.212   'indigo_aux_ppb' = Off // PegasusAstro Pocket Powerbox
12:14:15.212   'indigo_aux_upb' = Off // PegasusAstro Ultimate Powerbox
12:14:15.212   'indigo_mount_pmc8' = Off // PMC Eight Mount
12:14:15.212   'indigo_mount_rainbow' = Off // RainbowAstro Mount
12:14:15.212   'indigo_mount_synscan' = Off // SynScan Mount
12:14:15.212   'indigo_mount_temma' = Off // Takahashi Temma Mount
12:14:15.212   'indigo_dome_talon6ror' = Off // Talon6 ROR
12:14:15.212   'indigo_aux_sqm' = Off // Unihedron SQM
12:14:15.212   'indigo_aux_usbdp' = Off // USB Dewpoint
12:14:15.212 }
12:14:15.885 INDIGO Bus: property change request
12:14:15.886 'Mount Agent'.'FILTER_MOUNT_LIST' SWITCH UNDEFINED Idle 2.0 0 UNDEFINED {
12:14:15.886   'Mount Nexstar' = On 
12:14:15.922 }
12:14:15.922 INDIGO Bus: Change request - Device 'Mount Agent' token 0x0, Proprerty 'FILTER_MOUNT_LIST' token 0x0
12:14:15.922 INDIGO Bus: property update
12:14:15.922 'Mount Agent'.'FILTER_MOUNT_LIST' SWITCH rw Busy 2.0 0 OneOfMany {
12:14:15.923   'NONE' = On 
12:14:15.923   'Mount Nexstar' = Off 
12:14:15.923 }
12:14:16.134 INDIGO Bus: property update
12:14:16.134 'Mount Agent'.'FILTER_MOUNT_LIST' SWITCH rw Busy 2.0 0 OneOfMany {
12:14:16.134   'NONE' = Off 
12:14:16.134   'Mount Nexstar' = On 
12:14:16.134 }
12:14:16.277 INDIGO Bus: property enumeration request
12:14:16.278 'Mount Nexstar'.'CONNECTION' SWITCH rw Ok 2.0 0 OneOfMany {
12:14:16.278 }
12:14:16.278 INDIGO Bus: property definition
12:14:16.278 'Mount Nexstar'.'CONNECTION' SWITCH rw Ok 2.0 0 OneOfMany { // Connection status
12:14:16.278   'CONNECTED' = Off // Connected
12:14:16.279   'DISCONNECTED' = On // Disconnected
12:14:16.279 }
12:14:16.469 INDIGO Bus: property update
12:14:16.470 'Mount Agent'.'FILTER_MOUNT_LIST' SWITCH rw Busy 2.0 0 OneOfMany {
12:14:16.470   'NONE' = Off 
12:14:16.471   'Mount Nexstar' = On 
12:14:16.471 }
[... 2607 lines cut ...]
12:15:35.806 indigo_mount_nexstar[position_timer_callback:176, 0x70000cc5d000]: tc_get_tracking_mode(11) = -5 (Invalid argument)
12:15:36.812 indigo_mount_nexstar[position_timer_callback:160, 0x70000cc5d000]: tc_get_location(11) = -5 (Unknown error: 316)
12:15:36.812 indigo_mount_nexstar[position_timer_callback:176, 0x70000cc5d000]: tc_get_tracking_mode(11) = -5 (Invalid argument)
12:15:36.997 Can't connect to socket 127.0.0.1:7624 (Connection refused)
12:15:38.215 indigo_mount_nexstar[position_timer_callback:160, 0x70000cc5d000]: tc_get_location(11) = -5 (Unknown error: 316)
12:15:38.215 indigo_mount_nexstar[position_timer_callback:176, 0x70000cc5d000]: tc_get_tracking_mode(11) = -5 (Invalid argument)
12:15:39.026 indigo_mount_nexstar[position_timer_callback:160, 0x70000cc5d000]: tc_get_location(11) = -5 (Unknown error: 316)
12:15:39.027 indigo_mount_nexstar[position_timer_callback:176, 0x70000cc5d000]: tc_get_tracking_mode(11) = -5 (Invalid argument)
12:15:40.040 indigo_mount_nexstar[position_timer_callback:160, 0x70000cc5d000]: tc_get_location(11) = -5 (Unknown error: 316)
12:15:40.041 indigo_mount_nexstar[position_timer_callback:176, 0x70000cc5d000]: tc_get_tracking_mode(11) = -5 (Invalid argument)
12:15:40.794 indigo_mount_nexstar[position_timer_callback:160, 0x70000cc5d000]: tc_get_location(11) = -5 (Unknown error: 316)
12:15:40.794 indigo_mount_nexstar[position_timer_callback:176, 0x70000cc5d000]: tc_get_tracking_mode(11) = -5 (Invalid argument)
12:15:41.523 indigo_mount_nexstar[position_timer_callback:160, 0x70000cc5d000]: tc_get_location(11) = -5 (Unknown error: 316)
[... cut ...]
12:15:46.313 indigo_mount_nexstar[position_timer_callback:160, 0x70000cc5d000]: tc_get_location(11) = -5 (Unknown error: 316)
12:15:46.313 indigo_mount_nexstar[position_timer_callback:176, 0x70000cc5d000]: tc_get_tracking_mode(11) = -5 (Invalid argument)
12:15:47.313 Can't connect to socket 127.0.0.1:7624 (Connection refused)
12:15:48.291 indigo_mount_nexstar[position_timer_callback:160, 0x70000cc5d000]: tc_get_location(11) = -5 (Unknown error: 316)
12:15:48.291 indigo_mount_nexstar[position_timer_callback:176, 0x70000cc5d000]: tc_get_tracking_mode(11) = -5 (Invalid argument)
```

polakovic, 2021-11-21:
This part

12:08:40.497 libnexstar: write 56
12:08:40.527 libnexstar: read 02 02 23

shows it is definitely not SynScan hand controller on the other side. So the selected serial port is not the right one.

closed 2021-11-25 by rumengb