indigo-astronomy/indigo #421: Celestron Neximage 10 images aren't de-bayered

ckuethe, 2021-11-25:
Raw image from the sensor.

[image]

After de-bayering with siril, and I had to tell siril that this camera uses a GRBG bayer pattern :(
[image]

Neither Ain nor the imager web app are able to grab this as a color image, which suggests to me the problem is somewhere closer to ccd_uvc, possibly even the UVC kernel driver itself. Neither `guvcview` nor `uvccapture` on this armhf machine are able to capture from the camera, though indigo can.

Could you add a debayering control to the uvc driver?

rumengb, 2021-11-25:
Drivers do not debayer. They only write the correct pattern in the FITS header if known. Ain and other apps use this to debayer the image. Ain debayers only for preview. There are several algorithms to do so each with its strengths and weaknesses. So we leave this to the processing software. UVC driver does not know about the pattern. Maybe we can hard code this fr this camera but still it is not a 100% universal solution. When you use ROI it may turn in to other pattern. Well it can be handled for sure. But indigo will not debayer it anyway. We can only add hint for the processing software.

ckuethe, 2021-11-25:
Here's what v4l2-ctl thinks about this camera.

```
Driver Info:
	Driver name      : uvcvideo
	Card type        : NexImage 10
	Bus info         : usb-1c1c000.usb-1
	Driver version   : 5.4.88
	Capabilities     : 0x84a00001
		Video Capture
		Metadata Capture
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x04200001
		Video Capture
		Streaming
		Extended Pix Format
Media Driver Info:
	Driver name      : uvcvideo
	Model            : NexImage 10
	Serial           : 29914056
	Bus info         : usb-1c1c000.usb-1
	Media version    : 5.4.88
	Hardware revision: 0x00000000 (0)
	Driver version   : 5.4.88
Interface Info:
	ID               : 0x03000002
	Type             : V4L Video
Entity Info:
	ID               : 0x00000001 (1)
	Name             : NexImage 10
	Function         : V4L2 I/O
	Flags         : default
	Pad 0x01000007   : 0: Sink
	  Link 0x02000010: from remote pad 0x100000a of entity 'Extension 3': Data, Enabled, Immutable
Priority: 2
Video input : 0 (Camera 1: ok)
Format Video Capture:
	Width/Height      : 3872/2764
	Pixel Format      : 'GRBG' (8-bit Bayer GRGR/BGBG)
	Field             : None
	Bytes per Line    : 3872
	Size Image        : 10702208
	Colorspace        : Default
	Transfer Function : Default (maps to Rec. 709)
	YCbCr/HSV Encoding: Default (maps to ITU-R 601)
	Quantization      : Default (maps to Full Range)
	Flags             : 
Crop Capability Video Capture:
	Bounds      : Left 0, Top 0, Width 3872, Height 2764
	Default     : Left 0, Top 0, Width 3872, Height 2764
	Pixel Aspect: 1/1
Selection Video Capture: crop_default, Left 0, Top 0, Width 3872, Height 2764, Flags: 
Selection Video Capture: crop_bounds, Left 0, Top 0, Width 3872, Height 2764, Flags: 
Streaming Parameters Video Capture:
	Capabilities     : timeperframe
	Frames per second: 30.000 (30/1)
	Read buffers     : 0
                     brightness 0x00980900 (int)    : min=0 max=4095 step=1 default=168 value=168
                           gain 0x00980913 (int)    : min=100 max=383 step=1 default=100 value=100
              exposure_absolute 0x009a0902 (int)    : min=1 max=300000 step=1 default=333 value=333

```

I do see the bayer pattern being detected (https://gist.github.com/ckuethe/3433490385825c693bc6a5f879deb9fa#file-gistfile0-txt-L671) but it's not being written to the output FITS files.

I did a bunch of captures in RAW8, MONO8, unspecified; here are the headers... https://gist.github.com/9829dc7866265f460cad4e2d10ea68f7

Strangely I was unable to capture to MONO16, even though it claims to be supported. I see some broken pipe errors in the trace: https://gist.github.com/fb1a87932f05820ee8355c54987ad05d

polakovic, 2021-11-26:
Unfortunately, underlying libuvc library doesn't have API for bayer mask, so we can't set it on FITS images. But it is a good point, I'll look if it can be retrieved from over UVC. We don't use v4l2, it is not portable on macOS.

ckuethe, 2021-11-26:
Have another look at https://gist.github.com/ckuethe/3433490385825c693bc6a5f879deb9fa - I removed some of the irrelevant stuff at the beginning so you can see that on line 65 `indigo_ccd_uvc` detects the bayer pattern: `indigo_server: 		  GUID: 4752424700001000800000aa00389b71  (GRBG)`