indigo-astronomy/indigo #423: Canon550D no preview image with ptp driver

vdelord, 2021-12-11:
When I use the Imager Agent to try to use my DSLR Canon 550D (Kissx4) connected though USB, I hear the shutter release, the picture ends up in the capture directory but I get no image preview in the GUI.

It doesn't work on the Windows hard Client nor on the webclient.
My setup is distributed, I have all my devices connected to an IndigoSky raspberry.

I also have a Zwo camera for guiding and it's working fine.

This prevent me to do Focus evaluation.

rumengb, 2021-12-11:
I suppose you use Ain Imager? Are you? If so, it does not support native camera raw formats for preview yet. This is the TODO. So if you are shooting raw it will be downloaded and saved but not displayed.

vdelord, 2021-12-12:
Ah ok, sorry I didn't check the feature list 👍 
It will prevent me to use the Focusing tool but not the plate solving right ?

Thanks for your quick response.

rumengb, 2021-12-12:
For platesolving you can switch to jpeg frames. I am not sure how the autofocus will work. It uses INDIGO raw format but I am not sure the ptp driver will produce such frames. DSLRs are special case to be honest I never used DSLR for astro photography. Maybe Peter (he is the author of this driver) can answer this question.

polakovic, 2021-12-13:
Yes, it should work with JPEG format.

closed 2023-11-18 by rumengb