indigo-astronomy/indigo #427: Allow wheel offset to be grater than +-9999

Paolo97Gll, 2021-12-27:
I use a custom focuser driver which has a step range from 0 to 800000. Now the wheel offset limits are hard-coded to be between -9999 and 9999:

https://github.com/indigo-astronomy/indigo/blob/79c7e1543aad0b5c12f3accab1962c509f53f46e/indigo_libs/indigo_wheel_driver.c#L71

This prevent me to set the correct offset for my wheel (a sx wheel), since for some filter I have an offset of 15000 or 20000. It would be nice to add the possibility to change these limits in the driver settings!

polakovic, 2021-12-29:
fixed here: 039fdfa2948327a8c018634a9fbbf1b5d59d47f0

BTW, what driver did you write?

closed 2021-12-29 by polakovic

Paolo97Gll, 2021-12-29:
Thank you!

I write a driver for the PrimaLuceLab ESATTO 3". However, the setup of our dome is quite particular and so is the driver: we use HTTP GET requests to communicate, and the ESATTO is connected with a reverse proxy to a Raspberry Pi to make it reachable from all the LAN network (unfortunately the focuser has builtin wifi, but only in AP mode).