indigo-astronomy/indigo #429: No Serial Connect to PMC-8

grisu-112, 2022-01-02:
Wish you a happy new year 2022!

I am trying to connect to a new PMC-8 via serial port. 

indigo.log says 
`indigo_mount_pmc8[pmc8_command:198]: Command ESGv! failed`

A connection can be established and the mount controlled via a terminal program. Therefore I assume that the FTDI based USB-RS232 cable is working.

I retrieved the following TTY settings using the console command 
`stty -a -F /dev/ttyUSB0`

1. Working terminal session:
```
speed 115200 baud; rows 0; columns 0; line = 0;
intr = ^C; quit = ^\; erase = ^?; kill = ^U; eof = ^D; eol = <undef>; eol2 = <undef>; swtch = <undef>; start = ^Q; stop = ^S; susp = ^Z; rprnt = ^R;
werase = ^W; lnext = ^V; discard = ^O; min = 0; time = 0;
-parenb -parodd -cmspar cs8 hupcl -cstopb cread clocal -crtscts
-ignbrk -brkint -ignpar -parmrk -inpck -istrip -inlcr -igncr -icrnl -ixon -ixoff -iuclc -ixany -imaxbel -iutf8
-opost -olcuc -ocrnl onlcr -onocr -onlret -ofill -ofdel nl0 cr0 tab0 bs0 vt0 ff0
-isig -icanon -iexten -echo -echoe -echok -echonl -noflsh -xcase -tostop -echoprt -echoctl -echoke -flusho -extproc
```

2. Non-working INDIGO serial connection attempt:
```
speed 115200 baud; rows 0; columns 0; line = 0;
intr = <undef>; quit = <undef>; erase = <undef>; kill = <undef>; eof = <undef>; eol = <undef>; eol2 = <undef>; swtch = <undef>; start = <undef>;
stop = <undef>; susp = <undef>; rprnt = <undef>; werase = <undef>; lnext = <undef>; discard = <undef>; min = 0; time = 50;
-parenb -parodd -cmspar cs8 -hupcl -cstopb cread clocal -crtscts
-ignbrk -brkint ignpar -parmrk -inpck -istrip -inlcr -igncr -icrnl -ixon -ixoff -iuclc -ixany -imaxbel -iutf8
-opost -olcuc -ocrnl -onlcr -onocr -onlret -ofill -ofdel nl0 cr0 tab0 bs0 vt0 ff0
-isig -icanon -iexten -echo -echoe -echok -echonl -noflsh -xcase -tostop -echoprt -echoctl -echoke -flusho -extproc
```

Used Configuration:
INDIGO-SKY (Build downloaded in the last three days) updated to 2.0-166.
PMC-8 with newest "Universial Firmware" (ESGv! Result: `ESGvES20A01.1 Release 2021.08.08: ES1A01CB20A01, ESP8266, DST4!`).

Full indigo.log (Trace-Level):
```
13:03:50.945375 indigo_server: INDIGO Bus: property update
13:03:50.945749 indigo_server: 'Server'.'LOG_LEVEL' SWITCH rw Ok 2.0 0 (null) {
13:03:50.945822 indigo_server:   'ERROR' = Off 
13:03:50.945887 indigo_server:   'INFO' = Off 
13:03:50.945947 indigo_server:   'DEBUG' = Off 
13:03:50.946007 indigo_server:   'TRACE' = On 
13:03:50.946065 indigo_server: }
13:03:50.948059 indigo_server: 26 ← { "setSwitchVector": { "device": "Server", "name": "LOG_LEVEL", "state": "Ok", "items": [  { "name": "ERROR", "value": false }, { "name": "INFO", "value": f
alse }, { "name": "DEBUG", "value": false }, { "name": "TRACE", "value": true } ] } }
13:03:50.948400 indigo_server: 27 ← { "setSwitchVector": { "device": "Server", "name": "LOG_LEVEL", "state": "Ok", "items": [  { "name": "ERROR", "value": false }, { "name": "INFO", "value": f
alse }, { "name": "DEBUG", "value": false }, { "name": "TRACE", "value": true } ] } }
13:04:03.088441 indigo_server: JSON Parser: parser finished
13:04:03.088630 indigo_server: INDIGO Bus: client detach request (JSON Driver Adapter)
13:04:03.088726 indigo_server: 7 clients
13:04:03.088794 indigo_server: Worker thread finished
13:04:09.404698 indigo_server: 27 → {"newSwitchVector":{"device":"Mount PMC Eight","name":"CONNECTION","items":[{"name":"CONNECTED","value":true}]}}
13:04:09.405223 indigo_server: INDIGO Bus: property change request
13:04:09.405301 indigo_server: 'Mount PMC Eight'.'CONNECTION' SWITCH UNDEFINED Idle 2.0 0 (null) {
13:04:09.405369 indigo_server:   'CONNECTED' = On 
13:04:09.405433 indigo_server: }
13:04:09.405501 indigo_server: INDIGO Bus: Change request - Device 'Mount PMC Eight' token 0x0, Proprerty '(null)' token 0xae580810
13:04:09.405620 indigo_server: INDIGO Bus: property update
13:04:09.405692 indigo_server: 'Mount PMC Eight'.'CONNECTION' SWITCH rw Busy 2.0 0 (null) {
13:04:09.405756 indigo_server:   'CONNECTED' = On 
13:04:09.405820 indigo_server:   'DISCONNECTED' = Off 
13:04:09.405883 indigo_server: }
13:04:09.406428 indigo_server: 27 ← { "setSwitchVector": { "device": "Mount PMC Eight", "name": "CONNECTION", "state": "Busy", "items": [  { "name": "CONNECTED", "value": true }, { "name": "DI
SCONNECTED", "value": false } ] } }
13:04:09.406728 indigo_server: timer #0 (of 1) used for 0s
13:04:09.406867 indigo_server: timer callback: 0xb4a04870 started
[... 58 lines cut ...]
13:04:10.537027 indigo_server: }
13:04:10.537347 indigo_server: 27 ← { "deleteProperty": { "device": "Mount PMC Eight", "name": "MOUNT_EPOCH" } }
13:04:10.537413 indigo_server: INDIGO Bus: property removal
13:04:10.537474 indigo_server: 'Mount PMC Eight'.'MOUNT_SIDE_OF_PIER' SWITCH ro Ok 2.0 0 (null) {
13:04:10.537531 indigo_server: }
13:04:10.537851 indigo_server: 27 ← { "deleteProperty": { "device": "Mount PMC Eight", "name": "MOUNT_SIDE_OF_PIER" } }
13:04:10.537916 indigo_server: INDIGO Bus: property removal
13:04:10.537979 indigo_server: 'Mount PMC Eight'.'SNOOP_DEVICES' TEXT rw Ok 2.0 0 (null) {
13:04:10.538036 indigo_server: }
13:04:10.538357 indigo_server: 27 ← { "deleteProperty": { "device": "Mount PMC Eight", "name": "SNOOP_DEVICES" } }
13:04:10.538442 indigo_server: INDIGO Bus: property update
13:04:10.538505 indigo_server: 'Mount PMC Eight'.'CONNECTION' SWITCH rw Alert 2.0 0 (null) {
13:04:10.538564 indigo_server:   'CONNECTED' = Off 
13:04:10.538623 indigo_server:   'DISCONNECTED' = On 
13:04:10.538681 indigo_server: }
13:04:10.539097 indigo_server: 27 ← { "setSwitchVector": { "device": "Mount PMC Eight", "name": "CONNECTION", "state": "Alert", "items": [  { "name": "CONNECTED", "value": false }, { "name": "
DISCONNECTED", "value": true } ] } }
13:04:10.539212 indigo_server: timer callback: 0xb4a04870 finished
13:04:10.539275 indigo_server: timer #0 done
```

My search on the forum and on the search engine were unsuccessful. I am a beginner regarding INDIGO and hope to have given all the important information. 

Greetings from Germany.
Chris

polakovic, 2022-01-02:
Happy new year to you to! :)

There are two surprising issues - the first one is that the log shows, that ESGv! command doesn't reply anything and the second one is that in "ESGvES20A01.1 Release 2021.08.08: ES1A01CB20A01, ESP8266, DST4!" reply is no mount identification. What mount is it?

I'll need to upgrade my PCM-8 to find what's going on :(

grisu-112, 2022-01-02:
Hi,

I was also surprised and a bit overwhelmed as it is my first electronic mount and the feedback on "ESGv!" does not match the feedback given in the Programmer's Reference (DOC-ESPMC8-002 Release 2 2019_March 07).

Sorry! The mount is of type EXOS-II.

The firmware has to be universal now. I assume that e.g. the trace rates can be read out, so that no type would have to be read out via "ESGv!

I hope you can solve this.

polakovic, 2022-01-03:
To be honest, I think that the new firmware is broken :( The serial port either doesn't work at all or it reboots each time the is connected, that's why it doesn't connect, it needs some time to boot and to eat the splash screen...

grisu-112, 2022-01-03:
Oh, you're right. I haven't noticed that yet. After the splash screen, the version query and further communication seem to be successful. (At least, it looks like that to me.)
Could it be a solution to repeat the version query ("ESGv!") after say 10 seconds, without disconnecting after the first failed attempt, before the connection attempt is considered as failed? This should work equally well for both old and new firmware versions.

polakovic, 2022-01-03:
Yes, I'll do it. I'll also new mount type detection (for firmware 2+) and the possibility to do it manually if needed.

polakovic, 2022-01-03:
Pls. can you try it now? c90c7565b00e406db9be72fd2926c80002503f69

polakovic, 2022-01-03:
I made some more changes to handle some slewing inaccuracy...

grisu-112, 2022-01-03:
I apologize for the following question (my programming skills are pretty rusty): 
I have installed indigo-sky (SD Image) and now downloaded the source code from github.
Do I now need to run `make all` and replace indigo_mount_pmc8 (under /usr/bin) with the version from the build directory (build/drivers)?

polakovic, 2022-01-03:
Hm, it may be quite complicated on the pre-configured indigo-sky image :(

1. You need to install complete development tools

`sudo apt-get install build-essential autoconf autotools-dev libtool cmake libudev-dev libavahi-compat-libdnssd-dev libusb-1.0-0-dev libcurl4-gnutls-dev libz-dev git curl bsdmainutils`

2. stop default indigo service

`systemctl stop`

3. build binaries to the staging area

`make all`

4. run it from the staging area (providing your're in the root of the source files)

`build/bin/indigo_server -vvv`

Am I right @rumengb ?

grisu-112, 2022-01-03:
Hello Peter,

Thanks for the tutorial. I was able to start the build and could see the new driver in the indigo control panel now.

But, unfortunately the serial connection still does not work.

By running from staging area I got the following log in the stdout:
```
18:34:23.066261 indigo_server: indigo_mount_pmc8: Connected to /dev/ttyUSB0
18:34:24.178738 indigo_server: indigo_mount_pmc8[pmc8_command:246]: Command ESGv! failed
18:34:24.178836 indigo_server: INDIGO Bus: message sent 'Retrying connection in 10 seconds...'
18:34:24.179258 indigo_server: 21 ← { "message": "Retrying connection in 10 seconds..." }
18:34:35.291877 indigo_server: indigo_mount_pmc8[pmc8_command:246]: Command ESGv! failed
18:34:35.291982 indigo_server: INDIGO Bus: message sent 'Retrying connection in 10 seconds...'
18:34:35.292533 indigo_server: 21 ← { "message": "Retrying connection in 10 seconds..." }
18:34:46.405294 indigo_server: indigo_mount_pmc8[pmc8_command:246]: Command ESGv! failed
18:34:46.405404 indigo_server: INDIGO Bus: message sent 'Retrying connection in 10 seconds...'
18:34:46.405794 indigo_server: 21 ← { "message": "Retrying connection in 10 seconds..." }
18:34:56.406000 indigo_server: indigo_mount_pmc8[pmc8_open:185, 0xb32b53c0]: Failed to initialize to /dev/ttyUSB0
18:34:56.408734 indigo_server: indigo_mount_pmc8: Disconnected from /dev/ttyUSB0
```

polakovic, 2022-01-03:
Does the port really exist?

polakovic, 2022-01-03:
Hm, sorry, it does, it is connected...

I'll test it on RPi tomorrow...

polakovic, 2022-01-04:
Hm, just recompiled everything on RPi4 with indigosky image and it just works. I tried to connect and disconnect couple of times. Are you sure the cable is OK? I have similar symptoms with a broken cable, it sometimes worked and sometimes didn't :(

[image]

grisu-112, 2022-01-04:
Hi Peter,

Thanks for your test. I reinstalled my RPI today based on the latest Indigo Sky image and compiled the latest source code on it as well.

Unfortunately the result did not change. 

I hope that only the cable is defective and not the PMC-8, although it does surprise me that the communication with both the Serial Monitor of the Arduino IDE (on Ubuntu) and with the programs available at ES (on Windows) worked but did not with other programs like Putty, Microcom or Indigo.

I just ordered other cables and will post my results here (probably not until the weekend).

grisu-112, 2022-01-04:
Just saw that you are using the EXOS-100. This mount has a built-in FTDI (or similar). That means you are using a simple USB cable. The EXOS-2 has a RS-232 interface. Therefore I have to use an adapter cable. So it could possibly be an incompatibility of the chip installed in this adapter cable with some setting used/set by the affected programs. I will report after testing the ordered adapter cables.

aether34, 2022-01-29:
I can confirm that my EXOS-2GT also has this problem. I have tried both the latest version of Indigo Sky (2.0.0.3) and Indigo Control Panel (2.0.0.4). Below is the output from Indigo Control Panel.

```
12:54:59.928 INDIGO Bus: property update
12:54:59.928 'Mount PMC Eight'.'MOUNT_TYPE' SWITCH rw Ok 2.0 0 OneOfMany {
12:54:59.928   'G11' = Off 
12:54:59.928   'TITAN' = Off 
12:54:59.928   'EXOS-2' = On 
12:54:59.928   'iEXOS-100' = Off 
12:54:59.928 }
12:55:10.143 INDIGO Bus: property change request
12:55:10.143 'Mount PMC Eight'.'DEVICE_PORT' TEXT UNDEFINED Idle 2.0 0  {
12:55:10.143   'PORT' = '/dev/cu.usbserial-AM6ESVN9'
12:55:10.143 }
12:55:10.143 INDIGO Bus: Change request - Device 'Mount PMC Eight' token 0x0, Proprerty 'DEVICE_PORT' token 0x0
12:55:10.143 24 ← <newTextVector device='Mount PMC Eight' name='DEVICE_PORT'>

12:55:10.143 24 ← <oneText name='PORT'>/dev/cu.usbserial-AM6ESVN9</oneText>

12:55:10.143 24 ← </newTextVector>

12:55:10.143 INDIGO Bus: property update
12:55:10.143 'Mount PMC Eight'.'DEVICE_PORT' TEXT rw Ok 2.0 0  {
12:55:10.143   'PORT' = '/dev/cu.usbserial-AM6ESVN9'
12:55:10.143 }
12:55:16.454 INDIGO Bus: property change request
12:55:16.454 'Mount PMC Eight'.'CONNECTION' SWITCH UNDEFINED Idle 2.0 0 UNDEFINED {
12:55:16.454   'CONNECTED' = On 
12:55:16.454 }
12:55:16.454 INDIGO Bus: Change request - Device 'Mount PMC Eight' token 0x0, Proprerty 'CONNECTION' token 0x0
12:55:16.454 INDIGO Bus: property update
12:55:16.454 'Mount PMC Eight'.'CONNECTION' SWITCH rw Busy 2.0 0 OneOfMany {
12:55:16.454   'CONNECTED' = On 
12:55:16.454   'DISCONNECTED' = Off 
12:55:16.454 }
12:55:16.528 timer #0 (of 1) used for 0s
12:55:16.528 timer callback: 0x113fd4d60 started
12:55:16.528 indigo_mount_pmc8: Connected to /dev/cu.usbserial-AM6ESVN9
12:55:22.007 indigo_mount_pmc8[pmc8_command:246]: Command ESGv! failed
12:55:22.007 INDIGO Bus: message sent 'Retrying connection in 10 seconds...'
12:55:22.007 Retrying connection in 10 seconds...
12:55:37.562 indigo_mount_pmc8[pmc8_command:246]: Command ESGv! failed
12:55:37.562 INDIGO Bus: message sent 'Retrying connection in 10 seconds...'
12:55:37.562 Retrying connection in 10 seconds...
12:55:53.114 indigo_mount_pmc8[pmc8_command:246]: Command ESGv! failed
12:55:53.114 INDIGO Bus: message sent 'Retrying connection in 10 seconds...'
12:55:53.122 Retrying connection in 10 seconds...
12:56:03.118 indigo_mount_pmc8[pmc8_open:185, 0x70000d6a5000]: Failed to initialize to /dev/cu.usbserial-AM6ESVN9
12:56:03.118 indigo_mount_pmc8: Disconnected from /dev/cu.usbserial-AM6ESVN9
12:56:03.118 INDIGO Bus: property removal
12:56:03.118 'Mount PMC Eight'.'MOUNT_INFO' TEXT ro Ok 2.0 0  {
12:56:03.118 }
12:56:03.154 INDIGO Bus: property removal
12:56:03.154 'Mount PMC Eight'.'GEOGRAPHIC_COORDINATES' NUMBER rw Ok 2.0 0  {
12:56:03.154 }
12:56:03.177 INDIGO Bus: property removal
12:56:03.177 'Mount PMC Eight'.'MOUNT_LST_TIME' NUMBER ro Ok 2.0 0  {
12:56:03.177 }
12:56:03.197 INDIGO Bus: property removal
12:56:03.197 'Mount PMC Eight'.'MOUNT_PARK' SWITCH rw Ok 2.0 0 OneOfMany {
12:56:03.197 }
12:56:03.232 INDIGO Bus: property removal
12:56:03.232 'Mount PMC Eight'.'MOUNT_SLEW_RATE' SWITCH rw Ok 2.0 0 OneOfMany {
12:56:03.232 }
12:56:03.251 INDIGO Bus: property removal
12:56:03.252 'Mount PMC Eight'.'MOUNT_MOTION_DEC' SWITCH rw Ok 2.0 0 AtMostOne {
12:56:03.252 }
12:56:03.270 INDIGO Bus: property removal
12:56:03.270 'Mount PMC Eight'.'MOUNT_MOTION_RA' SWITCH rw Ok 2.0 0 AtMostOne {
12:56:03.270 }
12:56:03.289 INDIGO Bus: property removal
12:56:03.289 'Mount PMC Eight'.'MOUNT_TRACK_RATE' SWITCH rw Ok 2.0 0 OneOfMany {
12:56:03.289 }
12:56:03.308 INDIGO Bus: property removal
12:56:03.308 'Mount PMC Eight'.'MOUNT_TRACKING' SWITCH rw Ok 2.0 0 OneOfMany {
12:56:03.308 }
12:56:03.326 INDIGO Bus: property removal
12:56:03.326 'Mount PMC Eight'.'MOUNT_ON_COORDINATES_SET' SWITCH rw Ok 2.0 0 OneOfMany {
12:56:03.326 }
12:56:03.345 INDIGO Bus: property removal
12:56:03.345 'Mount PMC Eight'.'MOUNT_EQUATORIAL_COORDINATES' NUMBER rw Ok 2.0 0  {
12:56:03.345 }
12:56:03.363 INDIGO Bus: property removal
12:56:03.363 'Mount PMC Eight'.'MOUNT_HORIZONTAL_COORDINATES' NUMBER ro Ok 2.0 0  {
12:56:03.363 }
12:56:03.383 INDIGO Bus: property removal
12:56:03.383 'Mount PMC Eight'.'MOUNT_ABORT_MOTION' SWITCH rw Ok 2.0 0 AtMostOne {
12:56:03.383 }
12:56:03.402 INDIGO Bus: property removal
12:56:03.402 'Mount PMC Eight'.'MOUNT_EPOCH' NUMBER ro Ok 2.0 0  {
12:56:03.402 }
12:56:03.421 INDIGO Bus: property removal
12:56:03.422 'Mount PMC Eight'.'MOUNT_SIDE_OF_PIER' SWITCH ro Ok 2.0 0 OneOfMany {
12:56:03.422 }
12:56:03.444 INDIGO Bus: property removal
12:56:03.444 'Mount PMC Eight'.'SNOOP_DEVICES' TEXT rw Ok 2.0 0  {
12:56:03.444 }
12:56:03.465 INDIGO Bus: property update
12:56:03.466 'Mount PMC Eight'.'CONNECTION' SWITCH rw Alert 2.0 0 OneOfMany {
12:56:03.466   'CONNECTED' = Off 
12:56:03.466   'DISCONNECTED' = On 
12:56:03.466 }
12:56:03.498 timer callback: 0x113fd4d60 finished
12:56:03.499 timer #0 done
```

polakovic, 2022-01-31:
While googling about this issue, I found this comment in INDI driver implementation:

https://github.com/indilib/indi/blob/29056dfe1eb0db45b0e9cfa2ea93dd1155044174/drivers/telescope/pmc8driver.cpp#L318

I'll try to clear DTR on connect, but the unwanted side effect will be the mount reset :(

polakovic, 2022-01-31:
Pls. can you try it with this commit f70cdf9b2c923361ec851cc4f989b2a78683427f?

Make sure you set "Connection mode" to "Serial (clear DTR)" before trying to connect with a control panel. If it will work for you, you can make selection persistent with "Configuration control" > "Save".

aether34, 2022-02-06:
Finally got around to trying this and it seems to work, as far as I can tell. Once it connects, I get a stream of successful "ESG" commands in Indigo Server.

polakovic, 2022-02-06:
👍

closed 2023-11-18 by rumengb