indigo-astronomy/indigo #430: #Frames before Dither

astroman68, 2022-01-04:
Hi Rumen,

Is there a way (maybe in indgo settings) to set the number of frames are taken before a dither occurs? Right now, it appears to dither after every image. I use a Hyperstar and take many, many 15-30 sec frames. Dithering after every frame, plus the settle time, burns through a lot of imaging time. When I used PHD2 I would only dither once every 5-10 frames and generally this worked OK. If there isn't a setting, can you add to enhancement list?

Thanks, 
-Stephen

rumengb, 2022-01-04:
Hi Stephen,
I have transferred the issue to INDIGO as it is related to INDIGO not to Ain Imager.
There is no such functionality yet. I have labeled it as an enhancement and we will add this functionality in a future release.

astroman68, 2022-01-04:
OK, thanks Rumen. Much appreciated.

rumengb, 2022-01-06:
Peter added this to indigo. he was faster then me :)
However I added support for this in Ain (see Skip Frames 0 means dither after each frame 1 means skip one frame then dither etc...) 
It will be available in the next release.
[image]
.

closed 2022-01-06 by rumengb

astroman68, 2022-01-07:
You guys are just awesome!