indigo-astronomy/indigo #432: dsusb.a is 'incompatible'

RusDavies, 2022-01-09:
On a fresh build of Indigo, following the instructions in README.md for Fedora, results in this error: 

```bashgcc -o /home/russell/Development/indigo-astronomy/indigo/build/drivers/indigo_aux_dsusb indigo_aux_dsusb_main.o /home/russell/Development/indigo-astronomy/indigo/build/drivers/indigo_aux_dsusb.a -Lbin_externals/libdsusb/lib/Linux/x86/ -ldsusb -lindigo -lm -lrt -lusb-1.0 -pthread -L/home/russell/Development/indigo-astronomy/indigo/build/lib -Wl,-rpath=\$ORIGIN/../lib,-rpath=\$ORIGIN/../drivers,-rpath=. -lusb-1.0
/usr/bin/ld: skipping incompatible bin_externals/libdsusb/lib/Linux/x86//libdsusb.a when searching for -ldsusb
/usr/bin/ld: cannot find -ldsusb
collect2: error: ld returned 1 exit status
make[2]: *** [../../Makefile.drv:123: /home/russell/Development/indigo-astronomy/indigo/build/drivers/indigo_aux_dsusb] Error 1
make[2]: Leaving directory '/home/russell/Development/indigo-astronomy/indigo/indigo_drivers/aux_dsusb'
make[1]: *** [../Makefile.drvs:66: aux_dsusb/] Error 2
make[1]: Leaving directory '/home/russell/Development/indigo-astronomy/indigo/indigo_drivers'
make: *** [Makefile:148: all] Error 2
```

For reference:
```bash
[...]$ tree indigo_drivers/aux_dsusb/bin_externals/libdsusb/lib/Linux/x86/
indigo_drivers/aux_dsusb/bin_externals/libdsusb/lib/Linux/x86/
└── libdsusb.a
```
And also: 
```bash
[...] $ uname -a
Linux fedora 5.15.12-200.fc35.x86_64 #1 SMP Wed Dec 29 15:03:38 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
```

Since `-Lbin_externals/libdsusb/lib/Linux/x86/` seems correct for my OS and architecture, and given `libdssusb.a` is the only library in that directory, and yet yet that library is apparently 'incompatible', then what am I missing?

RusDavies, 2022-01-09:
Just tried on Ubuntu.  Builds without issues.

RusDavies, 2022-01-09:
And now I double-check myself, I see that I read too fast and that in fact it's picking up the wrong architecture.  Will investigate that.

RusDavies, 2022-01-09:
The issue is with the detection logic for ARCH_DETECTED in line 110 of the Makefile. 

I'm working on a patch.  However, I'm slowed because even having fixed that, other issues are preventing build on Fedora.  Working on it.

rumengb, 2022-01-09:
Actually this is a hack. We added this because need to distinguish x64 vs x32 in docker as we use docker to build it on different architectures. So if you know how can we distinguish those architectures in docker in a better way we will be glad...

RusDavies, 2022-01-09:
Ah, docker.  I'd have to look at that.  

For now, I've created a pull request, that at least allows building for Fedora x86_64.  Given this builds on a hack, perhaps it should be rejected in favour of a proper solution. 

https://github.com/indigo-astronomy/indigo/pull/433

rumengb, 2022-01-10:
fixed with commit f461784e8324fae812981414b74abcd97524be73

rumengb, 2022-01-10:
hey Russell,
Maybe it is a good idea to provide RPM build. Since you are Fedora user, Can you help us with this?

closed 2022-01-10 by rumengb

RusDavies, 2022-01-10:
Funnily enough, getting this packaged and into the Fedora repos was one of the outcomes I'm seeking to schedule. Let's see how far I can progress, with time available.