indigo-astronomy/indigo #438: ALPACA Agent: ImageReady never becomes true

Paolo97Gll, 2022-01-30:
Last night I made some tests with the ALPACA Agent and I found that the property `ImageReady` never becomes `true`, causing download problems.

Standing to the ASCOM standard, the property [`ImageReady`](https://ascom-standards.org/Help/Developer/html/P_ASCOM_DeviceInterface_ICameraV3_ImageReady.htm) should became `true` when an image is ready to be downloaded. If this flag is `false`, the property [`ImageArray`](https://ascom-standards.org/Help/Developer/html/P_ASCOM_DeviceInterface_ICameraV3_ImageArray.htm) will fail with an `InvalidOperationException`.

I tested two ASCOM software (MaxIm DL 6 and Astroart 8) and both of these failed in downloading the image. Here the behaviour I found by looking at the trace log of the ASCOM ALPACA driver.

- MaxIm DL looks at both [`CameraState`](https://ascom-standards.org/Help/Developer/html/P_ASCOM_DeviceInterface_ICameraV3_CameraState.htm) and `ImageReady`. When the exposure start, `CameraState` changes from 0 to 2 and then, at the end, it returns to 0. MaxIm, watching this "toggling", starts the download using `ImageArray`. However, the property `ImageReady` is always false and this trigger the `InvalidOperationException`.

- Astroart instead watch only at `ImageReady`, freezing in an infinite wait (since `ImageReady` never becomes true).

rumengb, 2022-01-30:
Can you provide an indigo trace log of the problem. It may be camera driver
related. I have checked the alpaca code and tested the camera simulator
against ASCOM conformance checker and it reported no issues.

ConformanceCheck ASCOM Device Conformance Checker Version 6.5.7500.22515,
Build time: 7/14/2020 12:30:30 PM
ConformanceCheck Running on: ASCOM Platform 6.5 SP1 6.5.1.3234

ConformanceCheck Driver ProgID: ASCOM.AlpacaDynamic1.Camera

Error handling
Error number for "Not Implemented" is: 80040400
Error number for "Invalid Value 1" is: 80040405
Error number for "Value Not Set 1" is: 80040402
Error number for "Value Not Set 2" is: 80040403
Error messages will not be interpreted to infer state.

05:16:11.742 Driver Access Checks              OK
05:16:13.086 AccessChecks                      OK       Successfully
created driver using late binding
05:16:14.007 AccessChecks                      OK       Successfully
connected using late binding
05:16:14.007 AccessChecks                      INFO     The driver is a COM
object
05:16:15.039 AccessChecks                      INFO     Driver does not
expose interface ICamera
05:16:15.055 AccessChecks                      INFO     Driver does not
expose interface ICameraV2
05:16:15.070 AccessChecks                      OK       Successfully
created driver using driver access toolkit
05:16:15.835 AccessChecks                      OK       Successfully
connected using driver access toolkit

Conform is using ASCOM.DriverAccess.Camera to get a Camera object
05:16:15.929 ConformanceCheck                  OK       Driver instance
created successfully
05:16:16.507 ConformanceCheck                  OK       Connected OK

Common Driver Methods
05:16:16.554 InterfaceVersion                  OK       3
05:16:16.602 Connected                         OK       True
05:16:16.665 Description                       OK       CCD Imager Simulator
05:16:16.728 DriverInfo                        OK       ASCOM Dynamic
Driver v6.5.1.3234 - REMOTE DEVICE: indigo_ccd_simulator
05:16:16.773 DriverVersion                     OK       2.0.0.18
05:16:16.819 Name                              OK       CCD Imager Simulator
05:16:16.867 CommandString                     INFO     Conform cannot test
the CommandString method
05:16:16.867 CommandBlind                      INFO     Conform cannot test
the CommandBlind method
05:16:16.867 CommandBool                       INFO     Conform cannot test
the CommandBool method
05:16:16.867 Action                            INFO     Conform cannot test
the Action method
05:16:16.898 SupportedActions                  OK       Driver returned an
empty action list

Can Properties
05:16:16.992 CanAbortExposure                  OK       True
05:16:16.992 CanAsymmetricBin                  OK       False
05:16:17.008 CanGetCoolerPower                 OK       True
05:16:17.008 CanPulseGuide                     OK       False
05:16:17.022 CanSetCCDTemperature              OK       True
05:16:17.022 CanStopExposure                   OK       False
05:16:17.022 CanFastReadout                    OK       False

Pre-run Checks

Last Tests
05:16:17.163 LastExposureDuration              OK
[... 127 lines cut ...]
UTC
05:16:28.726 ImageArray                        OK       Successfully read
32 bit integer array (1 plane) 1600 x 1200 pixels
05:16:35.757 ImageArrayVariant                 OK       Successfully read
variant array (1 plane) with System.Int32 elements 1600 x 1200 pixels

StartExposure error cases
05:16:35.867 StartExposure                     OK       Exception correctly
generated for negative duration
05:16:36.040 StartExposure                     OK       Exception correctly
generated for X size larger than binned chip size, Bin 1x1
05:16:36.275 StartExposure                     OK       Exception correctly
generated for Y size larger than binned chip size, Bin 1x1
05:16:36.491 StartExposure                     OK       Exception correctly
generated for X start outside binned chip size, Bin 1x1
05:16:36.697 StartExposure                     OK       Exception correctly
generated for Y start outside binned chip size, Bin 1x1

Post-run Checks
05:16:36.991 PostRunCheck                      OK       Camera returned to
initial cooler temperature

Conformance test complete

No errors, warnings or issues found: your driver passes ASCOM validation!!

rumengb, 2022-01-30:
And here is the indigo log:

15:20:16.225594 indigo_server:
indigo_agent_alpaca[alpaca_v1_api_handler:272]: < GET
/api/v1/camera/5/imageready ClientTransactionID=166&ClientID=11694
15:20:16.225700 indigo_server:
indigo_agent_alpaca[alpaca_v1_api_handler:322]: > { "Value": false,
"ErrorNumber": 0, "ErrorMessage": "", "ClientTransactionID": 166,
"ServerTransactionID": 445 }
15:20:16.225797 indigo_server: indigo_agent_alpaca[send_json_response:186]:
/api/v1/camera -> 200 OK
15:20:16.231636 indigo_server:
indigo_agent_alpaca[alpaca_v1_api_handler:272]: < GET
/api/v1/camera/5/camerastate ClientTransactionID=167&ClientID=11694
15:20:16.231715 indigo_server:
indigo_agent_alpaca[alpaca_v1_api_handler:322]: > { "Value": 2,
"ErrorNumber": 0, "ErrorMessage": "", "ClientTransactionID": 167,
"ServerTransactionID": 446 }
......
15:20:18.576115 indigo_server:
indigo_agent_alpaca[alpaca_v1_api_handler:272]: < GET
/api/v1/camera/5/imageready ClientTransactionID=225&ClientID=11694
15:20:18.576193 indigo_server:
indigo_agent_alpaca[alpaca_v1_api_handler:322]: > { "Value": true,
"ErrorNumber": 0, "ErrorMessage": "", "ClientTransactionID": 225,
"ServerTransactionID": 504 }
15:20:18.576276 indigo_server: indigo_agent_alpaca[send_json_response:186]:
/api/v1/camera -> 200 OK
15:20:18.581581 indigo_server:
indigo_agent_alpaca[alpaca_v1_api_handler:272]: < GET
/api/v1/camera/5/camerastate ClientTransactionID=226&ClientID=11694
15:20:18.581651 indigo_server:
indigo_agent_alpaca[alpaca_v1_api_handler:322]: > { "Value": 0,
"ErrorNumber": 0, "ErrorMessage": "", "ClientTransactionID": 226,
"ServerTransactionID": 505 }

so please let us know what camera driver you are using and provide the full
indigo trace log.

rumengb, 2022-01-30:
I have Atik, Starlight Express, ZWO ASI, meade DSI and QHY cameras, tested all of them... no issue whatsoever with the ASCOM conformance checker.

rumengb, 2022-01-30:
Installed Astroart 8 Demo and tested with several cameras. Here is a screenshot:
[image]
 So please send us the full indigo trace log! it might be something driver specific....

Paolo97Gll, 2022-01-30:
Sorry I was busy (I’m also a student). I made some tests in the past hours but there’s something strange, the behavior is sightly different from the one exposed above, despite the setup is the same. I will investigate and share with you all the logs and informations in the next hours or days.

Unfortunately I didn’t saved the indigo trace log yesterday, my fault.

rumengb, 2022-01-31:
Can you please share the camera and the driver involved?

rumengb, 2022-01-31:
Checked MaximDL seems working together with the mount focuser and everything. However I could not find how to connect the filter wheel. I never used maximDL.
[image]

Paolo97Gll, 2022-02-01:
> Can you please share the camera and the driver involved?

The camera is an ATIK 16200, so the driver is the `indigo_ccd_atik`

Paolo97Gll, 2022-02-01:
Ok, here the results.

## Informations

### Camera

- Atik 16200 mono

### Server

- Raspberry Pi 4 4GB with Ubuntu 20.04.3 LTS (kernel 5.4.0-1050-raspi)
- INDIGO 2.0-166 with only `indigo_ccd_atik` and `indigo_agent_alpaca` loaded

### Client

- Lenovo X1 Carbon 6th with Windows 11
- ASCOM 6.6 with Astroart 7

## Testing procedure

I loaded only the needed drivers, started the server right before starting the test and stopped it at the end of the test, so all the log informations in the log files are generated during the test (about two minutes).

I started the server with the following command:

```
indigo_server --enable-trace indigo_ccd_atik indigo_agent_alpaca > indigo.log 2>&1
```

The ASCOM Alpaca driver on the client is configured as following:

[image]

I also tried all the other "Image array transfer method" (and following what written in the [alpaca agent docs](https://github.com/indigo-astronomy/indigo/tree/master/indigo_drivers/agent_alpaca#ccd)), with the same results.

After the camera connection, I took a 1s exposure photo. After about 105 seconds (Astroart showed "Dowload ..." for all this time), it returned a completely dark image (see attachments below). I took a photo with the same exposure with Ain INDIGO Imager two minutes later and it showed the expected result (see attachments below). Then I disconnected the camera and shutted down the server.

## Attachments 

Here the file: [attachments.zip](https://github.com/indigo-astronomy/indigo/files/7982365/attachments.zip). I had to compress all because the `indigo.log` file is about 60MB. The ZIP file contains:

- `indigo.log` with the INDIGO trace log
- `ASCOM.AlpacaDynamic1.Camera.2316.227420.txt` with the ASCOM trace log
- `photo_astroart.fit` is the photo obtained using Astroart 7 and `photo_ain.fits` the same photo using Ain INDIGO Imager. _The photos do not show an astronomical object because the camera is not on the telescope at the moment._

polakovic, 2022-02-02:
As far as I can say there is no issue with the driver or ALPACA agent. Both CCD_IMAGE and CCD_EXPOSURE were set correctly and the state was propagated to the client. ASCOM protocol communication failed during image upload, server reports "Broken pipe" at 22:18:19.986346. It looks ALPACA client didn't asked for compression, so the server sent 1.8 millions integers as plain JSON :)

rumengb, 2022-02-02:
Can you please try the image transfer settings "JSON" and "GZip"
[image]
and maybe connect your rpi to a more reliable network. try LAN cable?

rumengb, 2022-02-02:
I managed to reproduce the issue with one of my larger cameras. And figured out that it works only with Json + GZip or Base64HandOff. What comes to BestAvaiable for some strange reason ASCOM Remote considers plain JSON as the best method, which transfers the image as human readable plain text numbers which on a slow Wifi may take forever. On a 100Mbit network Alpaca (with Astroart) in its fastest setting "Base64HandOff" took about 18 seconds to download an image from ASI071MC camera and INDIGO (with Ain Imager) takes about 4s with same windows and same indigo server and same camera. Everything boils down to the way the images are transferred. Unfortunately there is nothing we can do about this, Alpaca is slow by design :(

However I have updated the Readme 7a9c4841

Paolo97Gll, 2022-02-02:
> It looks ALPACA client didn't asked for compression, so the server sent 1.8 millions integers as plain JSON :)

I don't know why but it sounds funny.

> connect your rpi to a faster network. try LAN cable?

My RPi is already connected to a 1Gbps network using an ethernet cable. However it's located in one of the domes at the observatory and I'm here only in the weekends. The first time I reported this issue this sunday, I was at the observatory and I did every test using a cabled PC. However, yesterday I did the test behind a VPN and my internet connection is not the best in the world (20/30Mbps down, 2/3Mbps up), but the connection of the observatory is better (50Mbps down/up).

> I managed to reproduce the issue with one of my larger cameras. And figured out that it works only with Json + GZip or Base64HandOff. What comes to BestAvaiable for some strange reason ASCOM Remote [...]

For some reasons now, using JSON+GZip and Base64HandOff, it worked. This sunday no, I don't know why since I tried also these settings. Here the download times:

- Ain INDIGO Imager: 7 seconds
- JSON+GZip: 15 seconds
- Base64HandOff: 3 minutes

Being at home and not at the observatory, I find them good results (although the fastest method takes twice as long as native INDIGO).  The ASCOM support in our system is only an extra feature and we want to base everything on INDIGO, so I don't think it's a problem now that is known.

Since it seems an ASCOM related issue, I think we can close this issue. Thank you for the support!

P.S. Should I rename the issue now that the problem is known?

closed 2022-02-02 by Paolo97Gll

Paolo97Gll, 2022-02-02:
I'm going to test the ASCOM Alpaca bridge a lot in the next weeks/months, I will keep you informed of any other related issues or news.

rumengb, 2022-02-02:
> P.S. Should I rename the issue now that the problem is known?
No it is fine :)
Keep us informed :)