indigo-astronomy/indigo #441: indigo_ccd_qhy2 crashes the server if launced with the -i option

Paolo97Gll, 2022-02-12:
Following the documentation, I launched the `indigo_ccd_qhy2` driver using the `-i` option.

https://github.com/indigo-astronomy/indigo/blob/1b3d02bc82b877853d2093d0ec2ce3f41e18baf8/indigo_drivers/ccd_qhy2/README.md?plain=1#L34-L41

I used the following command:

```
indigo_server --enable-info --disable-bonjour indigo_ccd_asi indigo_ccd_atik -i indigo_ccd_qhy2 indigo_wheel_sx indigo_agent_alpaca
```

And then the server crashed when I tried to take a 0.1 second shot (it happens also with 0.001, 0.01 and 1). Here the log file: [indigo.log](https://github.com/indigo-astronomy/indigo/files/8054155/indigo.log).

Without the `-i` option, everything works well and no crash happen.

polakovic, 2022-02-12:
We'll check it, but for now try to avoid -i switch. For such a fast camera it may slow down everything by many orders of magnitude (because of very inefficient INDI legacy protocol enforced by -i switch).

@rumengb, it looks like a memory overrun in base64 decoding.

rumengb, 2022-02-13:
> Ok, thanks!

@Peter Polakovic ***@***.***>
I have checked the problem... it does not crash here but there is another
strange issue. Image retrieval hangs until timeout occurs...

08:16:33.070885 indigo_server: Worker thread started socket = 30
08:16:33.070973 indigo_server: 4 clients
08:16:33.112826 indigo_server: GET /blob/0x2fa39f10.fits -> OK (16240320
bytes)
08:16:34.114445 indigo_server: 3 clients
08:16:34.117261 indigo_server: Worker thread finished
08:17:04.917428 indigo_server: indigo_write(24): Broken pipe
08:17:04.928936 indigo_server: Worker thread started socket = 25

I was able to reproduce it with asi too...