indigo-astronomy/indigo #446: Indigo Control Panel, Imager... Not populating

JaySPA, 2022-04-16:
All,

New here.  Running 64 bit Bullseye on RPI 4 8GB. INDI fully loaded.. PHD2, EKOS, SkyX, etc. all running perfectly. I followed the download and build instruction for the Indigo Control Panel, Imager, Viewer, and the applications all load, but all the drop downs are blank and nothing is selectable even for Simulation.  Tried with a USB 3 connected ZWO 120 Mono and was not picked up as a device, whereas was seen by all the other packages..

I did the build multiple ways based on the following page (tried all three with the same result) ..

https://www.indigo-astronomy.org/downloads.html

Someone mentioned Indigo Server may be in conflict for a port with INDI server?

Appreciate any thoughts or leads..

Jay S.

rumengb, 2022-04-16:
There are several problems with this:
1. When you install indigo it will not be started by automatically. you will have to start it as a service with:
```
systemctl start indigo
```
or to enable it to be loaded at boot:
```
systemctl enable indigo
```
for this approach you will need to create user "indigo" member of the dialout group.

or you can start indigo_server by hand (but still the current user needs to be a member of dialout group too).
```
indigo_server
```
2. INDIGO will try to load on the same port as indi. so you need to start indigo on a different port please read:
https://github.com/indigo-astronomy/indigo/blob/master/indigo_docs/INDIGO_SERVER_AND_DRIVERS_GUIDE.md

3. do not try to control the same devices from indigo and indi at the same time. They will not work!
4. Ain Imager uses Indigo agents - a concept that INDI does not have. So it will not work with indi.
5. INDIGO has a service discovery - again a concept that is not present in indi. So Ain and indigo control panel will not see indi services. They need to be configured by hand as with native indi clients.

JaySPA, 2022-04-17:
> There are several problems with this:
> 
> 1. When you install indigo it will not be started by automatically. you will have to start it as a service with:
> 
> ```
> systemctl start indigo
> ```
> 
> or to enable it to be loaded at boot:
> 
> ```
> systemctl enable indigo
> ```

> for this approach you will need to create user "indigo" member of the dialout group.
>
> or you can start indigo_server by hand (but still the current user needs to be a member of dialout group too). [...]

Rumen..  Not following all the steps you're outlining..  I can start the service witg Systemctl easy enough.. 

On the next piece you talk about creating an INDIGO  member, but trying to understand why this would be necessary? RPIs are becoming more and more popular as a platform for Astronomy.  There's the potential for INDIGO packages to be broadly accepted like the CloudMaker packages are on OSX, but I think the install would need to be more "seamless" and less "build" oriented.  I know this is still true for other packages (some still won't work on 64 Bit RPI OS).

On the ports, TCP or UDP? If INDI is using a hardcoded one in INDIGO can an install query for an open port or is 7624 something specific.

Am reviewing the following page by you about INDIGO agents, but it's not clear how (or why) they would have to be manually started.  If "Agents" are the selections for product families, e.g. ZWO Cameras, Pegasus products, Syscan mounts, etc. I'm not getting a place where I can select those from a drop down of sorts, like in what I see on OSX.

https://github.com/indigo-astronomy/indigo/blob/master/indigo_docs/INDIGO_AGENTS.md

Again I really like the products on Cloudmakers and my initial route on this was to ask Peter about how this is being enabled on RPIs..  Again, just suggesting that the install process be developed more to be less "you have to do all the individual steps" and more toward a developed package.  I do recognize that's not the norm for a lot of Debian RPI packages yet..

Thanks for the help..

Jay S.

polakovic, 2022-04-17:
Jay, agents are not selections for product families, agents are an application logic running on the server side. So applications like Ain or AstroImager are just a front end to the agents. You can enable particular agent with any control panel on Server > Drivers.

Both INDI and INDIGO listens on TCP port and both listens on the port 7624 by default. You can change it by adding a command line switch (in this particular case by configuring the service) on either INDI or INDIGO side, but it is better to do it on INDIGO side as far as its clients discover the port automatically.

JaySPA, 2022-04-17:
Thanks Peter..  I see where I can add the "port" switch on the command line and will try to start the service and let you know how it goes....  Again, just thinking about bigger picture offerings like Astroimager and the family, but on the Raspberry Pi at some point.  If I can run full SkyX there (64 bit) just want to be able to show there are different ways to do the job (so to speak).  :-)

rumengb, 2022-04-17:
Have you seen INDIGO Sky?
https://www.indigo-astronomy.org/indigo-sky.html
it is a configured and optimized Linux distribution to run INDIGO. you just download the image frush it to the SD card and run it on a raspberry pi. You do not have to start anything... It boots and starts indigo service.

To install it as a service is a bit more complicated on linux. We do provide service file but linux itself needs some permissions granted to the users to access serial devices like mounts focusers etc... This permission is granted to the users in group "dialout" so you need to take care of this yourself. We have no idea which users are going to use indigo. It is not a good idea to run any service as root. So our service file requires user indigo in group dialout. So what we can do is to create this user and add it to the group on installation. But i do not think it is a good idea to run this service by default.  You will still need to run systemctl tool to start enable disable and stop indigo. There are some guy tools readily available to sart and stop services. I think I can do that: I can add silently indigo user in dialout group. I will do this in the the next release.

JaySPA, 2022-04-17:
Rumen... 

Thanks.  I did see INDIGO SKY and will give that a try as well.  I do think people will want to be able to run PHD2 (for example) as their guiding package and it appears (from some testing I just did) that the same port can be used for INDI and INDIGO as long as you're not trying to use a device on both at the same time.  So for an example, use PHD2 to guide with a ZWO camera, but AIN Imager to image with a different ZWO (or other) camera.

I mentioned to Peter, that after I created an IDAC file of drivers I wanted have started, the file was not found automatically, and also not found unless I changed the default on "LOAD" to "ALL FILES" .. Leaving it as looking for the IDAC extension didn't work..

Again, covered a lot of ground and will experiment more.  One reason I haven't looked more fully at the INDIGO SKY is that I don't boot from an SD card.  My RPIs are in an ARGON m.2 case, booting from a 500GB m.2 SSD.. SUPER fast!  I can load SkyX fully in 2 to 3 seconds (as an example).  Since it's not an SD card, it's not so easy to just boot a different image..

Jay

rumengb, 2022-04-17:
I do not know what is "IDAC file of drivers". IDAC means Indigo Device Access Control and there is no  "LOAD" to "ALL FILES" in yje indigo access control. Why do you need device access control? How do you expect this file to be found? I think you are trying to do too many things at once without understanding of the concepts. I would suggest to do the things one by one not everything at once. Indigo access control described here: https://github.com/indigo-astronomy/indigo/blob/master/indigo_docs/INDIGO_DEVICE_ACCESS_CONTROL_AND_LOCKING.md  ASCOM, INDI and other platforms do not offer such functionality and it is developed upon a request from professional observatory. They wanted to be able to lock devices used by somebody in order to avoid accidental takeover of control by other colleague and mess with the process. So I doubt you need it. 

Also Indigo sky can run form an SSD drive and it is explained on the indigosky page. There is a link to a document describing how to do this: https://github.com/indigo-astronomy/indigosky/blob/master/INSTALL.md

JaySPA, 2022-04-17:
Rumen..  Again appreciate the help.  I am not a master of Linux by any means, but I'm also not a complete novice..   :-)

As for IDAC, it "appeared" that the only way I could get the Control Panel to remember which Device Drivers I had selected to be loaded was by creating and saving the Device ACL file between shutdowns or reboots. That no longer seems to be the case, but until I had created it and loaded it, none of the selected devices were being carried across reboots.  Again, not seemingly happening now.  Thanks for the link and the function of ACL seems thought out for the purpose you're saying.  I think I'm trying things in a logical order.. I just could not get any of the selected drivers to "reload"..  Not sure why but will try again on a fresh installation.

On Indigo Sky.. I'm aware that I can load it to the SSD to boot from, but that image would wipe out the working system I've got there with SkyX, INDI and some of those applications..  For now PHD2 is still the preferred guide program.

A couple of other things I've run into (and could be something I'm doing wrong).  The instructions you provided using the Systemctl Indigo are not working.  See the attached file with the simple error messages about the service not being found.

Back to the port for a minute. There is another place where the 7624 is referenced it looks like based on the -h display in terminal.  There is a remote server host switch that can be set as well that looks like it also defaults to 7624.  Do I need to point that to a different port as well?

-r  | --remote-server host[:port]     (default port: 7624)

Lastly (as I keep going deeper), when I do start indigo_server manually, I cannot either shutdown or reboot unless I go to the task manager and kill the process before wanting to shutdown.  I've tried this one repeatedly, and this seems consistent.  If I never start Indigo_server, the system operates totally normally.. No issues shutting down or rebooting either as a sudo command or with the GUI. Is it necessary to shutdown the Server process before shutting down?  Does this go back to your earlier comments on dialout and systemctl to start/stop indigo process?  Again (for me) Systemctl does not seem to work.

[INDIGO.TXT](https://github.com/indigo-astronomy/indigo/files/8502443/INDIGO.TXT)

rumengb, 2022-04-17:
IDAC has nothing to do with the drivers. And control panel only instructs the server which drivers to load and unload. the server takes care of loading and unloading. Next time when you start the server it will load the drivers that were loaded last time. However this list will be overridden if you provide the drivers at startup on the command line.

About systemctl you are right, indigo.service is not provided in the deb. I will fix that along with creating indigo user.

to run indigo server on a specified port you need to  run it with "-p XXXX" or "--port XXXX" it is listed with the "indigo_server -h" and described in the document I have shown you https://github.com/indigo-astronomy/indigo/blob/master/indigo_docs/INDIGO_SERVER_AND_DRIVERS_GUIDE.md
it says
```
-p | --port
Set listening port for the indigo_server.
```

What comes to the reboot while indigo is running it may take longer but it will eventually reboot.

JaySPA, 2022-04-18:
Hi Rumen...

Peter sent me an email with the indigo.service file to create, but since I'm not creating an indigo "user" per se, I didn't know whether or not to just change that indigo.service text file to reflect the user name I'm using (pi in this case).

On the port, I'm using that switch to specify 7625 (instead of the default 7624).. My question was based on further down in the help section where 7624 is again referred to as a part of the remote server??
 
"-r | --remote-server host[:port] (default port: 7624)"

I waited on the shutdown/reboot..  Took about 90 seconds or so..  Not sure why it takes that time, but it did shutdown.

As I mentioned to Peter, I would like to include this in a demo, perhaps at AIC in San Jose mid May.. Hoping to get it more "end user" friendly as far as install, etc..  Again, I know there is Indigo Sky as well, but I think more users than not might want to have more control over apps.  Sounds like Indigo Sky is a similar thought process to the Astroberry offering as far as it being a bootable solution..

Thanks again..

Jay S.

rumengb, 2022-04-18:
About the system file:
changing the user to pi and the log path will be enough:
```
[Unit]
Description=INDIGO Server

[Service]
User=**indigo**
PAMName=login
ExecStart=/usr/bin/indigo_server
StandardOutput=file:/home/**indigo**/indigo.log
StandardError=file:/home/**indigo**/indigo.log

[Install]
WantedBy=multi-user.target

```
change the bold parts to whatever user you want as long as it exists :)

```
-r | --remote-server host[:port] (default port: 7624)
```
This is when you want one server to proxy some other server.  Or for example if you have 5 RPIs one to control your mount,  one for imaging camera one for guiding etc. and you want all of them to communicate with each other for example the imager RPI needs to dither the guiding RPI. You need to make them talk to each other. 

if you want instantaneous reboot/shutdown you either need to run it as a service or to run indigo_server without its auto recovery finction 
```
-- | --do-not-fork

By default indigo_server will start a child process called indigo_worker, and the drivers will be run in the boundaries of this process. This is done for a reason, if a single driver crashes this will crash indigo_worker process indigo server will detect it and start indigo_worker again. This switch prevents this behavior and if a driver fails the whole server will crash without an attempt to recover. This is is used for faulty driver debugging.
```

rumengb, 2022-04-19:
> Does seem a little like Chromium a little slower to load, but not sure.. didn't do any before/after performance numbers..
Lol this can not be true. INDIGO when idle does not take any CPU cycles and 0.7% of the 4G memory of my RPI.

Yes you can add "-p 7625" or whatever you need.

JaySPA, 2022-04-21:
Hi Rumen...

Peter mentioned you have a new(er) version ready and available.  Will what I have installed update (now that it's a service) or do I need to uninstall and reinstall the newer one?  Should I do that anyway just to test the install process for you?  What's the best way to make sure the older version is totally gone?  Thanks..  Found a typo on one of the panels.. will post it up.. just a mispelling..  :-)

Jay

rumengb, 2022-04-21:
> Rumen..
>
> Thanks! Will try today.. Should I erase the "indigo-service" service I
> created? Can I specify a different port with the new one (assuming there's
> a equivalent/simllar file to edit where I placed the indigo-service file).
>
> Think the typo was in AIN imager.. Will get that to you..
> Jay

I think you should keep it because indigo-derver service will most likely
be overwritten by the next update.

rumengb, 2022-04-21:
Thanks typo fixed!

About the update do you have indigo ppa added? if not please add it as described here:
https://www.indigo-astronomy.org/downloads.html

if you already have it try 
```
$ indigo_server -h
INDIGO server v.2.0-176 built on Apr 19 2022 23:43:35.
usage: indigo_server [-h | --help]
       indigo_server [options] indigo_driver_name indigo_driver_name ...
```
check the version if it says 2.0-176 it is the recent version.

if it is not then try removing the cache:
```
rm /var/lib/apt/lists/indigo-astronomy.github.io_indigo%5fppa_ppa_dists_indigo_*
```
and run update and upgrade again

JaySPA, 2022-04-21:
Rumen,

176 running..  Got an error message at the very end (sorry, forgot to copy) about a patch not being applied since this wasn't an Indigo Sky installation.  The RM command on the cache cleared the way for the update to work.

Indigo_server also apparently running, but when I launch the Indigo control panel, I believe it is still tied to the Indigo_Service I built because I had added the different port in that startup line  command "ExecStart=/usr/bin/indigo_server -p 7625"..

Jay

rumengb, 2022-04-21:
This is not an error message, This means that the underlying operating system is not INDIGO Sky image and does not have the optimizations and hooks needed to install the tools to control the OS from indigo, like reboots, system updates, WiFi configuration, internet sharing etc...

About the running indigo_server - how am i supposed to know? :) But what i can assure you indigo-server service will not start on its own by default. And indigo_control_panel is not tied to anything if the server is running and auto discovery is not disabled it will see it. You are doing something wrong but I have no idea what you are doing at all.
It would help a lot if you provide more specific information about what you are trying to achieve and how you are trying to achieve it step by step.

Man, one can think INDIGO is totally broken and does not work at all while reading this thread. Thousands of users are using indigo and nobody had so many problems :)

JaySPA, 2022-04-22:
Hi Rumen...

Oh the years and yearsof being a coder come rushing back after reading your reply..   LOL   I've been trying to give you positive feedback about ways to make the Indigo offerings more commercial on a RPI, certainly not tear them down.. I'm sure you're aware just how much of the industry ZWO owns and that the ASIAir offering run on a Raspberry Pi as well.  I like Indigo a lot and use it extensively on OSX as Peter knows..  64 Bit Pi OS is relatively new.  We're just beginning to see more and more acceptance of an RPI as a viable alternative to the bigger box offerings of Fusion, Eagle, etc..  Please understand there's no criticisms here and don't think anything I said was critical of you as a programmer or the offerings..

Back to what I was asking..  I don't understand your question .. How are you supposed to know what aspect ?  (appreciate the smiley face)  :-)   After I built the original Indigo_Server Service (under the name of Indigo_Service and updating the user name and adding the flag to change the port) it appears that I no longer needed to do a command line start.  After doing that, the control panel comes up, reflecting the port selection (7625) and all the devices I selected.  If I launched the Control Panel without first starting Indigo_Server (or having it now as a service), the fields were blank on the Control Panel.  I was under the impression from what you had instructed me earlier on was making the Indigo_Service file in lib/systemd/system was the way to make it start on boot. 

Again, maybe I misunderstood, but I think with the new code, there is now a Indigo_Server file in that same location with similar text (which I also think means both are starting). What I was saying was that the language I inserted in the Indigo_Service file has the port change and the Pi as the user name.  I made no mods to the Indigo_Server file in the same location.. Since the Control Panel comes up reflecting the 7625 port, I have to assume it is being launched at boot by the file I originally created and not the new one created by the new code..  Again, there seems to be no way that the control panel would pick up the port change without referring to that original file.

I don't think anyone is saying Indigo is broken... Again I love it on OSX (really I do!!) and again I'm really trying to make it so that an end user can pick it up and go with it.. Indigo Sky, while building a solution, doesn't have the hooks (I don't think) to run PHD2 if someone wanted to (and most may want to), as my understanding is (at least on an RPI) it requires all the INDI hooks and libraries. I run INDI and INDILIB side by side on a Mac.. Maybe that's not 100% yet on Linux RPI, but I'm trying to help get it there as much as possible.. AND have it on a demo in May at AIC for hopefully lots of people to see!!!  :-)

Hope we're all good!
Jay

rumengb, 2022-04-22:
Hi Jay, no bad feelings :) Everything is ok :)

1. About starting indigo_server:
On Linux the standard way to manage services is through systend. And to do so the systemctl command is used. The *.service files are instructions to systemd how to start/stop and in what environment to run the corresponding service. The presence of this file in a corresponding location means that systemd will know how to manage the service. You can have multiple service files for one service running the service with different configurations and environments. Now you have 2, indigo listening on port 6725 and the one distributed with the deb package running indigo on the default port. From now on you can use them to start stop indigo with different configs.

To start service xxxx (it will not survive a reboot):
```
systemctl start xxxx 
```

To make xxxx service to start at boot:
```
systemctl enable xxxx
```

To stop xxxx (it will start on boot if enabled):
```
systemctl stop xxxx
``` 

to make xxxx service not start at rboot:
```
systemctl disable xxxx
```

to check the status of xxxx service:
```
systemctl status xxxx
```

here is how it looks on indigosky:
[image]

2. control panel comes up unpopulated.
you can run both services as long as they use different ports to listen but there is a catch - you will need to change the bonjour name too in order make the second service auto discoverable. There can not be two services with the same name and by default indigo is using the hostname. To make the two serviced autodiscoverable you need to change the name of one of them. To do so you need to use -b switch
```
-b  | --bonjour name                  (default: hostname)
```
Also Indigo_control_panel will be unpopulated if there is no indigo service discoverable. indigo server announces what devices and what properties are available and if nothing is available it comes up unpopulated. 

In mac Peter decided it another way: each application runs its own server and this is why it is always populated.
so if you do not see it populated he service is either not running, it is not advertising itself or you deliberately ignored this service from control panel (you can do so from "File -> Manage services"
[image]
here you can even connect to unannounced services like INDI by entering the host name and port.

3. PHD2
PHD2 is bound to INDI library and it needs INDI installed to run. We do not install indi on INDIGOSky. So you need to install indi too in order to make it work on inigosky. 
Please remember PHD2 uses INDI to communicate with INDIGO. INDI image transfer by design is in orders of magnitude slower than the native INDIGO. Also INDIGO has an excellent guiding. We have several users comparing HD2 and indigo guiding and came up with the conclusion that the 2 performed equally well but  indigo is better in some cases and:
a) We have a full frame guiding called Donuts which can guide i poor seeing and on highly unfocused stars. However Elon Musk is giving us troubles with this guiding as satellites can mess with the guiding... but I am working on this to ignore them.
b) Multi star guiding - it is more advanced than the one used in PHD2 for several reasons which I am not going to discuss here
c) we have built in Periodic Error Correction without the need for training. You just need to configure 2 parameters.

I am not sure you know but for more than 10 years I was working on lin_guider - an autoguider for linux which by the was was copy-pasted in Ekos and this is how Ekos got his guiding. So I knew a thing or two about guiding when we started indigo.

So if you want to see how really indigo should be used I would not recommend using PHD2 :)

JaySPA, 2022-04-22:
Rumen..

Thanks.. This helps a lot in understanding what's happening and, if needed, how to explain it.  People want turn key these days, which on one hand is good, but on the other hand means they really don't understand what's happening when it breaks.  I try to be both as much as possible.  :-).   Unhooking PHD2 from people's minds will be hard, but promise I will try it here (and I like what Elon does with rockets and capsules, not satellites!!) ..  Then there's the built in guiding in SkyX as well!!  :-).  Will also build a complete Indigo Sky system on another microSD card..  Like I said, trying to get as much built into these great little boxes as possible to be able to demo..  Let me know if there are any "product literature" files I can start linking to as well..

Jay

rumengb, 2022-04-22:
Yes this is what people want turn key, and this is exactly what INDIGO Sky is :) Power up and plug your devices and off you go...
I understand the PHD2 obsession. People would not switch easily. But the truth is that they will never switch if they do not know about the other options.

JaySPA, 2022-04-24:
Rumen..  What is the initial login id and password for Indigosky?  I'm trying indigosky for both.. Not talking about SSID, etc..  Getting the detailed login instructions rolling on screen and then asks for logon ID and password??  Thanks!

polakovic, 2022-04-24:
wifi password is indigosky, user login and password is indigo :)

https://www.indigo-astronomy.org/indigo-sky.html

JaySPA, 2022-04-24:
Yep.. found it..
:-)

svwbeckwith, 2022-11-07:
Rumen,

I have been working with Peter to try and get my RPi working with Indigo, so far with no success. He has been trying to help me, but my C program using Indigo libraries runs fine on his RPi computer and not on mine. I have a code that works well on a Mac, and I need to get it working on the RPi. Peter has been more than generous with his time. It seems clear that he, too, is baffled by my problem described in the next paragraph.

Having followed all the instructions on the Downloads page successfully to install Indigo, AIN Imager, and AIN Control Panel, I find that when I bring up AIN Imager, for example, it does not see any camera plugged into the USB port. When I run "lsusb" from the command line, it sees the camera (SVBony 305m) just fine. When I try to discover the camera with my own C-code using the SVBCameraSDK library that works well on my Mac, it does not see the camera and in fact does not even print out any internal (to the library) messages as it normally does on my Mac when the camera is not plugged in. I have tried reinstalling all the Indigo files, and that makes no difference. 

So there is something subtle going on here that a user like myself is unable to diagnose. I have been writing code in various languages since 1970 (PLC, Fortran, FORTH, 8088/8087 assembler, more recently Python and now C), so I am not a complete novice. It does appear, however, that JaySPA's comments are on the mark for me. I don't have any diagnostic tools to understand what is happening to stop the RPi from working with the Indigo apps or drivers. 

I don't fully understand the answers you have given JaySPA above. It does appear, however, that there are switches or settings in the RPi operating system or environment that do not make themselves evident but, nevertheless, prevent the Indigo system from operating normally. Any insights you might have will be most welcome.

Steve Beckwith, UC Berkeley

closed 2023-11-18 by rumengb