indigo-astronomy/indigo #457: Mount LX200 does not detect TCP network disconnection

je-lamiaud, 2022-08-16:
Following on my previous submission about failed network connection : when the mount is now connected and a disconnection occurs, this disconnection is not handled. The commands sent to the mount end in failure, but it is not obvious to the layperson what causes these failed states on several LX200 Indigo properties.

The following patch detects the TCP disconnection when a command is attempted. Which is quite fast for the mount itself, but can be much longer if, for instance, only the focuser is opened.

[indigo_mount_lx200.c.patch.txt](https://github.com/indigo-astronomy/indigo/files/9352567/indigo_mount_lx200.c.patch.txt)

BTW, I disabled the Nagle algorithm on the LX200 connection, to avoid potential latency issues (LX200 commands being delayed on Indigo side).

polakovic, 2022-08-18:
Thank you! Pls. can you make a pull request for your change? I'm away and can't apply your patch now :(

je-lamiaud, 2022-08-19:
> Thank you! Pls. can you make a pull request for your change? I'm away and can't apply your patch now :(

OK. I'm currently not completely happy with this fix, and will work further on it before creating the pull request.

PR #460 (2022-08-20): Handling of LX200 TCP disconnections (Issue #457)
The disconnection is only detected when trying to send a LX200 command. This is quite fast with the mount driver, which constantly polls the server, but depends on current activity for the guider and focuser. Anyway, opening the guider or the focuser alone must be a very uncommon use case.

Upon disconnection, the drivers are cleanly stopped as if it were a disconnection on user's request ; but, as it was not, the connection property is put in alert state.

This change was tested on macOS x86_64 and Linux ARM.

Fixes #457
closed 2022-08-20 by rumengb