indigo-astronomy/indigo #458: INDIGO A1 ASTAP solver error reading image

DarthMurse, 2022-08-19:
I'm using INDIGO 4.5 on Macbook Pro M1 2020, and I use Canon EOS 7D as my main camera. The rest functions of the software is just fine, but when I use the ASTAP agent to solve the image there's an error message "ASTAP agent: Error reading image file. Since I do not encounter this problem when using my QHY 5-II-L-M camera, I guess this has something to do with the image directory, but I don't know what's wrong. And this has been very frustrating.

Also I don't see the Astrometry agent in the plate solving panel. I thought that ASTAP is kind of deprecated, but I don't know why it's still in use, and the newer Astrometry agent is nowhere to be found.

polakovic, 2022-08-19:
The most probable reason is that you used RAW format on Canon. This issue is fixed in 4.6 pre-release, you can download the most recent build always here:

https://www.cloudmakers.eu/download/

or here:

https://bb.cloudmakers.eu/viewforum.php?f=15&sid=8f5060f7647693a53339703f77225788

BTW, for DSLRs is intended INDIGO D1, not A1.

Astrometry agent is always present on any recent INDIGO, it become active once Astrometry application is detected on the host. You can download it from here:

https://www.cloudmakers.eu/astrometry/

DarthMurse, 2022-08-20:
Thanks, I fixed this problem using the Astronomy local web server. Hope INDIGO get better and better.

closed 2023-11-18 by rumengb