indigo-astronomy/indigo #463: Client force server setting to update image to the client

Ghir8, 2022-09-10:
Hi,
we are using INDIGO to command a dome remotely, it would be nice if the images could be saved into the (in loco) server for backup and to avoid loosing them by some internet hiccups.
We select the "Upload and save" and we save the configuration but when we select the camera in the client program (Ain) the setting is overwrite to "Upload to client".

Is this a bug or an intended behavior? How can we force the desired behavior?
Thanks a lot!

rumengb, 2022-09-10:
Ain forces images to be downloaded to the client as it is not yet ready to work in client less environment. I need to solve several problems before I can make Ain to work in this setup.

Ghir8, 2022-09-10:
Thank you for the replay! 
Ok, we will wait for future Ain update then :)

rumengb, 2022-09-23:
Released!

closed 2022-09-23 by rumengb