indigo-astronomy/indigo #465: Improve file download in remote chained-server environment

Ghir8, 2022-09-24:
Tonight we have tested the new indigo features. The clientless operations are awsome and are steps in a very cool direction. We have just a little improvement to submit.

In a chained-server system, where the camera is connected to the remote server (using the `--remote-server` on the main server) and the client uses the main server imager_agent, the images are saved on the remote server and not the main one. This means that the client, when reconnecting, is unable to download the images, as they are not found on the main server. We solved it by creating a symlink on the main server (`/home/indigo/images/`) to the remote server folder using nfs, but it would be more user-friendly if the images are saved directly on the main server. The user does not expect to have to switch agents in the middle or at the end of the session to download any images not already present on its client.

rumengb, 2022-09-24:
We are aware of this limitation. Unfortunately it is not easy to be resolved on indigo side. So one should either use the imager agent and camera driver on the same host or provide a common storage for the agent and the driver (as you did). The images are stored by the driver theretofore they are always stored on the driver host.

Paolo97Gll, 2022-09-24:
We had thought that there was some low-level cause like this. Thanks for
your explanation! Maybe this can be written in the documentation.

rumengb, 2022-09-25:
done with commit 52e7e9c146024359d4f083ad5c749533db868b5a

closed 2022-09-25 by rumengb