indigo-astronomy/indigo #466: "sideofpier" property in alpaca_agent

Paolo97Gll, 2022-10-03:
I'm using a SynScan EQ-8 mount with the `mount_synscan` driver. The indigo driver exposes the mount side of pier and I can see it from the control panel and ain imager. However, the corresponding alpaca request return "Property or method not implemented".

Here an example taken from my server:

Request:

```
/api/v1/telescope/1/sideofpier
```

Response:

```
{
"Value": 0,
"ErrorNumber": 1024,
"ErrorMessage": "Property or method not implemented",
"ClientTransactionID": 0,
"ServerTransactionID": 158757
}
```

I think that the expected behaviour, given the official description in [this section](https://www.ascom-standards.org/api/#/Telescope%20Specific%20Methods/get_telescope__device_number__sideofpier) of the ASCOM Alpaca Device API, should be to report the current side of pier since this method is implemented in indigo (it's a read-only permission, so a get request it's ok).

polakovic, 2022-10-04:
Stupid typo :( Hopefully fixed here 4e5a14c13411251a5b8c6319464c6aa22c32d4e5, but not tested yet...

Paolo97Gll, 2022-10-04:
I can test it in the next days, I will let you know if it works

Paolo97Gll, 2022-10-04:
I can confirm that it works as expected and results are consistent with the ones reported in the indigo server, thanks!

closed 2022-10-04 by Paolo97Gll

polakovic, 2022-10-05:
👍

BobDenny, 2023-01-04:
Just a quickie... in case your typo was case-related, the newest release of [Conform Universal](https://github.com/ASCOMInitiative/ConformU) has strict casing, runs native on Linux, Mac, and will catch this sort of thing because it speaks native Alpaca (not like the WIndows Conform that runs through the Chooser Dynamic Drivers. So you can test right on your Linux system.

polakovic, 2023-01-04:
No, it was not case related. It was misplaced } :)