indigo-astronomy/indigo #467: Use a custom permanent set of placeholders in CCD file name "PREFIX"

Paolo97Gll, 2022-11-07:
Using AIN imager, the PREFIX (ccd driver property -> Camera -> Save on server -> File name prefix) when starting a batch is set to `name_%-D_%F_%C_%M`, where `name` is the one inserted in AIN imager. Is it possible to use a custom permanent set of placeholders (e.g. `_%-D_%-H_%F_%C_%M`  instead of `_%-D_%F_%C_%M`)?

rumengb, 2022-11-07:
I never intended to make sorting by filename the same as sorting by date. Actually what you mentioned is only part of the "problem". The other part is merging the filenames with the already saved files. And still there is yet another problem when you relay on file names only.  For example, if your preference is to see how many similar frames you have, including the time in the name will not make it obvious as the sequential numbers will be hallways "0". As it is now you look at the last frame number and you have it, but no grantee in the file name order. If you prefer to have the files sorted by name to be sorted by creation time you will not have clear view of how many frames you have in a single filter, ir how many darks you have etc.. So this is a matter of personal preference...
Also including the time makes the filename even longer and harder to distinguish. The sequential number gives a lot clearer distinction. On the other hand if the processing is acquisition time sensitive, like photometry, the software used should be able to read the header information and to get date, time, filter etc. Filename is not a reliable source of information it is just for convenience. A good example is the time: time is the file creation time not observation time which for a 30 min exposure is 30 minutes earlier :) 
Once I add custom file name templates everyone will be able to make the filenames as they wish :) So this is not a bug... this is just a preference :)

closed 2022-11-07 by rumengb

rumengb, 2022-11-08:
With the next INDIGO release the files will be downloaded in order of creation not alphabetically. This way files with sequential numbers will will still be alphabetically ordered (as before) but the files with MD5 digests will be ordered in order of creation so no more random file sequencing using MD5. Still files will not be merged time wise with the locally saved but it is far better that having downloaded files scrambled.