indigo-astronomy/indigo #468: AGadget FocusDreamPro FOCUSER_LIMITS bugfix required

amyznikov, 2022-11-13:
Hello somebody fro indigo team!

 It seems there is a bug in AGadget FocusDreamPro driver in file
 
indigo/indigo_drivers/focuser_focusdreampro/indigo_focuser_focusdreampro.c.

The values for FOCUSER_LIMITS_MAX_POSITION_ITEM->number.max and FOCUSER_LIMITS_MIN_POSITION_ITEM->number.max  are not set in the code.
As result the indigo_enumerate_properties() returns default values for these numbers which are not appropriate for normal work with focuser limits.

The affected lines are 120 an 121 in source file:

    FOCUSER_LIMITS_MIN_POSITION_ITEM->number.target = FOCUSER_LIMITS_MIN_POSITION_ITEM->number.value = FOCUSER_POSITION_ITEM->number.min = 0;
    FOCUSER_LIMITS_MAX_POSITION_ITEM->number.target = FOCUSER_LIMITS_MAX_POSITION_ITEM->number.value = FOCUSER_POSITION_ITEM->number.max = 1000000;

Can somebody fix these lines  by replacing the lines 120 and 121 with  following (or equivalent) ? 

    FOCUSER_POSITION_ITEM->number.min = FOCUSER_LIMITS_MIN_POSITION_ITEM->number.value = FOCUSER_LIMITS_MIN_POSITION_ITEM->number.target = FOCUSER_LIMITS_MIN_POSITION_ITEM->number.min = FOCUSER_LIMITS_MAX_POSITION_ITEM->number.min = 0;

    FOCUSER_POSITION_ITEM->number.max = FOCUSER_LIMITS_MAX_POSITION_ITEM->number.value = FOCUSER_LIMITS_MAX_POSITION_ITEM->number.target = FOCUSER_LIMITS_MIN_POSITION_ITEM->number.max = FOCUSER_LIMITS_MAX_POSITION_ITEM->number.max = 1000000;

This error prevents me from working properly with the my FocusDreamPro focuser,  I fixed this bug locally in my local build, but I would like to fix it in most recent version of indigo too in order to avoid the patching the code each time when indigo code is updated.
    
Thank you a lot in advance !
      Andrey Myznikov

rumengb, 2022-11-13:
Fixed with commit ae57881685669bc6a73f7253ab490427c5196bad can you please test?

amyznikov, 2022-11-13:
Thank you for the update !
Will test in nearest time slot, probably today.

amyznikov, 2022-11-13:
@rumengb ,

Tested the commit right now, it works as expected.  

Thank you a lot !!!
 Andrey

amyznikov, 2022-11-13:
The bug is fixed.

closed 2022-11-13 by amyznikov