indigo-astronomy/indigo #470: Install error on Mac OS

joellama, 2022-12-13:
I'm following the guide and can't get past `make all`. I get the following error

```
/Library/Developer/CommandLineTools/SDKs/MacOSX12.1.sdk/System/Library/Frameworks/Foundation.framework/Headers/NSBundle.h:91:143: error: function does not return NSString
- (NSAttributedString *)localizedAttributedStringForKey:(NSString *)key value:(nullable NSString *)value table:(nullable NSString *)tableName NS_FORMAT_ARGUMENT(1) NS_REFINED_FOR_SWIFT API_AVAILABLE(macos(12.0), ios(15.0), watchos(8.0), tvos(15.0));
                                                         ~~~~~~~~~~~~~~                                                                       ^                  ~
/Library/Developer/CommandLineTools/SDKs/MacOSX12.1.sdk/System/Library/Frameworks/Foundation.framework/Headers/NSObjCRuntime.h:103:48: note: expanded from macro 'NS_FORMAT_ARGUMENT'
        #define NS_FORMAT_ARGUMENT(A) __attribute__ ((format_arg(A)))
                                                      ^          ~
1 error generated.
make[2]: *** [indigo_aux_joystick.o] Error 1
make[1]: *** [aux_joystick/] Error 2
make: *** [all] Error 2
```

Any suggestions would be very much appreciated!

polakovic, 2022-12-14:
To be honest, I can't reproduce it. Can you send us whole "make all" output? Pls. use "make clean-all" first.

What macOS and XCode versions do you use? "/Library/Developer/CommandLineTools/..." doesn't exists on any of my systems (even on macOS 12).

DavidObfuscator, 2023-03-18:
@joellama you should upgrade your command line tools. It has a stripped version of macOS sdk and its a issue with the 12.1 sdk

@polakovic command line tools can be downloaded separately from xcode; it has a smaller version of the SDK, but clang, toolchain is the same. It is dedicated for building command line tools. You can download it from dev portal or just use `xcode-select --install`

tclaus, 2023-11-02:
On my Apple M1 Pro with macOS 14.1 (23B74) it builds directly after cloning the repo without any problems.

closed 2023-11-18 by rumengb