indigo-astronomy/indigo #471: Image Transfer Method - agent_alpaca

BobDenny, 2023-01-04:
I noticed this on the [alpaca_bridge README](https://github.com/indigo-astronomy/indigo/blob/master/indigo_drivers/agent_alpaca/README.md) under [Selecting "Image array transfer transfer method"](https://github.com/indigo-astronomy/indigo/blob/master/indigo_drivers/agent_alpaca/README.md#selecting-image-array-transfer-transfer-method).  In 2019, the new ImageBytes image transfer method was introduced and was made a standard part of ASCOM PLatform 6.6 in January 2022 (a year ago not just now). You might want to revise the README to select ImageBytes as the preferred method. We're hoping those older transfer schemes will simply go away.

polakovic, 2023-01-04:
Yep, README is outdated and we have to update it. ImageBytes transfer mode is already implemented.

> Also if there is anything we can do to help with, or make provisions for, cross-compatibility between INDI and Alpaca, please engage at the [ASCOM Driver and Application Development Support Forum](https://ascomtalk.groups.io/g/Developer) on Groups.io.

As far as this is Alpaca Bridge is an INDIGO agent, it will never work with INDI, just with INDIGO.

polakovic, 2023-01-04:
README updated...

closed 2023-01-04 by polakovic

BobDenny, 2023-01-04:
Peter — Thanks very much for the fast response!! If there is something that we can do  to  make this easier or better please feel free to ask on the ASCOM Developer Forum at https://ascomtalk.groups.io/g/Developer