indigo-astronomy/indigo #475: Error on start, /dev/tty: Inappropriate ioctl for device

Paolo97Gll, 2023-04-02:
We can't start indigo_server 2.0-228, the server crashes with the following error:

```
07:24:34.391371 indigo_server: port_type(): path = /dev/tty, type = 0, res = -1 error = 'Inappropriate ioctl for device'
```
After the crash, the server restarts immediately and automatically.

Complete error log in trace mode:

```
07:26:23.130337 indigo_server: INDIGO server 2.0-228 built on 2023-03-30T11:40:31+00:00 production
07:26:23.130987 indigo_server: Server sodoma.astrogeo.va.it:7624 thread started
07:26:23.157317 indigo_server: Driver indigo_focuser_esatto 2.0.0.9 loaded
07:26:23.157420 indigo_server: B <- Attach device 'PrimaLuceLab ESATTO Focuser'
07:26:23.157452 indigo_server: 1 devices attached
07:26:25.744404 indigo_server: Can't connect to socket 192.168.40.82:7624 (No route to host)
07:26:31.888453 indigo_server: Can't connect to socket 192.168.40.82:7624 (No route to host)
07:26:33.200096 indigo_server: port_type(): path = /dev/ttyS31, type = 0, res = 0
07:26:33.200286 indigo_server: port_type(): path = /dev/ttyS30, type = 0, res = 0
07:26:33.200408 indigo_server: port_type(): path = /dev/ttyS29, type = 0, res = 0
07:26:33.200521 indigo_server: port_type(): path = /dev/ttyS28, type = 0, res = 0
07:26:33.200657 indigo_server: port_type(): path = /dev/ttyS27, type = 0, res = 0
07:26:33.200793 indigo_server: port_type(): path = /dev/ttyS26, type = 0, res = 0
07:26:33.200910 indigo_server: port_type(): path = /dev/ttyS25, type = 0, res = 0
07:26:33.201008 indigo_server: port_type(): path = /dev/ttyS24, type = 0, res = 0
07:26:33.201117 indigo_server: port_type(): path = /dev/ttyS23, type = 0, res = 0
07:26:33.201259 indigo_server: port_type(): path = /dev/ttyS22, type = 0, res = 0
07:26:33.201376 indigo_server: port_type(): path = /dev/ttyS21, type = 0, res = 0
07:26:33.201509 indigo_server: port_type(): path = /dev/ttyS20, type = 0, res = 0
07:26:33.201630 indigo_server: port_type(): path = /dev/ttyS19, type = 0, res = 0
07:26:33.201729 indigo_server: port_type(): path = /dev/ttyS18, type = 0, res = 0
07:26:33.201825 indigo_server: port_type(): path = /dev/ttyS17, type = 0, res = 0
07:26:33.201913 indigo_server: port_type(): path = /dev/ttyS16, type = 0, res = 0
07:26:33.202027 indigo_server: port_type(): path = /dev/ttyS15, type = 0, res = 0
07:26:33.202136 indigo_server: port_type(): path = /dev/ttyS14, type = 0, res = 0
07:26:33.202227 indigo_server: port_type(): path = /dev/ttyS13, type = 0, res = 0
07:26:33.202313 indigo_server: port_type(): path = /dev/ttyS12, type = 0, res = 0
07:26:33.202405 indigo_server: port_type(): path = /dev/ttyS11, type = 0, res = 0
07:26:33.202491 indigo_server: port_type(): path = /dev/ttyS10, type = 0, res = 0
07:26:33.202605 indigo_server: port_type(): path = /dev/ttyS9, type = 0, res = 0
07:26:33.202695 indigo_server: port_type(): path = /dev/ttyS8, type = 0, res = 0
07:26:33.202818 indigo_server: port_type(): path = /dev/ttyS7, type = 0, res = 0
07:26:33.202934 indigo_server: port_type(): path = /dev/ttyS6, type = 0, res = 0
07:26:33.203048 indigo_server: port_type(): path = /dev/ttyS5, type = 0, res = 0
07:26:33.203156 indigo_server: port_type(): path = /dev/ttyS4, type = 0, res = 0
07:26:33.203264 indigo_server: port_type(): path = /dev/ttyS3, type = 0, res = 0
07:26:33.203372 indigo_server: port_type(): path = /dev/ttyS2, type = 0, res = 0
07:26:33.203462 indigo_server: port_type(): path = /dev/ttyS1, type = 0, res = 0
07:26:33.204676 indigo_server: port_type(): path = /dev/tty, type = 0, res = -1 error = 'Inappropriate ioctl for device'
07:26:35.617563 indigo_server: INDIGO server 2.0-228 built on 2023-03-30T11:40:31+00:00 production
07:26:35.618190 indigo_server: Server sodoma.astrogeo.va.it:7624 thread started
07:26:35.637435 indigo_server: Driver indigo_focuser_esatto 2.0.0.9 loaded
07:26:35.637507 indigo_server: B <- Attach device 'PrimaLuceLab ESATTO Focuser'
07:26:35.637534 indigo_server: 1 devices attached
```

I can't say if the error is due to version 2.0-228 or an early version. I can only say that in indigo 2.0-208 the error is present but the server doesn't crash. 

The server is running Ubuntu server 22.04.

rumengb, 2023-04-02:
Hi, the message is just a debug message and has nothing to do with the
server crash.
Also we do not have indigo_focuser_esatto driver. Is this some custom
driver? Most likely this is the driver that crashes.
Unfortunately, The log does not give any information about what happened.
So I want to ask you to do something that will give us more information:

1. Instal valgrind:
$ sudo apt-get install valgrind

2. Run the indigo server as you normally would but add extra "--" option
and run it through valgrind:
$ valgrind indigo_server -- -vvv .... >indigo_valgrind.log 2>&1

if you are running indigo with sudo add it before valgrind:
$ sudo valgrind indigo_server -- -vvv .... >indigo_valgrind.log 2>&1

Then send us indigo_valgrind.log

However, I am almost sure that you should send a report to the driver
developer.

Thank you !
Rumen

polakovic, 2023-04-02:
I had a plan to do it next week :) But I'll wait if you did already...

Paolo97Gll, 2023-04-03:
Hi, sorry for the late reply, I was out!

It was my fault. As @rumengb pointed out, we use custom drivers, but I totally forgot to compile them with the new indigo 2.0-228 (they were compiled with indigo 2.0-208). Now everything works!

Thank you for the help!

closed 2023-04-03 by Paolo97Gll

rumengb, 2023-04-04:
@Paolo97Gll, We have other users asking for this driver. Would you be willing to share it with Us so that we include it in the main distribution?

polakovic, 2023-04-12:
Thank you for the explanation, I understand. I saw INDI drivers already and even more useful was the discussion on INDI developers forum with communication logs, so I think I have an idea how it works. I also noticed that  USB Control Specification Document and I know it is not available anywhere from public sources.

I tried to connect PrimaLuceLab couple of times with no reaction from their side at all :(