indigo-astronomy/indigo #480: Custom tracking rate in PMC 8 EXOS II INDIGO driver not available.

rohan5sep, 2023-06-14:
Not sure if this is the right place to post this issue...I a newbie here. So apologies if this is the not the correct place.
I come from NINA (ASCOM) and after testing INDIGO for few days I am seriously considering making a switch to INDIGO. Everything is working as expected except I do not see the option to set custom tracking rate in the mount driver/agent. It just lists the standard sidereal, solar and lunar rates. Please see the attached image.
 To be able to set custom track rates is important to me as my mount has a huge RA drift. I compensate that by decreasing the tracking rate by -1.502.
My equipment –
1.	PMC 8 EXOS II mount. 
2.	Explore scientific 102mm refractor.
ASCOM driver for PMC 8 EXOS II has “offset” setting where I can do this adjustment.
Is there a similar setting for INDIGO driver too?
INDIGO Version 2.0-240
INDIGO Server: Service indigosky running on Raspberry PI 4 (8 gigabytes memory), connected over Gigabit LAN
INDIGO Client: Ain INDIGO Imager Version 1.0-1 (64 bit) Windows 10 PC, connected over Gigabit LAN

[image]

polakovic, 2023-06-15:
Yep, this is the right place :) There is no such setting in INDIGO driver, but I'll check what we can do.

rohan5sep, 2023-06-16:
> Yep, this is the right place :) There is no such setting in INDIGO driver, but I'll check what we can do.

Thanks for replying...OK -- until that happens I was wondering if I could just tweak the driver code and hardcode the settings I need? How do I build the code? Any build instructions? Would you recommend doing this?

polakovic, 2023-06-16:
Hi, instructions are at the end of top level README or here:

https://www.indigo-astronomy.org/for-developers.html

Just make sure you will load the right driver (that built one, not the one distributed with INDIGO Sky).

rohan5sep, 2023-07-17:
Hello Peter,

Thanks for the information. I was able to successfully build and make the desired changes the PMC 8 driver -- really happy that I was to do this given that I am totally new to Linux. I have just one question: Is there a way to install indigo on RPM based systems like Fedora 38? I did not find the code in DNF.

polakovic, 2023-07-17:
Well, I'm afraid the only way for RP based systems is to build it from source now. @rumengb, any other idea?

rumengb, 2023-07-18:
Currently we build DEBs only and we do not have RPM build support yet.
However we are looking for somebody to build and amaintaing indigo on RPM
distributions...

rohan5sep, 2023-07-21:
Thanks! I would give it a try.

closed 2023-11-18 by rumengb