indigo-astronomy/indigo #482: ccd_ptp: indigo_server crash when connecting Nikon cameras

tail-feather, 2023-07-07:
This problem only occurs in Raspberry Pi with INDIGO Sky.

Using image: 2023-04-09-indigosky-armhf.img
Camera:

* Nikon D5300
* Nikon Z 5

Crash while trying to decode `ptp_property_nikon_MovScreenSize`.

tail-feather, 2023-07-07:
Crash point:
https://github.com/indigo-astronomy/indigo/blob/e6958a8312af77493aee901411959ccefbd93292/indigo_drivers/ccd_ptp/indigo_ptp_nikon.c#L761

tail-feather, 2023-07-07:
If the following properties are skipped, the connection is successful.

* `ptp_property_nikon_MovScreenSize` (`0xd0a0`)
* `ptp_property_nikon_LensType` (`0xd0e2`)
* Unknown (`0xd0e7`)

polakovic, 2023-07-08:
About month ago I made one change on PTP driver which may affect it - unknown properties are not decoded to avoid a similar crash with Fuji XT5. I'll check if it may be related.

polakovic, 2023-07-08:
I tested with D750 now and it worked and unused properties was not decoded. Are you sure that INDIGO was already updated on that image to the most recent version?

tail-feather, 2023-07-08:
Yes. I built and tested master on a Raspberry Pi (INDIGO Sky).

I tried it on Ubuntu and did not have this problem.
This issue seems to occur only on the Raspberry Pi...

tail-feather, 2023-07-11:
I got PTP response dump.

Apparently, the problem occurs when parsing 64-bit integers.

```
d0a0:
88 bytes, 6 lines
[0]: a0 d0 08 00 01 00 00 3c 00 38 04 80 07 00 00 1e 
[1]: 00 38 04 80 07 02 08 00 00 00 1e 00 70 08 00 0f 
[2]: 00 00 19 00 70 08 00 0f 00 00 18 00 70 08 00 0f 
[3]: 00 00 3c 00 38 04 80 07 00 00 32 00 38 04 80 07 
[4]: 00 00 1e 00 38 04 80 07 00 00 19 00 38 04 80 07 
[5]: 00 00 18 00 38 04 80 07 
------------------------------------------------------------------------
code: a0 d0 : d0a0
type: 08 00 : 0008 -> ptp_uint64_type
writable: 01 : true
skip: 00 00 3c 00 38 04 80 07
value: 00 00 1e 00 38 04 80 07 : 07800438001e0000
form: 02 -> ptp_enum_form
count: 08 00 : 0008
- 00 00 1e 00 70 08 00 0f 
- 00 00 19 00 70 08 00 0f
- 00 00 18 00 70 08 00 0f 
- 00 00 3c 00 38 04 80 07
- 00 00 32 00 38 04 80 07 
- 00 00 1e 00 38 04 80 07
- 00 00 19 00 38 04 80 07 
- 00 00 18 00 38 04 80 07 

d0e2:
22 bytes, 2 lines
[0]: e2 d0 08 00 00 01 00 00 00 00 00 00 00 00 00 00 
[1]: 00 00 00 00 00 00 
------------------------------------------------------------------------
code: e2 d0 : d0e2
type: 08 00 : 0008 -> ptp_uint64_type
writable: 00 : false
skip: 01 00 00 00 00 00 00 00
value: 00 00 00 00 00 00 00 00
form: 00 : ptp_none_form

d0e7 [115]
22 bytes, 2 lines
[0]: e7 d0 08 00 00 00 00 00 00 00 00 00 00 80 00 00 
[1]: 00 00 00 00 00 00 
------------------------------------------------------------------------
code: e7 d0 : d0e7
type: 08 00 : 0008 -> ptp_uint64_type
writable: 00 : false
skip: 00 00 00 00 00 00 00 00
value: 80 00 00 00 00 00 00 00
form: 00  : ptp_none_form
```

tail-feather, 2023-07-11:
And I got coredump.

```
Program terminated with signal SIGBUS, Bus error.
#0 ptp_decode_uint64(source=source@entry=0xf5518d05 "", target=0xf3cfd488, target@entry=0xf3cfd480) at indigo_ptp.c:463
643            *target = *(uint64_t *)source;
```

Is this an alignment issue on a 32-bit OS?

polakovic, 2023-07-11:
Hm, this may be the issue. Pls. can you try to replace that line 463 with something like this?

memcpy(target, source, sizeof(uint64_t));

polakovic, 2023-07-12:
The bad news is, that I can't reproduce the issue even on 32 bit INDIGO Sky :(

polakovic, 2023-07-12:
Pls. can you test it with the following change?

df0866842461dbe082551fb54d22b58da290baf0

tail-feather, 2023-07-13:
It worked!
Thank you!

closed 2023-07-13 by tail-feather