indigo-astronomy/indigo #488: Script function indigo_enable_blob not working as expected

je-lamiaud, 2023-08-16:
When calling indigo_enable_blob in a script, if the argument is "true", BLOB transfer is disabled.
The script line is:
`indigo_enable_blob(CcdName, "CCD_IMAGE", true)`
The log traces:
` <enableBLOB device='Atik Large Format Camera' name='CCD_IMAGE'>Never</enableBLOB>`

When called with argument "false", the blob transfer is enabled, and my script is able to save the image.
The log traces:
`<enableBLOB device='Atik Large Format Camera' name='CCD_IMAGE'>Also</enableBLOB>`

Note: as expected, without calling indigo_enable_blob at all, the BLOB transfer is disabled.

polakovic, 2023-08-16:
Hopefully fixed here: fe18595c284201e34d115ece9a94a5cca9a8accb

3rd parameter false should map to "Never" and true to "URL" (for INDIGO) or "Also" (for legacy INDI backend).

je-lamiaud, 2023-08-17:
Yes, it works fine.
Thanks.

closed 2023-08-19 by polakovic