indigo-astronomy/indigo #490: Exposure times out likely because of permission issues.

rohan5sep, 2023-08-26:
First of all I am new to Linux and have a very limited understanding of file permissions. It is highly likely that I have missed something here...

I get -- pull_callback() was not called in time -- error when running with below setup.  Initially, I thought the error could be because of exposure watchdog timeout. But, then it starts to work fine if I manually start the indigo server from terminal using “sudo”.

I tried elevating file permissions shown below without success -

1. /usr/bin/indigo_ccd_touptek
2. /usr/bin/indigo_ccd_touptek.so
3. /usr/lib/libtoupcamsdk.so

What am I missing here? I have attached the debug trace as well. 

Equipment:

Toupcam branded as BRESSER Full HD Deep-Sky Camera

Setup 

INDIGO Server 242 running on Ubuntu server LTS 22.04, connected over Gigabit LAN
INDIGO Client: Ain INDIGO Imager Version 1.0-1 (64 bit) Windows 10 PC, connected over Gigabit LAN.

Drivers loaded: indigo_ccd_touptek

Configuration

Enabled following indigo-service using below commands --

    1. sudo systemctl enable indigo-servicesmbsmb://drive@ocean.local/share/Toup/PullCallback.txt://drive@ocean.local/share
    2. sudo systemctl start indigo-service
[PullCallback.txt](https://github.com/indigo-astronomy/indigo/files/12445059/PullCallback.txt)

rumengb, 2023-08-26:
> 1. sudo systemctl enable ***@***.******@***.***/share

Are you running indigo installed from a deb package or you build it on your
own?

If you install it from deb - mostlikely we are missing permission rule for
your camera. in this case we will need to find the USB vid and pid of the
camera to add it.

if you build it on your own you need to install the touptek rules file so
that the access to the camera is granted.

please let me know how you install indigo so that I can help ypu how to
proceed.

rohan5sep, 2023-08-26:
> Are you running indigo installed from a deb package or you build it on your
> 
> own?
> 
> 
> 
> If you install it from deb - mostlikely we are missing permission rule for
> 
> your camera. in this case we will need to find the USB vid and pid of the
> 
> camera to add it.
> 
> 
> 
> if you build it on your own you need to install the touptek rules file so
> 
> that the access to the camera is granted.
> 
> 
> 
> please let me know how you install indigo so that I can help ypu how to
> 
> proceed.
> 
> 
> 
> 
> 
> 
> > First of all I am new to Linux and have a very limited understanding of
> 
> > file permissions. It is highly likely that I have missed something here...
> 
> >
> 
> > I get -- pull_callback() was not called in time -- error when running with
> 
> > below setup. Initially, I thought the error could be because of exposure
> 
> > watchdog timeout. But, then it starts to work fine if I manually start the
> 
> > indigo server from terminal using “sudo”.
> 
> >
> 
> > I tried elevating file permissions shown below without success -
> 
> >
> 
> >    1. /usr/bin/indigo_ccd_touptek
> 
> >    2. /usr/bin/indigo_ccd_touptek.so
> 
> >    3. /usr/lib/libtoupcamsdk.so
> 
> >
> 
> > What am I missing here? I have attached the debug trace as well.
> 
> >
> 
> > Equipment:
> 
> >
> 
> > Toupcam branded as BRESSER Full HD Deep-Sky Camera
> 
> >
> 
> > Setup
> 
> >
> 
> > INDIGO Server 242 running on Ubuntu server LTS 22.04, connected over
> 
> > Gigabit LAN
> 
> > INDIGO Client: Ain INDIGO Imager Version 1.0-1 (64 bit) Windows 10 PC,
> 
> > connected over Gigabit LAN.
> 
> >
> 
> > Drivers loaded: indigo_ccd_touptek
> 
> >
> 
> > Configuration
> 
> >
> 
> > Enabled following indigo-service using below commands --
> 
> >
> 
> > 1. sudo systemctl enable ***@***.******@***.***/share
> 
> > 2. sudo systemctl start indigo-service
> 
> >
> 
> > PullCallback.txt
Hi Rumen,

I used this link to install from apt repository -- I did not build the code.

https://www.indigo-astronomy.org/downloads.html

Thanks,
Rohan

rumengb, 2023-08-26:
>    1. /usr/bin/indigo_ccd_touptek.so
>    1. /usr/lib/libtoupcamsdk.so
>    1. sudo systemctl start indigo-service

in that case touptek are missing the vid and pid of your camera model in
the provided rules. can you run lsusb command  without camera connected and
with camera connected the device that appeared is your camera. this way I
can identify the camera. can you send us both outputs so that I can fix the
rules?

rohan5sep, 2023-08-26:
> in that case touptek are missing the vid and pid of your camera model in the provided rules. can you run lsusb command without camera connected and with camera connected the device that appeared is [...]

[image]

[image]

I have also attached text files with lsusb -v option in case that is needed. 
[WithCameraVerbose.txt](https://github.com/indigo-astronomy/indigo/files/12445601/WithCameraVerbose.txt)
[WithoutCameraVerbose.txt](https://github.com/indigo-astronomy/indigo/files/12445602/WithoutCameraVerbose.txt)

rumengb, 2023-08-26:
excellent! I will add it to the rules file and will send it to you to
replace the original when I am home tomorrow. if it works we will fix it in
the new indigo release.

rohan5sep, 2023-08-26:
> excellent! I will add it to the rules file and will send it to you to replace the original when I am home tomorrow. if it works we will fix it in the new indigo release.
> […](#)

Sounds good. Thanks.

rumengb, 2023-08-27:
The rules are ok.
So i have read that ubuntu server does not run systemd-udevd by default.
so you may need to check if udevd is running.... Which makes some sense you
do not need usb hotplug on a server system...

rohan5sep, 2023-08-28:
Hi Rumen,

I confirmed that udevd is already running. However, the process was showing warnings related to nvme not mounting. This is because of a [bug](https://bugs.launchpad.net/ubuntu/+source/systemd/+bug/1966203). The problem still exists even after fixing these warnings.

Apart from this I have tried several things like stating the indigo service from root account, adding "indigo" to sudo group. I even created a special usbuser group and added udevd rules for USB and made "indigo" member of this group. But none of that worked. 

**Finally, changing the default user in indigo-server.service from "indigo" to "root" fixes the problem.** It is odd that this problem only appears -- at least for me -- on headless Ubuntu server LTS 22.04 installs. It works perfectly on Linux Mint running Cinnamon for example. 

Thanks

closed 2023-11-18 by rumengb