indigo-astronomy/indigo #491: LX200 - Support OpenAstroTech mounts

jimtjames, 2023-08-27:
The current LX200 driver does not support [OpenAstroTech](https://wiki.openastrotech.com/) mounts. These mounts have an LX200-like serial protocol documented [here](https://wiki.openastrotech.com/Knowledge/Firmware/MeadeCommands).

Some key differences are a 19200 baudrate and extra properties for add-ons (GPS, level, etc.).  I've started work on supporting this mount on  [this fork](https://github.com/jimtjames/indigo), mostly just using the Meade flavor of commands with a 19200 baudrate.

polakovic, 2023-08-27:
I’ll look at it.

jimtjames, 2023-08-27:
Great, thank you! The fork I wrote is able to connect to the mount and move it in Ain, though I haven't tested it much beyond that.

closed 2023-09-12 by polakovic