indigo-astronomy/indigo #507: Mount Simulator/Mount Synscan : park position stuck at north pole

wgoltz, 2023-11-15:
I am in the southern hemisphere. There doesn't appear to be any way to tell the Mount Simulator to park at the south pole. Similarly the hour angle of Home position is always 6 hrs and can't be changed. Using Synscan the park position on startup (unparked) appears to correctly showing the south pole but pressing park button tries to send the scope to the north pole (not nice!)

rumengb, 2023-11-16:
When you set the latitude to a negative number as shown below:
[image: image.png]
The park position will automatically change to -90 degrees as shown below:
[image: image.png]
Also, you can enter custom Park Position here.
If you want to preserve this settings you need to go to: [Mount name] ->
Main ->Profile selection and select Profile #0

[image: image.png]
Then select Configuration Control as shown below and click on Save:
[image: image.png]
This way the park position will be preserved for this mount
hope this helps!

rumengb, 2023-11-16:
What driver are you using?

wgoltz, 2023-11-16:
I have been experimenting with the Mount Simulator, Mount SynScan and Mount Nexstar and the Mount Agent with a Skywatcher AZeq5.

I am using the web interface and Skychart (Carte du Ciel) and gcx programs (my version of gcx is on github). I run indigo_server on a remote Pi controlling the telescope and I have indigo_server running on my laptop which just provides mousepad support. Skychart I run on the laptop, gcx I run on the Pi as it gives better performance running there. It is all connected together over WiFi using the home broadband modem.

I have been having problems with the Pi freezing or at least running very slowly. This problem with parking is the first definite error that seems to be related to indigo.

wgoltz, 2023-11-16:
Simplest way to reproduce problem for me: Starting with telescope at home position (some hour angle, -90) but unparked (what you get with AZeq5 after starting at home position). Initialize indigo by restarting server, connect nexstar mount, load nexstar configuration, check nexstar mount position displays (0, -90), it does. Now select park button in nexstar mount - the telescope heads of to the north pole!

rumengb, 2023-11-16:
I have checked the Mount simulator and the Nexstar driver with my AZ-EQ6 mount. 
1. mount simulator - yes there is a bug in the simulator - it always starts fron +90degrees regardless of the latitude - I will fix it.
2. Nexstar - no I can not reproduce the problem. But I see what you are missing. Setting the  park position is not preserved between the sessions. To make it permanent choose "[mount name] -> Main -> Profile selection -> Profile #0" as shown:

[image]

then save current configuration with  "[mount name] -> Configuration Control -> Save"

[image]

if saved to profile #0 it will be auto-loaded at start.

This way your park position will be made permanent and toy can enter whatever HA and Dec you like.

wgoltz, 2023-11-16:
Using the browser interface and Mount Nexstar, profile #0 is NOT loaded automatically and the park position shows as (6, 90). When I manually click load in Configuration Control the position shows correctly (0, -90). Perhaps auto-loading does not always work correctly with the browser interface? I have been confused before not knowing whether profiles have been loaded, not just with Mount Nexstar.

rumengb, 2023-11-16:
Sorry, it is my bad. it is autoloaded by the mount agent. So you need to load the configuration manually if you are not using it through the agent.

wgoltz, 2023-11-16:
Using the agent, the park position shows as (0, 90) when the manually loaded park position was (0, -90).

rumengb, 2023-11-16:
Just checked it and it works:
[image]
you need to select the mount from the mount list as shown above.

wgoltz, 2023-11-17:
When everything starts up fresh, the Mount Agent consistently shows (0, 90) (north pole) as the current coords . Mount Nexstar and the display on the controller both agree they are pointing at the south pole, though they slightly differ. I can slew to a nearby star. The coords in Mount Agent agree much the same as coords on the controller, difference in epoch between the two? Check park position in Mount Nexstar (0, -90), good. Click park button in Mount Agent - off to north pole again!

Same result as reported earlier.

wgoltz, 2023-11-17:
It looks like I might have been editing options without loading presets first. My changes get saved back with other default options which are wrong.

I can get SynScan to park at south pole. There is still a problem with SynScan home/park positions - the home hour angle always returns to 6hrs and the park position is off by 6hrs to what I select.

Nexstar still goes to the north pole when parked.

As an aside, the Pi got stuck with the Mount Agent sky display drifting whilst I was playing with SynScan options. Response improved when cable to mount was disconnected. I am not sure whether the cable should be directly connected to the mount or through the synscan controller. It seems both ways work, but the direct connection was immediately slower and coincided with the Pi getting stuck (maybe just coincidental).

wgoltz, 2023-11-17:
There is no MOUNT_HOME_POSITION saved to ~/.indigo/SynScan.config so it cannot be restored from there (if that is the way it is supposed to work).

Checking for something similar with Nexstar, I reset the actual telescope and the server, started from park and slewed to a nearby star. When I switched from Mount Agent to Mount Nexstar the equatorial coords got zeroed out somehow. Restarted server and the coords were ok again. Tried a park and the telescope looked like it was headed to the south pole ... but no - it went straight though and off to the north pole again.

I haven't had any success getting the log file to work. I deleted the existing log file at /home/indigo/indigo.log but it hasn't reappeared.

rumengb, 2023-11-17:
I will try to reproduce it.

wgoltz, 2023-11-17:
I have sorted out the log files. Turning on Debug in log levels and using Nexstar the log shows:

indigo_mount_nexstar[mount_handle_park:445]: Going to park position: HA = 0.00000 Dec = -90.00000

before it then goes and parks at the north pole.

When it is parked correctly, the synscan controller internal axes always show as (0, 90), but I expect that is irrelevent.

rumengb, 2023-11-17:
Yes, you are right MOUNT_HOME_POSITION was not saved. I have fixed it in git. Most likely nobody ever changed home position. Probably, because the home position is nonchangeable and is a fixed position marked by a sensor. Skywatcher and Celestron are one of these. For these mounts, there is also no park position supported by the protocol and it is also emulated by the driver.

wgoltz, 2023-11-17:
Great! A fix. 

I never actually use the park through indigo as there was obviously a problem the first time I tried. I always use the controller. That is not operator free, which is what (any sane, armchair astronomer) is aiming for.

closed 2023-11-23 by rumengb