indigo-astronomy/indigo #509: Mount Synscan with AZeq5 hang with log and deadlock files

wgoltz, 2023-11-18:
A hang during with AZeq5 and Mount Synscan driver

Operations to get the hang 
[indigo_state.txt](https://github.com/indigo-astronomy/indigo/files/13399149/indigo_state.txt)

tail indigo.log 
[indigo_log.txt](https://github.com/indigo-astronomy/indigo/files/13399150/indigo_log.txt)

deadlock output 
[indigo_deadlock.txt](https://github.com/indigo-astronomy/indigo/files/13399151/indigo_deadlock.txt)

rumengb, 2023-11-18:
I assume you are using the web interface as there is no sky chart in the mount agent itself. After a server restart you may need to reload the page. and also web interface is just for a demo not for a real use.

The attached indigo log is not complete and probably not a trace log therefore is useless.

indigo deadlock log shows no blocked threads let alone deadlocked threads. Actually the server looks like it is completely idle.

What operating system, what platform and what browser are you using?

wgoltz, 2023-11-18:
Thank you for your comments. 

I was challenged for time and not aware of what might be significant and where to start.

I am using the web interface as I thought it was a significant part of the distribution. So I am wrong there, and will use indigo_control_panel for further testing.

The joystick is attached to my laptop running linux and indigo, which just provides game-pad for hand control. indigosky runs on a Pi connected to the telescope, with everything connected by WiFi over the home router.

By slow, I meant that the response was already slow at that point. The indigo.log was a very long list of the same messages. It is just a sample to get some comments so I can get on the right track! Do I have to start a trace log before it hangs or can it be started after the response gets bad? As yet I have no clue as to what is causing my problems.

Yes, I was expecting it to pick up the coordinates ready for a slew, but there was no visible response.

rumengb, 2023-11-18:
Let me see if I understand you:
So you are running two Indigo servers
Server 1: running on Raspberry Pi with the telescope attached to it.
Server 2: running on an Intel laptop running Linux with an attached gamepad.

In that case, you need to run the indigo_server on the laptop with -r or --remote-server to connect to the Rpi.
and control everything from the indigo_server running on the laptop as it will proxy the devices running in the remote server.

If this is the case please bear in mind that if you restart the remote indigo_server the connection will be lost and such behaviour can be expected. 

Also please note, that we are a team of developers and astronomers, there are no clairvoyants on our team. Please consider this handicap of ours when explaining the issue you are experiencing.

wgoltz, 2023-11-18:
Thankyou for your reply,

I do appreciate that you are not clairvoyants. On the other hand, I know very little about your system and it takes me a fair while to write down my problem attempting to explain everything when I know nothing!

I have been running with the main server running on the pi with -r my_laptop. I will try the other way around.

wgoltz, 2023-11-19:
I have changed my server setup on my laptop so that indigo-server runs with -r indigosky, and reset the setup on indigosky to its default (I modified the appropriate files in /usr/lib/systemd/system). The laptop enables the drivers HID-gamepad and Mount Agent. indigosky enables the drivers Mount Simulator, Mount SynScan and Mount Nexstar which I want to test.

When I run indigo_control_panel on indigosky, the left hand pane of the screen lists all the loaded drivers @ indigosky and the output is [indigo_control_panel_startup_indigosky.txt](https://github.com/indigo-astronomy/indigo/files/13401764/indigo_control_panel_startup_indigosky.txt)

The browser interface seems to work for both localhost:7624 and indigosky.local:7624.

When I run indigo_control_panel on the laptop, the screen is empty. I get this output [indigo_control_panel_startup.txt](https://github.com/indigo-astronomy/indigo/files/13401750/indigo_control_panel_startup.txt)

What am I missing here?

rumengb, 2023-11-19:
Goto "File -> Manage services" and you should see both services. Then click the checkbox to connect to the service. 
you may want to check another setting: "Settings -> Auto connect new services".

rumengb, 2023-11-19:
And one more thing. In this setup, you need to use the Indigo service on the laptop. Otherwise, you will not see devices exported by the laptop. In other words the service on the laptop is a proxy for the indigosky.

wgoltz, 2023-11-20:
I did as you suggested and indigo_control_panel correctly showed the drivers at remote indigosky. But it did not show drivers on my laptop (namely, the gamepad). The browser interface worked correctly.

I tried to get the local (laptop services) to appear by removing "-r indigosky" when starting indigo_server on the laptop, but get the error:

avahi: Failed to resolve service 'indigosky': Timeout reached
 
I played with .config/indigo_control_panel.services (trying removing the reference to indigosky, and deleting the file altogether).

With indigosky server running and server on laptop run as: indigo_server -vv -r indigosky

indigo_control_panel console messages (run on laptop):
[indigo_control_panel_initial_startup.txt](https://github.com/indigo-astronomy/indigo/files/13405030/indigo_control_panel_initial_startup.txt)

With server on laptop running as: indigo_server -vv
(no reference to indigosky, and with  .config/indigo_control_panel.services  deleted)

indigo_control_panel console messages (run on laptop):
[indigo_control_panel_startup.txt](https://github.com/indigo-astronomy/indigo/files/13405036/indigo_control_panel_startup.txt)

indigo_control_panel now seems always to be looking for indigosky, regardless of whether it should be looking; and it should be able to find it (the browser interface can). It stops looking for anything else (the local drivers) when indigosky can't be found. I end up with nothing displayed in indigo_control_panel window when I click on manage services. All the devices, local and remote, are there in the browser interface, and get correctly set as the servers change state.

wgoltz, 2023-11-20:
Reboot of indigosky machine got us back to where we were - indigosky drivers visible but not the laptop drivers.

wgoltz, 2023-11-20:
Shutdown indigosky and removed reference to indigosky in .config/indigo_control_panel.services. Running indigo_control_panel no longer looks for indigosky. 

Nor does it find the services on the local machine. Output from indigo_control_panel: 
[indigo_control_panel_noindigosky.txt](https://github.com/indigo-astronomy/indigo/files/13405462/indigo_control_panel_noindigosky.txt)

I get blank screen when clicking manage_services in indigo_control_panel, though I have the joystick and a couple of other devices selected according to the browser interface.

wgoltz, 2023-11-21:
Starting "indigo_server -r indigosky.local" on the laptop seems to have worked wonderfully well! 

Initial experiments show mount simulator and ccd camera simulator are working perfectly, but yet to try with the real telescope and camera.

Thankyou.

closed 2023-11-23 by rumengb