indigo-astronomy/indigo #515: Crash in generation of preview image

tail-feather, 2024-01-07:
Crashes when RAW16 setting and preview generation is enabled.

Camera: ZWO ASI294MM Pro
CCD_MODE = RAW 16 8288x5644
CCD_PREVIEW = Enabled

Crashes do not occur for RAW8 settings.

Oddly enough, this problem is less likely to occur with fewer pixels cameras (e.g. ASI120MM Mini). But, it does occur rarely.

polakovic, 2024-01-07:
Hm, maybe JPEG is larger than RAW, @rumengb, what do you think?

rumengb, 2024-01-07:
I have been trying to reproduce the issue for several hours now with no
success.
On what OS are you running Indigo? If Linux can you try running it through
valgrind:

valgrind indigo_server -- -v [drivers] >indigo_valgrind.log 2>&1

reproduce the crash and send us indigo_valgrind.log

Rumen

tail-feather, 2024-01-08:
I got the log.
It appears that it is trying to allocate a size that exceeds `INT_MAX`.

```
08:39:18.135027 indigo_server: indigo_ccd_asi[asi_read_pixels:403]: ASIGetDataAfterExp(0) = 0
==1587== Warning: silly args (-256206080,1) to calloc()
indigo_server: /indigo-2.0-264/indigo_libs/indigo/indigo_bus.h:797: indigo_safe_malloc: Assertion `pointer != NULL' failed.
--1587-- WARNING: unhandled syscall: 407
```

Is this a problem here?
https://github.com/indigo-astronomy/indigo/blob/f828273bb3b6846fdfa508a6116477eb71566194/indigo_libs/indigo_ccd_driver.c#L938

rumengb, 2024-01-08:
can you please provide the complete log please!

tail-feather, 2024-01-08:
Sure.
[indigo_valgrind.log](https://github.com/indigo-astronomy/indigo/files/13857991/indigo_valgrind.log)

rumengb, 2024-01-08:
Thanks a lot.
Are you running it on 32 bit Raspberry Pi OS with 64 bit kernel?
as valgrind reports a lot of nonsense which it does in this wired new
raspberry pi os configuration.

so I would ask you to provide a RAW file that provokes this crash. Just
stop jpeg preview and set format to indigo raw. and save this raw file and
send it to me.
Thank you!

tail-feather, 2024-01-08:
Sorry I forgot report that.
I used INDIGO Sky (armhf) & Raspberry Pi 4.

* [RAW8.zip](https://github.com/indigo-astronomy/indigo/files/13858346/RAW8.zip)
* [RAW16.raw](https://filedepo.astroarts.co.jp/get/Ysw8IMSHg8VqKiWqLr0H) (Download limit at 2024-02-07. Used our uploader due to file size limitations on GitHub.)

```
35f1106c415d4379e8eda50340915ca8a50a120e  RAW16.raw
28a2fc60c726a3c99d0e6b3ba604d90319e81b16  RAW8.raw
```

rumengb, 2024-01-08:
fixed with commit #3e3568249f220afd19a187bc6b6ce813a82910d1
but please note this version is a work in progress and it has some issues that need to be taken care of

closed 2024-01-08 by rumengb

tail-feather, 2024-01-08:
I built 3e3568249f220afd19a187bc6b6ce813a82910d1 and confirmed that no problems occurred.
Thank you!