indigo-astronomy/indigo #530: PegasusAstro FocusCube 3

chrsoo, 2024-04-14:
The PegasusAstro Focuser DMFC driver indicates support for...

- PegasusAstro Motor Focus Kit
- PegasusAstro FocusCube
- PegasusAstro FocusCube 2

Doing due diligence for potential purchase; [FocusCube 2 is now discontinued](https://pegasusastro.com/products/focuscube-2-universal/) and replaced by FocusCube 3.

Does anybody know if the newer [FocusCube 3](https://pegasusastro.com/products/focuscube-3-universal/) is also supported?

If not, any plans of supporting it in the near future?

polakovic, 2024-04-14:
I doubt it will work out of box, but it should be easy to add FC3 support. I'll ask Pegasus for protocol description.

polakovic, 2024-04-14:
Evangelos promised to send documents once they return from NEAF24 (around 24 April).

chrsoo, 2024-06-21:
@polakovic - any news from Evangelos / PegasusAstro?

polakovic, 2024-06-23:
Not yet :( I'll try to ping Evans again...

polakovic, 2024-07-04:
Still nothing, I pinged him again today...

polakovic, 2024-07-18:
Hopefully done, I had to make new driver for FocusCube3+ devices, because the protocol is substantially different:

https://github.com/indigo-astronomy/indigo/tree/master/indigo_drivers/focuser_focuscube

closed 2024-07-18 by polakovic