indigo-astronomy/indigo #535: Streaming of SVBONY camera does not stop

tail-feather, 2024-05-09:
I found a condition that does not terminate when trying to abort streaming.

Normally, if exposure aborted, the `SVBGetVideoData` returns an error (`SVB_ERROR_TIMEOUT`) and streaming stops.

If the `SVBGetVideoData` does not return an error, the `CCD_ABORT_EXPOSURE_PROPERTY` remains busy.
(But it can be stopped by executing the abort process twice.)

-----

I have a question about the aborting process.

In the `svb_clear_video_buffer` function, the `SVBGetVideoData` function is called until it returns an error (`SVB_ERROR_TIMEOUT`).

Am I correct in my understanding that in the `streaming_timer_callback` function, it stops when the `SVBGetVideoData` function fails?

tail-feather, 2024-05-09:
If streaming aborted, `break` is executed on this block.
https://github.com/indigo-astronomy/indigo/blob/41cc06eed5a8bbc8b43d478589b1404840eb6bdc/indigo_drivers/ccd_svb/indigo_ccd_svb.c#L556-L559

Since the `res` variable is only updated by the above `SVBSendSoftTrigger` function, it appears that streaming is not aborted as long as this function continues to succeed.

https://github.com/indigo-astronomy/indigo/blob/41cc06eed5a8bbc8b43d478589b1404840eb6bdc/indigo_drivers/ccd_svb/indigo_ccd_svb.c#L551

rumengb, 2024-05-09:
hopefully fixed: 7480328a9580b12b87a82cdb1712dd1ce613ed01

tail-feather, 2024-05-10:
I've tried it.
It seems to work. Thank you very much!

closed 2024-05-10 by tail-feather