indigo-astronomy/indigo #537: After update to latest version, mount park declination gets set to +90

wgoltz, 2024-05-19:
As per title.

I need park dec to be -90.

I have been using Mount Simulator and Mount Nexstar.

I have to do a unpark/park to get indigo to the correct state. I get into all sorts of trouble if I don't realize an update has been done!

rumengb, 2024-05-20:
I understand your frustration. but please be more descriptive and explain what you are trying to achieve and what happens. Send us the steps to reproduce. Also, a good log from indigo server will help to better understand what is going on as described in the last two sections of https://www.indigo-astronomy.org/indigo-sky.html

Thank you!

wgoltz, 2024-05-21:
I'm sorry that I am being a pain. I think the Indigo project is wonderful, and it fits neatly into the project I have been working on - maintaining a version of gcx-1.5.

There are lots of things in Indigo which are mysterious to me. There are things not documented, or the documentation is hard to find, or I am not trying hard enough. I have attempted to compile it from source, trying to use the debugging features of QtCreator to explore things in more detail, but I haven't managed to get that to work.

The problem with the park position can be easily avoided by never using the park position. Sometimes I forget, and it takes some tens of minutes to figure out what's wrong and rectify the problem, unless I just give up for the night. I would have thought the source of the problem is that the park position is not saved and restored - the wrong values are hard coded into the system somewhere, and that should be easy to find for someone familiar with system.

Regards William

wgoltz, 2024-05-21:
Here is my indigo.log file. Indigo-server was restarted, the mount connected and profile loaded. The park position is (0, -90) and the mount is parked, but the equatorial coordinates show declination +90 (it should be -90).

Doing an unpark then park moves the telescope declination to -90, but, if it gets restarted, indigo-server sets the declination back to +90.

rumengb, 2024-05-22:
Unfortunately, the log can not be sent this way. can you send it to the email mentioned on the page above: [indigo@cloudmakers.eu](mailto:indigo@cloudmakers.eu)

rumengb, 2024-05-27:
I'm sorry for not getting back to you sooner. I was at a conference. This is a bug in the Mount simulator driver. It always starts from +90 degrees regardless of the park position.  We will try to fix it before the next release.

polakovic, 2024-05-28:
Can you test if this fixed your issue b3aa9db6dce39a41914f9b25f86b5ba74383e3ee ?

wgoltz, 2024-05-29:
That has fixed the problem for the Mount Simulator.

Mount Nexstar has a similar (probably the same) problem. (It might apply to all the mount drivers?) It would be great if that was fixed too.

Thankyou for your help.

Regards William

rumengb, 2024-05-29:
Can you provide a log. It is not the same problem.

rumengb, 2024-05-29:
I have checked the driver. I can not see how this can happen. The nexstar mount does not have park command and it does not know anything about park position. Park is simulated by the driver and park state is preserved until next connect. So in general when you park disconnect and then connect again the mount will not move but the driver will not report parked state any more. And it will report the coordinates as reported by the mount, which should be the coordinates of the park poison. This means that if the mount is parked at -90 it should report -90. So I really need a log.

rumengb, 2024-05-29:
can you try this: e38b95f940aeb8bcdf7ddde2cbc54a4803174eea.
I believe it is fixed.

closed 2024-06-11 by rumengb