indigo-astronomy/indigo #559: Driver support for AstroLink 4 Pi

wannabecoffeenerd, 2024-08-31:
Using indigo sky on a Raspberry Pi is very nice, but it would be even nicer to use it on an all-in-one device like the [AstroLink 4 Pi](https://shop.astrojolo.com/product/astrolink-4-pi/). This is based on a Raspberry Pi compute module, so everything should work out of the box, but it would be great if the drivers for focuser controller, switchable power outlets, etc would also be available as an indigo driver. There is an indi driver, not sure how hard it would be to port that over to an indigo driver: https://github.com/astrojolo/astrolink4pi.

If this is something that is interesting to the indigo devs, I'm happy to take on some of the driver porting work if I can get some pointers on what exactly needs to be done.

rumengb, 2024-09-02:
Hi,
we have a driver called "indigo_aux_rpio" it already supports the 2 PWMs and the POWER/HEATER outlets along with several inpits. So We already have something pretty close to what you need and however power GPIOs may need to be remapped to match the ports of the actual device. and also the motors are not supported as we did not implement software PWM we only ise the two hardware ones on the RPi.

INDI drivers can not be easily ported to  Indigo as the structure is very different. But the good news is that we already have the basic ports support and it can be extended with the focuser support etc...

Another very similar driver is "indigo_aux_asiair" which can be used as a base too. it also supports the same ports out of the box (with some remaping ofc.)

Rumen

wannabecoffeenerd, 2024-09-09:
@rumengb Thanks for the pointers and the suggestion. It would be great to add support for the focus motor controller as well. I'm happy to try and take this on if possible. Any suggestions on what focuser driver I can use as a good starting point maybe?

astrojolo, 2024-09-17:
Hello,

In AstroLink 4 Pi INDI driver we use pigpio library ( https://abyz.me.uk/rpi/pigpio/ ) to control all the GPIO pins and functions. There are the following functions available:

1. three on/off pins to control switchable DC outputs
2. two PWM pins to control dew cap heaters outputs
3. several on/off pins to control the focusing stepper motor controller (step, direction, microstepping resolution, enable, reset)
4. one PWM pin to control the focusing stepper motor current
5. I2C buc control (also with pigpio) to read values from environmental sensors and also internal voltage / current / power sensors

There is also RTC module in AstroLink 4 Pi, but I assume this does not need to be supported in the driver. 

Unfortunately, we have zero experience in indigo development :( 

Lucas

rumengb, 2024-09-17:
Hi,
it is easy to add the support for all this things.
 And we do not need the pigpio library. It is quite complicated, (it runs
additional daemon) and is specific to RPI.

what we have out of the box with the RPI OS is:
1. control over every GPIO pin via /sys/class/gpio interface
2. control over 2 hardware PWMs via /sys/class/pwmchipX interface
3. one I2C bus via /dev/i2c-X  interface

so what we miss is stepper motor control and one software PWM. This is easy
to add - it would be less than 100 lines of code for both.
What bothers me though is the busy cycles of the PWM you can easily load
one cpu to the max for a high frequency PWM. So in general it is easy to
add. but I do not have such a device to be able to test the changes. but on
the other hand i2c and hardware pwms are on a specific pins so they should
work out of the box. abowd the three DC outputs we need to map them
accordingly. So the support is not hard to add  i just do not have the
hardware to test the changes.  Otherwise it can be done in a weekend.

Rumen

astrojolo, 2024-09-17:
Hi Rumen,

Thank you for the response. 
AstroLink 4 Pi device is based on RPi, so it is RPi-specific by design. 

There is however a small pin mapping problem, because we use in our device pins 19, 20, and 26 for PWM outputs, and as far as I know only pin 19 is available for RPi OS hardware PWM. That is why we used pigpio, because it exposes hardware PWM on all GPIO (at least that is stated in the description).

I2C should work out of the box, it is connected to the I2C pins.

Lucas

rumengb, 2024-09-29:
Sorry for the late reply, I was very busy with other stuff and kind of forgot about this. Excuse me!
There are only 2 PWMs on a Rpi:
PWM0: Can be mapped to GPIO 12, GPIO 18, GPIO 40, and GPIO 52.
PWM1: Can be mapped to GPIO 13, GPIO 19, GPIO 41, and GPIO 45.

Any other PWM source is a software PWM and therefore CPU intensive.

So what I can do is add all you need I can make the software PWM. but I am not sure I can make the motor control without a description of the signals. 

Can you please provide a description of everything you need and on which GPIO.

wannabecoffeenerd, 2024-10-10:
Looks like here is the pin mapping for the astrolink4pi: https://github.com/astrojolo/astrolink4pi/blob/main/astrolink4pi.cpp#L45

@astrojolo might be able to provide more details aboout this.

wannabecoffeenerd, 2024-11-18:
@astrojolo Pinging on this issue to provide a description of the signals for motor control and other pin mapping details needed to implement the driver.

astrojolo, 2024-11-18:
From the linked file in the two posts above we have:

```
#define EN_PIN 15
#define M0_PIN 17
#define M1_PIN 18
#define M2_PIN 27
#define RST_PIN 22
#define STP_PIN 24
#define DIR_PIN 23
#define OUT1_PIN 5
#define OUT2_PIN 6
#define PWM1_PIN 26
#define PWM2_PIN 19
#define MOTOR_PWM 20
#define FAN_PIN 13
```

Pins RP4, RP5, and CHK_IN are for checking the AstroLink 4 Pi revision and Raspberry Pi version, but let us leave them free for now and assume we have the most recent AstroLink revision working with Raspberry Pi 5. The pin description is as follows:

- EN - motor enable pin
- M0, M1, M2 - motor microstepping resolution pin
- RST - motor reset pin
- STP - motor step pin
- DIR - motor direction pin
- OUT1, OUT2 - switchable 12V DC outputs pins
- PWM1, PWM2 - regulated PWM outputs for heating tapes
- MOTOR_PWM - PWM signal that controls stepper motor current. It should be current [mA]/20, so if the required current is 400mA, it should be 20 (255 is max). PWM frequency should be 1kHz minimum, 5kHz is optimal. 
- FAN - turns on the internal fan, when the CPU temp is over 65C

The stepper motor driver is DRV8255. 

Many thanks,
Lucas