indigo-astronomy/indigo #564: Slow slew cause ALERT on some mounts

DavidObfuscator, 2024-09-10:
During debugging some meridian flip issues on UMI17 mount discord, we face with a issue, when it doesnt work well.
Mount is spec'ed to 3degree/s, running OnStep. Mount has a speed-up, slow-down proces during the slew. So when a meridian flip is triggered, it turns around 180degree on single axis, which is exactly 60seconds, adding the warm start overhead, it will exceed the 60second limit in the Mount Agent.

Issue is the 60k times waiting for 1000us:
https://github.com/indigo-astronomy/indigo/blob/d80a92ecd267cd53994d5b1bfb38c9dff2482ca3/indigo_drivers/agent_mount/indigo_agent_mount.c#L739

It would be great if the timeout is increased or even better, monitor the coordinates and abort it, if the coordinates doesnt change for a while

rumengb, 2024-09-11:
I will fix the issue. probably we should allow 1degree/s and make the timeout 180s?

wannabecoffeenerd, 2024-10-10:
@rumengb In the short term, is there a way for me to incorporate this change into indigosky on my end? Maybe with a custom build of some indigo component or something?

wannabecoffeenerd, 2024-11-18:
@rumengb Looks like this was changed to 180s in the main branch now, does that mean this issue can be closed and this shouldn't be an issue with the latest version of indigo server?
https://github.com/indigo-astronomy/indigo/blob/master/indigo_drivers/agent_mount/indigo_agent_mount.c#L216

rumengb, 2024-11-18:
Yes, 180s should be enough for mounts running at speeds faster than 1degree/sec, so it should not cause problems.

closed 2024-11-19 by rumengb