indigo-astronomy/indigo #566: License clarification

artyom-beilis, 2024-09-29:
> Redistribution in source or binary form for fee is permitted only as part of another different product or as product derived from this software.

What does it mean? 

- If I use a indigo in a product/project - it seems I can take fee on it.
- If I modify a line and redistribute indigo I can take a fee

But it means that for example I put it on a CD and sell the CD via e-mail (not that it is needed) I can't send it unless I change a line and call it derived product.

Why there is a line? When can't I "sell it". It makes it incompatible with GPL software - but not clear from what reason? To prevent it being used in GPL projects?

artyom-beilis, 2024-09-29:
While it isn't a problem for a project I own since I can add an exception but if I take some other GPL code I may not be able to use it - since indigo does not fall under definition of system library

polakovic, 2024-09-29:
The point should be that you can sell product based on INDIGO or containing INDIGO, but not INDIGO itself without any added value.

We know that in the extreme interpretation the derived product is what you mentioned. But for me personally even 2 clause BSD license is acceptable :)

One of reasons why we started INDIGO development was toxic GPL license used for INDI. But I don't think our license will limit you in any way to link it together with GPL code.

artyom-beilis, 2024-09-29:
> But for me personally even 2 clause BSD license is acceptable :)

If you can change it to 2 clause BSD it would effect significantly the community - non-OSI approved and GPL incompatible license is a huge red-flag. We in OSS community read our licenses `:-)`

At first I thought it was just BSD like license till I read the details.

> But I don't think our license will limit you in any way to link it together with GPL code.

It is an issue and there is a wide spread consensus that it is a problem.

It is same case as with OpenSSL 4-Clause BSD license that is not compatible with GPL. It created lots of headache - but unlike indigo OpenSSL can be considered nowadays a system library that comes with OS (GPL has an exception for system libraries)

I myself develop libraries under MIT or Boost, sometime time ago I used LGPLv3 but now rarely. Only applications I develop under GPL.

> One of reasons why we started INDIGO development was toxic GPL license used for INDI.

I think Indi is LGPL which is way more permissive (ASIAir uses one)

artyom-beilis, 2024-09-29:
BTW I want this library is because I want to have Android port for it for [OpenLiveStacker](https://github.com/artyom-beilis/OpenLiveStacker). My problem with indi is that it is client-server architecture and running server on Android is quite a big problem. Also indi has huge amount of dependencies.

So indigo looked like perfect candidate for porting but the license is quite a red-flag.

polakovic, 2024-09-29:
I personally agree to change it to 2 clause BSD license if it will help, I just want to stay away from GPL/LGPL of any sort. But there are many other authors which have to agree too.

> I think Indi is LGPL which is way more permissive (ASIAir uses one)

INDI has GPL in the root folder :)

https://github.com/indilib/indi/blob/master/COPYING.GPL

Also read e.g. this :)

https://www.reddit.com/r/AskAstrophotography/comments/11e2jrc/zwos_asiair_imager_binary_appears_to_be/

artyom-beilis, 2024-09-29:
> I personally agree to change it to 2 clause BSD license if it will help, I just want to stay away from GPL/LGPL of any sort. But there are many other authors which have to agree too.

It would be fantastic! And I understand that contributors - copyright holders must agree.

Foreseeing it in one of my projects I developed under LGPLv3, I requested copyright transfer specifically since I wasn't sure if I'll keep using LGPLv3. And finally I switched to MIT license.

> INDI has GPL in the root folder :)

I think there is a mix - AFAIR client is LGPL but server GPL (but I may be wrong). So you can use it in proprietary code with relative ease as long as you play by the rules.

But anyway it is their right, and if ASIAir violates they need to stop it. They have very simple requirement. It isn't a big deal to handle LGPL properly - it only asks keep original sources open and user option to fix them.

License need to be respected, and if ZWO here behaves improperly it - gives every ground to be sued or requested to fix the situation. 

I also noticed some camera libraries do not link correctly to libusb. I wouldn't tell here which major vendor but it is easy to find it out. Bottom line whoever chooses the license has their own ideas on what is right. But license need to be respected.

polakovic, 2024-09-29:
> I also noticed some camera libraries do not link correctly to libusb. I wouldn't tell here which major vendor but it is easy to find it out. Bottom line whoever chooses the license has their own ideas on what is right.

Do you mean INDI or INDIGO?

artyom-beilis, 2024-09-29:
> Do you mean INDI or INDIGO?

No - some proprietary SDKs by vendors. Bot indi and indigo can link statically since they open-source.

rumengb, 2024-09-29:
Well, we changed the license to 2-clause BSD, which should be compatible with GPL. But strictly speaking, GPL versions are not compatible with themselves.  So let us stop this. We use some vendor-provided binary SDKs which god knows what they use. So if you are nitpicking you will find GPL violations everywhere, even in GPL licensed projects.

artyom-beilis, 2024-09-29:
Amazing! 

It would make things simpler - and I totally agree that GPL problematic when using proprietary libraries - and has its own inconsistencies.

It just makes things simpler.

Thanks you guys and I'll start work on porting (at least mount part) to Android!

closed 2024-09-29 by artyom-beilis