indigo-astronomy/indigo #567: Debayering issues with Fujifilm files

esox13, 2024-10-01:
Hello,

There is something I can't manage to fix in AIN. I use a Fuji DSLR (X-H2) and if I create FITS files they are in greyscale. If I import them in SIRIL I have this message (french translated by chatGPT, so the original message in english may differ) :

`
The Bayer pattern found in the header (GRBG) is different from the Bayer pattern set in the preferences (RGGB). The latter has been replaced.`

In fact I think Fuji Xtrans is a 6 pixels pattern, not 4. I purchased the  macOS version, indigo A1, I think it has the same issue.

rumengb, 2024-10-03:
The issue is fixed. It will be part of the next indigo release.

esox13, 2024-10-03:
Thanks. Will it also fix the flipped upside down issues in FITS file from FUJI/IndigoA1/AIN issue ?

I didn't notice it becaue I tried on pictures that are difficult to say they are upside down, at least I didn't pay attention to that, but I tried with a non astro image where orientation can be easily noticed and all FITS images from my Fuji camera are flipped upside down, not 180° rotation but upside down flipped. The RAF raw are fine but the FITS aren't. In SIRIL, ASTAP and FITS liberator. But FITS previw app opens it in the right orientation, but of it can't debayer it. I guess there is a explanation to that. I don't have other software to test them on mac or linux. ASTAP also opens RAF files upside down...

The problem is that if I flip the images (with a numpy script) and apply the BAYERPAT I gave you above, it doesn't debayer correctly again, the xtrans pattern also has to be flipped upside down...

Maybe I shouldn't worry about this flipped problem, as I told you I'm not beginner as a photographer but a beginner in astrophoto.

I wrote a small astropy/numpy script that can correct the 
[fits_fuji_bayerpat.py.zip](https://github.com/user-attachments/files/17244726/fits_fuji_bayerpat.py.zip)
FITS so they are opened and processed correctly in SIRIL. It may also be a SIRIL issue, but SIRIL has no problem processing FUJI RAF files. This script has a flipped xtrans matrix and make the FITS fils (apprently) processed correctly in SIRIL.

I join you the script.
[fits_fuji_bayerpat.py.zip](https://github.com/user-attachments/files/17244735/fits_fuji_bayerpat.py.zip)

polakovic, 2024-10-03:
I tried FITS in both PixInsight and FITS Preview and images are not flipped compared to RAF. So it looks like SIRIL and ASTAP issue (as we consider PixInsight as most reliable "standard").

polakovic, 2024-10-03:
Mystery solved, SIRIL doesn't use ROWORDER in FITS metadata.

rumengb, 2024-10-03:
I did some experiments and it looks like Siril does not observe the ROWORDER keyword.  and it assumes the images are always bottom-up. And we produce TOP-DOWN images and indicate this in the HEADER.

esox13, 2024-10-03:
Ok thanks for the answer. Understood, the funny thing is that the RAF are OK in SIRIL. I keep my script for SIRIL use, as it is the tool I'm using now.

Can't wait for the update ;)

rumengb, 2024-10-03:
It will take some time as we are preparing a big release. We are still
developing. Then we need to freeze and test everything which will take
several weeks.

esox13, 2024-10-04:
Hi !

Here is the answer of Vincent Hourdin one of SIRIL's developpers :

`that’s a funny thing to say because ROWORDER was created by the Siril team.

So that’s not RAW images if they are FITS, and if the acquisition software is not able to give the correct BAYERPAT, it’s not really our problem.

Yes Siril displays images bottom-up, like the FITS convention: [Siril - FAQ 2](https://siril.org/faq/#why-are-my-images-upside-down) . When plate solving the result image in the usual processing workflow, if it’s flipped it will be put back in the correct orientation.`

[https://siril.org/faq/#why-are-my-images-upside-down](url)

rumengb, 2024-10-04:
I'm sorry for my ignorance, I did not know that but I am glad for SIRIL developers :)
Please bear in mind, that half of the software threats ROWORDER in one way and the other half in another. And it looks like we are in the wrong camp, together with PixInsight and many others. Life is tough and we have to live with that... Sorry, we are not going to change that.

closed 2024-10-23 by rumengb