indigo-astronomy/indigo #568: Mount Nexstar over-shooting on GOTO and sometimes stopping in wrong place.

wgoltz, 2024-10-07:
I have observed Mount Nexstar GOTO's stopping in unexpected places. I have studied this using _Skychart_ (Carte du Ciel) program and its _"telescope->slew to chart center"_ function, and created an Indigo trace log showing the 4 telescope goto commands that were traced.

(for a starting position looking West in southern hemisphere, image aligned N top, W right)

Case 1, _moving to the top left_ : the telescope goes through the desired new position, but then overshoots in both **RA** and **Dec** before returning to desired position.

Case 2 _moving to top right_ : the telescope moves to the desired position, but then overshoots in **Dec** before returning to desired position.

Case 3 _moving to bottom right_ : the telescope moves to desired position with no overshoot.

Case 4 _moving to bottom left_ : the telescope moves to the desired position, but then overshoots in **RA**. It **does not** return to the correct position and **remains offset in RA**.

I have not seen any strange behaviour using the Mount Simulator driver using _Skychart_, so infer that this is a problem with the Mount Nexstar driver rather than with _Skychart_.

rumengb, 2024-10-09:
Hi,
Your log is partial and I do not see the mount model. What mount are you using? However, the log shows no driver issue. This overshooting behavior is normal. Some mounts (especially the budget ones) overshoot and then return to the desired position. The simulator does not do this. Please note that the driver does not control the mount motors. It is the mount firmware that does it. The driver only issues commands like "goto X Y" and waits for the response that the operation is complete or failed. How it would be executed is a responsibility of the firmware. 
When it comes to the last slew when the mount remains offset in RA, it is not a driver issue too. The log clearly shows that the driver asked the mount to go to 6.89211h but the mount decided to stop at 6.94316h and reported that the slew is complete. So it is likely a firmware issue.

Rumen

rumengb, 2024-10-09:
Closing as it is not a driver issue.

closed 2024-10-09 by rumengb

wgoltz, 2024-10-09:
Thankyou for that.

The mount is an Sky-Watcher AZ-EQ5 Pro.  I will report the problem to the manufacturer.