indigo-astronomy/indigo #572: Documentation - no support of AZ Mounts

artyom-beilis, 2024-10-16:
Reading this https://github.com/indigo-astronomy/indigo/blob/master/indigo_drivers/mount_synscan/README.md  I was sure that AZ GTi as is supported in its default Alt-Az mode.

Here also marked that NexStar SE is supported: https://github.com/indigo-astronomy/indigo/blob/master/indigo_drivers/mount_nexstar/README.md

Now reading [this message on the forum](https://bb.cloudmakers.eu/viewtopic.php?p=8869#p8869) and trying one it is clear that Alt-Az mounts are not really supported.

This is something that should be very explicit in the documentation of indigo and BTW indi/indigo comparison.

polakovic, 2024-10-16:
Actually it means, that Alt-Az mode is not supported.

Both AZ-GTi and SE can be used in Alt-Az or EQ mode (on wedge). If you use some of hand controllers for these mounts and NexStar driver, you can use them in Alt-Az mode too, because driver doesn't really know the physical state. But if you don't use hand controller and NexStarAUX or SynScan driver, whole math is in the drivers and they don't support Alt-Az mode.

INDIGO itself doesn't impose any limits on it, but nobody implemented this feature as the focus is on astrophotographers.

artyom-beilis, 2024-10-16:
> INDIGO itself doesn't impose any limits on it, but nobody implemented this feature as the focus is on astrophotographers.

Thanks, I understand that you have your priorities. I just wish it was mentioned in the documentation explicitly and not located in a forum after extensive search.

If I knew it in first place I wouldn't start working with indigo for OpenLiveStacker, since I concentrate on EAA and alt-az are mounts perfectly fine for this task and these are ones I actually use most of the time.

Thanks again for your work and being very responsive. Just please update the docs.

rumengb, 2024-10-16:
> If I knew it in first place I wouldn't start working with indigo for OpenLiveStacker, since I concentrate on EAA and alt-az are mounts perfectly fine for this task and these are ones I actually use most of the time.

I am sorry, live is tough sometimes :) And you are welcome to implement the alt-az mode. Indigo is free and opensource. And the beauty of open source is that if you miss something you can implement it yourself :). There are exactly two drivers that do not support Alt-Az mode - NexStarAUX and SynScan. Those drivers use low-level motor control drivers and basically implement the mount firmware in themselves. Synscan is, by far, the most complex mount driver in INDIGO for this reason. And honestly, there is little to no demand for alt-az support.

And yes I agree we do not have the best documentation ever but if you find some documentation flaw as (I said) INDIGO is open source - correct it and make a PR :)

artyom-beilis, 2024-10-16:
> And the beauty of open source is that if you miss something you can implement it yourself :).

I know. I myself FOSS developer and tell it to my users frequently `;-)` - just it is little bit too much for me at this point. I'll try to port indi to Android fist `:-)`. Indigo has much better architecture not forcing me to use multiple process and thus is much simpler to port to Android. But ironically I have two mounts AZ GTi I use elusively in at Alt Az mode and NexStart SE8 with broken motor bearing (I may be will fix some day)

So I'll see what I can do. And it is amazing to work with you. You are one of most responsive FOSS developers I have ever seen.

Thanks Again
Artyom