indigo-astronomy/indigo #573: Indigo A1 / EQ6-R Pro USB connexion doesn't connect

esox13, 2024-10-18:
Hello,

On Indigo A1 (on Mac Sonomo 14.4.1), using Synscan driver and connectin the EQ6-R pro via the USB socket of the mount, the mount seems to connect to what is supposed to be the Bluetooth port of the MB Pro (Intel). When I save the log, then Indigo A1 crashes.

In the serial port section I have no other choice than `/dev/cu.BLTH`

`I 17:31:54.91	macOS 14.4.1 HW MacBookPro16,1 / x86_64 GPU AMD Radeon Pro 5300M / F8FFC26AC23B
I 17:31:54.91	INDIGO 2.0_295 (2024-09-03T17:18:31+02:00 0cfdb67d5)
I 17:31:54.91	INDIGO A1 5.5_863 (2024-09-03T20:21:40+02:00 f20d373+)
I 17:31:55.04	Client started
V 17:31:55.04	Driver indigo_agent_astrometry 2.0.0.16 loaded
D 17:31:55.04	indigo_platesolver_validate_executable: /Applications/INDIGO A1.app/Contents/MacOS/solve-field
D 17:31:55.04	indigo_platesolver_validate_executable: /Applications/INDIGO A1.app/Contents/MacOS/image2xy
D 17:31:55.04	indigo_platesolver_validate_executable: /usr/bin/curl
V 17:31:55.04	indigo_agent_astrometry: 'Astrometry Agent' attached
I 17:31:56.07	Service browser started
D 17:31:56.07	Service browser started
V 17:31:56.20	Driver indigo_agent_auxiliary 2.0.0.3 loaded
V 17:31:56.21	indigo_agent_auxiliary: 'Auxiliary Agent' attached
V 17:31:56.28	Driver indigo_agent_config 2.0.0.5 loaded
V 17:31:56.28	indigo_agent_config: 'Configuration Agent' attached
V 17:31:56.35	Driver indigo_agent_guider 2.0.0.35 loaded
V 17:31:56.37	indigo_agent_guider: 'Guider Agent' attached
V 17:31:56.46	Driver indigo_agent_imager 2.0.0.44 loaded
V 17:31:56.53	indigo_agent_imager: 'Imager Agent' attached
V 17:31:56.57	Driver indigo_agent_mount 2.0.0.17 loaded
V 17:31:56.60	indigo_agent_mount: 'Mount Agent' attached
V 17:31:56.70	Driver indigo_agent_scripting 2.0.0.8 loaded
V 17:31:56.74	Hight level scripting API installed
D 17:31:56.74	indigo_agent_scripting[agent_device_attach:929]: boot.js executed
V 17:31:56.74	indigo_agent_scripting: 'Scripting Agent' attached
I 17:31:56.81	Executing on-load scripts
V 17:31:56.86	Driver indigo_mount_synscan 2.0.0.18 loaded
V 17:31:56.86	indigo_mount_synscan: 'Mount SynScan' attached
V 17:31:56.92	indigo_mount_synscan: 'Mount SynScan (guider)' attached
D 17:31:57.04	Internal server device attached
I 17:31:57.04	Server started
V 17:31:57.04	Server started on port 49465
V 17:31:57.04	Service registered as INDIGO A1
V 17:32:15.72	indigo_mount_synscan: Connected to /dev/cu.BLTH @ 115200-8N1
D 17:32:15.73	indigo_mount_synscan[synscan_command_unlocked:124]: CMD: [:e1]
E 17:32:20.73	indigo_mount_synscan[synscan_read_response:154, 0x70000c9a2000]: SYNSCAN_TIMEOUT
E 17:32:20.73	indigo_mount_synscan[synscan_read_response:165, 0x70000c9a2000]: Reading response failed
D 17:32:20.75	indigo_mount_synscan[synscan_command_unlocked:124]: CMD: [:e1]
E 17:32:25.74	indigo_mount_synscan[synscan_read_response:154, 0x70000c9a2000]: SYNSCAN_TIMEOUT
E 17:32:25.76	indigo_mount_synscan[synscan_read_response:165, 0x70000c9a2000]: Reading response failed
E 17:32:25.76	indigo_mount_synscan[synscan_configure:109, 0x70000c9a2000]: ERROR GETTING FIRMWARE
V 17:32:25.80	indigo_mount_synscan: Disconnected from /dev/cu.BLTH
V 17:32:26.05	indigo_mount_synscan: Connected to /dev/cu.BLTH @ 9600-8N1
D 17:32:26.07	indigo_mount_synscan[synscan_command_unlocked:124]: CMD: [:e1]
E 17:32:31.06	indigo_mount_synscan[synscan_read_response:154, 0x70000c9a2000]: SYNSCAN_TIMEOUT
E 17:32:31.06	indigo_mount_synscan[synscan_read_response:165, 0x70000c9a2000]: Reading response failed
D 17:32:31.08	indigo_mount_synscan[synscan_command_unlocked:124]: CMD: [:e1]
E 17:32:36.08	indigo_mount_synscan[synscan_read_response:154, 0x70000c9a2000]: SYNSCAN_TIMEOUT
E 17:32:36.08	indigo_mount_synscan[synscan_read_response:165, 0x70000c9a2000]: Reading response failed
E 17:32:36.10	indigo_mount_synscan[synscan_configure:109, 0x70000c9a2000]: ERROR GETTING FIRMWARE
V 17:32:36.14	indigo_mount_synscan: Disconnected from /dev/cu.BLTH
`
If I use the special cable from hand controler to USB (not using the hand controler it self but the hand control socket on the mount), then it connects, if I use this port : `/dev/cu.usbserial-AQ02XQRF`

It first tries to connect @115200-8N1, which returns the Timeout / Error getting firmware error, then tries to connect @ 9600-8N1 and the mount seems to work (didn't try it on a real image capture session yet).

On AIN Linux I can connect with both USB/Handcontroler mount socket.

polakovic, 2024-10-19:
Sorry, I'm not sure I understand. Did you select /dev/cu.usbserial-AQ02XQRF as serial port for SynScan driver?

For SynScan driver we can't use auto-detection, so you have to set the correct port manually.

esox13, 2024-10-19:
With the cable that fists in the handcontroler on the mount and has a USB plug at the other hand, wgen 8 did an autodetect in indigo A1 / ports, /dev/cu.usbserial-AQ02XQRF came out and I seleted it.

What is the procedure to find the right port ?

polakovic, 2024-10-20:
There is refresh item in the port list. So try it with and without USB-to-serial adapter plugged in and the difference in the list is the right one,

Alternatively you can try terminal command

`ls /dev/cu.*`

esox13, 2024-10-20:
If I use the EQMOD cable and BT turned on then :

`ls /dev/cu.*
/dev/cu.BLTH
/dev/cu.usbserial-AQ02XQRF`

If I use the EQMOD cable and turn BT off :

`ls /dev/cu.*
/dev/cu.BLTH	
/dev/cu.Bluetooth-Incoming-Port	
/dev/cu.usbserial-AQ02XQRF`

If I don't use the EQMOD cable but a USB2 cable from the USB socket of the EQ6-R mount to the MacBook Pro :

then :

`ioreg -p IOUSB | grep -w "class AppleUSBDevice"

  | +-o FaceTime HD Camera (Built-in)@80200000  <class AppleUSBDevice, id 0x10000054c, registered, matched, active, busy 0 (0 ms), retain 13>
  | +-o Touch Bar Backlight@80700000  <class AppleUSBDevice, id 0x100000550, registered, matched, active, busy 0 (0 ms), retain 13>
  | +-o Ambient Light Sensor@80300000  <class AppleUSBDevice, id 0x100000555, registered, matched, active, busy 0 (0 ms), retain 11>
  | +-o IOUSBHostDevice@80800000  <class AppleUSBDevice, id 0x10000055f, registered, matched, active, busy 0 (0 ms), retain 14>
  | +-o Apple T2 Controller@80100000  <class AppleUSBDevice, id 0x100000563, registered, matched, active, busy 0 (2 ms), retain 13>
  | +-o Headset@80400000  <class AppleUSBDevice, id 0x10000056a, registered, matched, active, busy 0 (0 ms), retain 11>
  | +-o Touch Bar Display@80600000  <class AppleUSBDevice, id 0x100000576, registered, matched, active, busy 0 (0 ms), retain 13>
  | +-o Apple Internal Keyboard / Trackpad@80500000  <class AppleUSBDevice, id 0x10000569d, registered, matched, active, busy 0 (2 ms), retain 19>
    +-o USB2.1 Hub@14400000  <class AppleUSBDevice, id 0x1000057f6, registered, matched, active, busy 0 (0 ms), retain 15>
      +-o USB2.1 Hub@14410000  <class AppleUSBDevice, id 0x100005809, registered, matched, active, busy 0 (0 ms), retain 16>
        +-o ASI220MM Mini@14411000  <class AppleUSBDevice, id 0x100005826, registered, matched, active, busy 0 (1 ms), retain 14>
        +-o USB-Serial Controller @14412000  <class AppleUSBDevice, id 0x100005876, registered, matched, active, busy 0 (1 ms), retain 14>`

This item :

`+-o USB-Serial Controller @14412000  <class AppleUSBDevice, id 0x100005876, registered, matched, active, busy 0 (1 ms), retain 14>`

Appears in the list if I plug in the EQ6-R from the mount USB socket to the computer.

On linux :

`lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 003: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 003 Device 005: ID 8087:0aaa Intel Corp. Bluetooth 9460/9560 Jefferson Peak (JfP)
Bus 003 Device 009: ID 0bda:5411 Realtek Semiconductor Corp. RTS5411 Hub
Bus 003 Device 010: ID 0bda:5411 Realtek Semiconductor Corp. RTS5411 Hub
Bus 003 Device 011: ID 03c3:2209 ZWO ASI220MM Mini
Bus 003 Device 012: ID 067b:23a3 Prolific Technology, Inc. ATEN Serial Bridge
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 004 Device 004: ID 0bda:0411 Realtek Semiconductor Corp. Hub
Bus 004 Device 005: ID 0bda:0411 Realtek Semiconductor Corp. Hub
`

This item is the one that appears if I connect the mount and the computer using the USB socket of the mount to the computer :

`Bus 003 Device 012: ID 067b:23a3 Prolific Technology, Inc. ATEN Serial Bridge`

And it works perfectly fine with indigo_server and AIN.

Do that mean that there is no way on Indigo A1 on macOS  to connect the mount and the computer via a simple USB2 cable as it is ppossible on Linux ? Only the EQMOD cable works ? Or do I maybe have to choose another driver than the syncon one ?

polakovic, 2024-10-20:
> | +-o FaceTime HD Camera ***@***.*** <class AppleUSBDevice, id
> | +-o Touch Bar ***@***.*** <class AppleUSBDevice, id 0x100000550,
> | +-o Ambient Light ***@***.*** <class AppleUSBDevice, id
> | +-o ***@***.*** <class AppleUSBDevice, id 0x10000055f,
> | +-o Apple T2 ***@***.*** <class AppleUSBDevice, id 0x100000563,
> | +-o ***@***.*** <class AppleUSBDevice, id 0x10000056a, registered,
> | +-o Touch Bar ***@***.*** <class AppleUSBDevice, id 0x100000576,
> | +-o Apple Internal Keyboard / ***@***.*** <class AppleUSBDevice,
> +-o USB2.1 ***@***.*** <class AppleUSBDevice, id 0x1000057f6,
> +-o USB2.1 ***@***.*** <class AppleUSBDevice, id 0x100005809,
> +-o ASI220MM ***@***.*** <class AppleUSBDevice, id 0x100005826,

If the serial port doesn't appear when the mount is plugged in (USB cable
to USB socket on mount head), you may need to install an operating system
level driver:
https://www.prolific.com.tw/US/ShowProduct.aspx?p_id=229&pcid=41

esox13, 2024-10-20:
Thanks. In fact on the prolific link the version are outdated (Big Sur). But the PL2303 driver can be installed via AppStore. And the in Indigo A1 I can select /dev/cu.PL2303G-USBtoUART1430 and the mount seems to be active. Didn't check yet in real situation but it loads and moves.

There is also a /dev/cu.Plser available but seems to freeze/crash Indigo A1.

Thanks !

polakovic, 2024-10-21:
Pls. can you send me such crash log? 

Opening wrong device can be generally dangerous, but more for the device itself, A1 shouldn't crash :(

esox13, 2024-10-22:
It is complicated as if the app crashes or freezes I can't save logs. I just had a new "instant shutdown" crash when using the direction arrows in the imager agent when mount activated. Then Indigo A1 can't be launched anymore and is still on the "force to quit" list with no possibility to force quit. I guess only a reboot of the MBpro can clean everything. I'm using an Intel MBpro, not a M.

esox13, 2024-10-22:
I also noticed this red warning in front of the imager tab.
[image]

polakovic, 2026-10-02:
Resolved by installing the Prolific PL2303 system driver; the mount then connects through the /dev/cu.PL2303G-* port. Closing.

closed 2026-10-02 by polakovic