indigo-astronomy/indigo #576: Fujifilm cameras disconnect with new version

esox13, 2024-10-27:
Everything was working fine with the previous version with my Fujifilm X-H2 camera. But since today's update, the camera keeps disconnecting at each command sent t the camera (iso change, preview...).

I updated via apt update/upgrade

Current version :  framework  2.0-296, AIN 1.7.0

`15:42:19.027 Server @ astrobuntu.LOAD: Driver indigo_agent_imager (Imager Agent) loaded
15:42:19.075 Server @ astrobuntu.LOAD: Driver indigo_agent_guider (Guider Agent) loaded
15:42:19.116 Server @ astrobuntu.LOAD: Driver indigo_agent_mount (Mount Agent) loaded
15:42:19.178 Server @ astrobuntu.LOAD: Driver indigo_agent_astrometry (Astrometry Agent) loaded
15:42:19.186 Server @ astrobuntu.LOAD: Driver indigo_agent_config (Configuration Agent) loaded
15:42:51.879 Server @ astrobuntu.LOAD: Driver indigo_agent_imager (Imager Agent) loaded
15:42:51.925 Server @ astrobuntu.LOAD: Driver indigo_agent_guider (Guider Agent) loaded
15:42:51.991 Server @ astrobuntu.LOAD: Driver indigo_agent_mount (Mount Agent) loaded
15:42:52.055 Server @ astrobuntu.LOAD: Driver indigo_agent_astrometry (Astrometry Agent) loaded
15:42:52.065 Server @ astrobuntu.LOAD: Driver indigo_agent_config (Configuration Agent) loaded`

Also, can't launch control panel from AIN File menu

`16:46:15.684 Error: INDIGO Control Panel could not be started. Is it in your path?
16:46:42.504 Error: INDIGO Control Panel could not be started. Is it in your `path?``

Edit :

 I finally found a 2.0-294 deb version. I purged everything, reinstalled per instruction, same issue. Then I purged everything again, erased the repo, installed the 2.0-294 version, (with the new versions of ain and control panel) then it worked fine again. I reinstalled the repo, made the update to 296 and then things are OK again. 

But Fuji debayering is still off for fits files. Maybe worse than it was. More annoying, on focus panel, when using focusing option (meaning looping previews) even if native is selected, the image displayed for focusing is badly debayered. Making the function unusable.  Preview option works fine if native is selected but focus option is badly debayered.

I didn't reveive my electronic focuser yet, but would that mean that electronic focusing will be impossible with Fuji cameras ?

Edit 2 : the disconnexion happened again when I unplugged the camera to zoom focus on the camera display, then replugged the camera and reselected it in the drop down list on AIN. When you do that with the 294 version of indigo server everything is fine, with the 296 version you can't reconnect the camera even after stopping and restarting the server. So for the moment I keep 294 version of the server.

But with the new 296 Fuji debayer pattern the colors seem to be correctly displayed in SIRIL and Kstars viewer.

[ain_indigo_2024-10-27.log.zip](https://github.com/user-attachments/files/17534906/ain_indigo_2024-10-27.log.zip)

esox13, 2024-10-27:
Thanks. I'm on x86 64.

esox13, 2024-10-29:
> try this build

Hello and thanks.

I installed it, the disconnection problem is solved.

I still have some issues with the linux version (didn’t really test the Mac version yet) :

I started a 500 images batch (with 2 sec delay between images). The AIN-Imager shut down after 100+ images (I wasn’t in front of the computer so I can’t tell exactly when it happened, but it seem to be always happening when I test this 500 images batch). It was between 100 and 200 images. But he camera was still shooting. I relaunched AIN-Imager and it started with everything connected (didn’t have to reselect the Camera in the drop down list) and the message window displayed the correct image count. Then it stopped definitively after 273 images (I have 273 images recorded on the external nvme drive). See attached log file.
The FITS files from my Fuji X-H2 camera are now correctly debayered by SIRIL but are not correctly displayed in the preview window of AIN-Imager. In the focus tab it is even more unusual : images are not displayed the same way if you do a single preview or a loop one.

The setup is : MiniPC Mele quieter 4C, Ubuntu 24.04.1 LTS, Microsoft RDP from a MacBook Pro Intel 2019 on the Linux MiniPC (xfce session).

Regards,

Jean-Martin

￼

rumengb, 2024-10-29:
The images from Fuji X-H2 will not be displayed correctly. as the pattern of thee pixels is quite unusual. and I guess this pixel arrangement is the root cause of the crash you observe in ain. it may be related to libraw. but I can not be sure without any logs.

polakovic, 2026-10-02:
The disconnect issue was fixed in 627c7a6df and confirmed by the reporter. The X-Trans preview rendering is a client-side (AIN) limitation and is not tracked here. Closing.

closed 2026-10-02 by polakovic