indigo-astronomy/indigo #577: Select TIFF image format causes Segmentation fault in both Preview and Expose

chtay, 2024-10-29:
Hi
I tried 2.0-294 and 2.0-296, but will produce Segmentation fault (core dumped) if select TIFF image format, and then click preview or expose in Ain INDIGO imager and as well as in webGUI. It happen on CCD Image Simulation and as well as SVBony SV305 Pro camera (I didn't try other camera).

The version I used is for Linux, running on openSUSE Leap 15.6. I tried compiled it (successful, no error) from source code, and as well as from the deb package. It has no problem on FITS and JPEG.

Here is the last few lines, right after I pressed Preview:
06:51:43.643492 indigo_server: B <+ Update 'Imager Agent'.'CCD_EXPOSURE' NUMBER rw Busy 2.0 0 {
06:51:43.643575 indigo_server: B <+   'EXPOSURE' = 1, target = 1 
06:51:43.643651 indigo_server: B <- }
06:51:43.643799 indigo_server: 18 <- <setNumberVector device='Imager Agent' name='CCD_EXPOSURE' state='Busy'>
06:51:43.643907 indigo_server: 18 <- <oneNumber name='EXPOSURE' target='1'>1</oneNumber>
06:51:43.643997 indigo_server: 18 <- </setNumberVector>
06:51:43.644138 indigo_server: 18 <- <setNumberVector device='CCD Imager Simulator' name='CCD_EXPOSURE' state='Busy'>
06:51:43.644242 indigo_server: 18 <- <oneNumber name='EXPOSURE' target='1'>1</oneNumber>
06:51:43.644331 indigo_server: 18 <- </setNumberVector>
06:51:43.644428 indigo_server: B <+ Update 'CCD Imager Simulator'.'CCD_IMAGE' BLOB ro Busy 2.0 0 {
06:51:43.644511 indigo_server: B <+   'IMAGE' (3844800 bytes, '.fits', '')
06:51:43.644585 indigo_server: B <- }
06:51:43.644694 indigo_server: B <+ Update 'Imager Agent'.'CCD_IMAGE' BLOB ro Busy 2.0 0 {
06:51:43.644783 indigo_server: B <+   'IMAGE' (3844800 bytes, '.fits', '')
06:51:43.644859 indigo_server: B <- }
06:51:43.645017 indigo_server: 18 <- <setBLOBVector device='Imager Agent' name='CCD_IMAGE' state='Busy'>
06:51:43.645119 indigo_server: 18 <- </setBLOBVector>
06:51:43.645277 indigo_server: timer #7 - reusing (0x7f5e7c104ab0)
06:51:43.645426 indigo_server: timer #7 - sleep for 1s (0x7f5e9e5ff8f0)
06:51:44.415452 indigo_server: B <+ Update 'CCD Imager Simulator'.'CCD_EXPOSURE' NUMBER rw Busy 2.0 0 {
06:51:44.415600 indigo_server: B <+   'EXPOSURE' = 0, target = 1 
06:51:44.415678 indigo_server: B <- }
06:51:44.415782 indigo_server: B <+ Update 'Imager Agent'.'CCD_EXPOSURE' NUMBER rw Busy 2.0 0 {
06:51:44.415849 indigo_server: B <+   'EXPOSURE' = 0, target = 1 
06:51:44.415901 indigo_server: B <- }
06:51:44.416100 indigo_server: 18 <- <setNumberVector device='Imager Agent' name='CCD_EXPOSURE' state='Busy'>
06:51:44.416345 indigo_server: 18 <- <oneNumber name='EXPOSURE' target='1'>0</oneNumber>
06:51:44.416444 indigo_server: 18 <- </setNumberVector>
06:51:44.416803 indigo_server: 18 <- <setNumberVector device='CCD Imager Simulator' name='CCD_EXPOSURE' state='Busy'>
06:51:44.416948 indigo_server: 18 <- <oneNumber name='EXPOSURE' target='1'>0</oneNumber>
06:51:44.417099 indigo_server: 18 <- </setNumberVector>
06:51:44.645613 indigo_server: timer #7 - callback 0x7f5e9e609240 started (0x7f5e9e5ff8f0)
Segmentation fault (core dumped)

rumengb, 2024-10-30:
> 06:51:43.643907 indigo_server: 18 <- 1
> 06:51:43.644242 indigo_server: 18 <- 1
> 06:51:44.416345 indigo_server: 18 <- 0
> 06:51:44.416948 indigo_server: 18 <- 0

it would be helpful if you run it through Valgrind something like:

valgrind indigo_server [other options] >valgrind.log 2>&1

and send us the whole output

chtay, 2024-10-30:
I tried the valgrind from another Linux system (openSUSE Leap 15.5) and same Segmentation fault.
Using this:
valgrind indigo_server -- >valgrind.log 2>&1

Here is the log output:
==32571== Memcheck, a memory error detector
==32571== Copyright (C) 2002-2022, and GNU GPL'd, by Julian Seward et al.
==32571== Using Valgrind-3.20.0 and LibVEX; rerun with -h for copyright info
==32571== Command: indigo_server --
==32571== 
==32571== Thread 20:
==32571== Invalid read of size 1
==32571==    at 0x483E902: strlen (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==32571==    by 0x589DA6E: __vfprintf_internal (in /lib64/libc-2.31.so)
==32571==    by 0x58B0C4E: __vsnprintf_internal (in /lib64/libc-2.31.so)
==32571==    by 0x4EE5A14: add_key.constprop.6 (indigo_ccd_driver.c:1160)
==32571==    by 0x4EEDB79: raw_to_tiff (indigo_ccd_driver.c:1238)
==32571==    by 0x4EEDB79: indigo_process_image (indigo_ccd_driver.c:1975)
==32571==    by 0x822DDE2: create_frame (indigo_ccd_simulator.c:588)
==32571==    by 0x822F2B8: exposure_timer_callback (indigo_ccd_simulator.c:597)
==32571==    by 0x4EA8BBB: timer_func (indigo_timer.c:89)
==32571==    by 0x58076E9: start_thread (in /lib64/libpthread-2.31.so)
==32571==    by 0x593958E: clone (in /lib64/libc-2.31.so)
==32571==  Address 0x2 is not stack'd, malloc'd or (recently) free'd
==32571== 
==32571== 
==32571== Process terminating with default action of signal 11 (SIGSEGV): dumping core
==32571==  Access not within mapped region at address 0x2
==32571==    at 0x483E902: strlen (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==32571==    by 0x589DA6E: __vfprintf_internal (in /lib64/libc-2.31.so)
==32571==    by 0x58B0C4E: __vsnprintf_internal (in /lib64/libc-2.31.so)
==32571==    by 0x4EE5A14: add_key.constprop.6 (indigo_ccd_driver.c:1160)
==32571==    by 0x4EEDB79: raw_to_tiff (indigo_ccd_driver.c:1238)
==32571==    by 0x4EEDB79: indigo_process_image (indigo_ccd_driver.c:1975)
==32571==    by 0x822DDE2: create_frame (indigo_ccd_simulator.c:588)
==32571==    by 0x822F2B8: exposure_timer_callback (indigo_ccd_simulator.c:597)
==32571==    by 0x4EA8BBB: timer_func (indigo_timer.c:89)
==32571==    by 0x58076E9: start_thread (in /lib64/libpthread-2.31.so)
==32571==    by 0x593958E: clone (in /lib64/libc-2.31.so)
==32571==  If you believe this happened as a result of a stack
==32571==  overflow in your program's main thread (unlikely but
==32571==  possible), you can try to increase the size of the
==32571==  main thread stack using the --main-stacksize= flag.
==32571==  The main thread stack size used in this run was 8388608.
==32571== 
==32571== HEAP SUMMARY:
==32571==     in use at exit: 55,979,071 bytes in 1,156 blocks
==32571==   total heap usage: 103,898 allocs, 102,742 frees, 1,012,893,287 bytes allocated
==32571== 
==32571== LEAK SUMMARY:
==32571==    definitely lost: 0 bytes in 0 blocks
==32571==    indirectly lost: 0 bytes in 0 blocks
==32571==      possibly lost: 13,023 bytes in 73 blocks
==32571==    still reachable: 55,966,048 bytes in 1,083 blocks
==32571==         suppressed: 0 bytes in 0 blocks
==32571== Rerun with --leak-check=full to see details of leaked memory
==32571== 
==32571== For lists of detected and suppressed errors, rerun with: -s
==32571== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)

chtay, 2024-10-30:
Update:
I download and compile 2.0-290 version and it works for TIFF.
Now you got a working version to compare with.

rumengb, 2024-11-09:
fixed in 2.0-298

closed 2024-11-09 by rumengb