indigo-astronomy/indigo #583: SIGPIPE in `indigo_write` during Fuji X-T5 preview with `indigo_ccd_ptp`

evanmayer, 2024-12-06:
### Summary

While running the `indigo_ccd_ptp` agent to control a Fujifilm X-T5, attempting to take a preview frame causes `ain_imager` lose connection to the server and reconnect. No preview frame is taken.

I can connect to the camera from the Indigo control panel and browse/change settings, and I can select it from the dropdown in `ain_imager`, so I think the ptp connection is ok.

* Camera: Fujifilm X-T5
  * Wired tether shooting fixed mode
  * ISO dial: C mode
  * Shutter speed dial: B mode
* OS: Ubuntu 22.04

### Steps to Reproduce

1. Start `indigo_server`: `indigo_server indigo_ccd_ptp`
2. Plug in camera
3. Turn on camera
4. Start `ain_imager`: `ain_imager`
5. Capture tab > Select "Fujifilm Camera"
6. Click preview
7. Preview button turns yellowish
8. Dialog below reads "Preview: 0 of 1 seconds elapsed..."
9. All items disappear from dropdowns: imager agent, camera, wheel
10. Inside control panel, all drivers disappear, then reload themselves
11. `ain_imager` window appears identical to step 4.

### Expected Behavior

Shutter fires for 1s, file transfer occurs, image appears in preview window.

### Details

[STDOUT and STDERR during the bug (github gist)](https://gist.github.com/evanmayer/c14d1a00bd8bb4026232cb578c2953dc)

`gdb` backtrace during SIGPIPE:

```
19:55:33.621435 ain_imager: [REMOVE REMOVE] .

Thread 21 "Thread (pooled)" received signal SIGPIPE, Broken pipe.
[Switching to Thread 0x7fffc6000640 (LWP 100231)]
__GI___libc_write (nbytes=44, buf=0x7fff78101f50, fd=37) at ../sysdeps/unix/sysv/linux/write.c:26
26	../sysdeps/unix/sysv/linux/write.c: No such file or directory.
(gdb) bt
#0  __GI___libc_write (nbytes=44, buf=0x7fff78101f50, fd=37)
    at ../sysdeps/unix/sysv/linux/write.c:26
indigo-astronomy/indigo_imager#1  __GI___libc_write
    (fd=fd@entry=37, buf=buf@entry=0x7fff78101f50, nbytes=nbytes@entry=44)
    at ../sysdeps/unix/sysv/linux/write.c:24
indigo-astronomy/indigo_imager#2  0x00007ffff784dee9 in indigo_write
    (handle=37, buffer=0x7fff78101f50 "<getProperties version='1.7' switch='2.0'/>\n", length=<optimized out>) at indigo_io.c:456
indigo-astronomy/indigo_imager#3  0x00007ffff784e00e in indigo_printf
    (handle=handle@entry=37, format=format@entry=0x7ffff79c2220 "<getProperties version='1.7' switch='%d.%d'/>\n") at indigo_io.c:477
indigo-astronomy/indigo_imager#4  0x00007ffff786da40 in xml_client_parser_enumerate_properties
    (device=0x7fff78000f80, client=<optimized out>, property=0x7ffff7ab23e0 <INDIGO_ALL_PROPERTIES>) at indigo_client_xml.c:98
indigo-astronomy/indigo_imager#5  0x00007ffff7841d3c in indigo_enumerate_properties
    (client=0x0, property=0x7ffff7ab23e0 <INDIGO_ALL_PROPERTIES>)
    at indigo_bus.c:542
indigo-astronomy/indigo_imager#6  0x000055555564f985 in  ()
indigo-astronomy/indigo#425  0x00007ffff62cff92 in  () at /lib/x86_64-linux-gnu/libQt5Core.so.5
indigo-astronomy/indigo#430  0x00007ffff62ccca1 in  () at /lib/x86_64-linux-gnu/libQt5Core.so.5
indigo-astronomy/indigo_imager#9  0x00007ffff5e94ac3 in start_thread (arg=<optimized out>)
    at ./nptl/pthread_create.c:442
indigo-astronomy/indigo_imager#10 0x00007ffff5f26850 in clone3 ()
--Type <RET> for more, q to quit, c to continue without paging--
    at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81
(gdb) 
```

SIGPIPE might be a red herring...backtrace seems to suggest a file is missing, or maybe my OS is yoinking a file `ain_imager` thinks should exist. What can I do to help get the expected behavior?

polakovic, 2024-12-06:
It shouldn't crash (I'll look at it), but it will not work with these settings. Don't use B as shutter time and I'm not sure about C as ISO. LiveView is a very limited (= useless) feature on most cameras and this one is not an exception. You usually have no control over any of camera settings and the resolution is very low (for camera display).

evanmayer, 2024-12-06:
@rumengb Thanks for moving this to the proper place.

@polakovic Just to make sure I was being clear, B for Bulb mode and C for Camera mode refer to the positions of the physical dials on the camera, an attempt to make sure there's nothing limiting the setting of the exposure by the PC, which is in control according to my selection in Indigo Control Panel. Although with the PC in control, in principle the physical dial positions shouldn't matter, I suppose. I have no problems setting the proper ISO or shutter speed for the camera using the Indigo Control Panel when the PC is in control, and I see the proper settings updated when the camera is in control.

If there's stuff about LiveView in the output files I sent, that was unintentional, I don't want to be using that feature. I want to press the Preview button in `ain_imager` that commands the camera to fire the shutter and take a normal image using the applied settings, but not part of a sequence or autofocus run. This, the most basic test, is what I am trying to run, before moving on to sequencing and autofocusing (I have a Moonlite-compatible focuser, but let's set that aside for now.)

Thanks for the speedy replies! I'll get that gdb/valgrind output tonight. I'm familiar with both and it should be no problem to get that for you.

evanmayer, 2024-12-07:
From GDB, looks like a segfault:

```
Thread 16 "indigo_server" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffeea00640 (LWP 6406)]
0x00007ffff772247a in ptp_set_switch_property (value=4, property=0x7fffef62b050, device=0x7fffe8000b70) at indigo_ptp_fuji.c:546
546	indigo_ptp_fuji.c: No such file or directory.
(gdb) bt
#0  0x00007ffff772247a in ptp_set_switch_property (value=4, 
    property=0x7fffef62b050, device=0x7fffe8000b70) at indigo_ptp_fuji.c:546
#1  ptp_fuji_exposure (device=0x7fffe8000b70) at indigo_ptp_fuji.c:596
#2  0x00007ffff7706ac7 in handle_exposure (device=0x7fffe8000b70)
    at indigo_ccd_ptp.c:343
#3  0x00007ffff7859124 in timer_func (timer=0x7fffe98281a0)
    at indigo_timer.c:89
#4  0x00007ffff6c94ac3 in start_thread (arg=<optimized out>)
    at ./nptl/pthread_create.c:442
#5  0x00007ffff6d26850 in clone3 ()
    at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81
```

[gdb output (gist)](https://gist.github.com/evanmayer/d4e84a91b1d84a98e9ca67926c325c2c)

[valgrind output (gist)](https://gist.github.com/evanmayer/e4fda9369f0f8beacf7a4648faa050e5)

I haven't looked at these closely yet, but since it's inside the Fuji-specific version of the ptp driver, I'll be paying attention to test any fixes or gather more info. I'll also take a look at these logs and the code.

evanmayer, 2024-12-07:
More debugging.

Null pointer dereference on one of the properties:

```Thread 21 "indigo_server" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffff5600640 (LWP 11401)]
0x00007ffff772247a in ptp_set_switch_property (value=4, property=0x7fffef62b050, device=0x7fffe8000b70) at indigo_ptp_fuji.c:546
546	indigo_ptp_fuji.c: No such file or directory.
(gdb) bt
#0  0x00007ffff772247a in ptp_set_switch_property (value=4, 
    property=0x7fffef62b050, device=0x7fffe8000b70) at indigo_ptp_fuji.c:546
#1  ptp_fuji_exposure (device=0x7fffe8000b70) at indigo_ptp_fuji.c:596
#2  0x00007ffff7706ac7 in handle_exposure (device=0x7fffe8000b70)
    at indigo_ccd_ptp.c:343
#3  0x00007ffff7859124 in timer_func (timer=0x7fffd410c430)
    at indigo_timer.c:89
#4  0x00007ffff6c94ac3 in start_thread (arg=<optimized out>)
    at ./nptl/pthread_create.c:442
#5  0x00007ffff6d26850 in clone3 ()
    at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81
(gdb) p *property
$1 = {code = 53772, type = 4, form = 2 '\002', writable = 1 '\001', 
  count = -1, value = {number = {value = 1, min = 1, max = 2, step = 3}, 
    text = {
      value = "\001\000\000\000\000\000\000\000\001\000\000\000\000\000\000\000\002\000\000\000\000\000\000\000\003\000\000\000\000\000\000\000\004", '\000' <repeats 222 times>}, sw = {value = 1, values = {1, 2, 3, 4, 
        0 <repeats 1020 times>}}, sw_str = {
      value = "\001\000\000\000\000\000\000\000\001\000\000\000\000\000\000\000\002\000\000\000\000\000\000\000\003\000\000\000\000\000\000\000\004", '\000' <repeats 222 times>, values = {'\000' <repeats 255 times> <repeats 16 times>}}}, 
  property = 0x0}
(gdb) 
```

I'm assuming the code implies it's related to `ptp_property_fuji_CardSave = 0xd20c` = 53772 (`indigo_ptp_fuji.h`).

evanmayer, 2024-12-07:
On a hunch, turning off "Delete downloaded image" allows the preview exposure to succeed:

[image]

[image]

But this is simply because it does not exercise the code path on `indigo_ptp_fuji.c:596` with the call to `ptp_set_switch_property(device, card_save, FUJI_CARD_SAVE_OFF)`:

```c
	if (result && card_save) {
		if (DSLR_DELETE_IMAGE_ON_ITEM->sw.value) {
			result = result && ptp_set_switch_property(device, card_save, FUJI_CARD_SAVE_OFF);
		} else {
			result = result && ptp_set_switch_property(device, card_save, FUJI_CARD_SAVE_ON_RAW_JPEG);
		}
	}
```

This behavior is the same for camera GUI's USB shooting mode "USB TETHER SHOOTING AUTO"  or "USB TETHER SHOOTING FIXED".

This is an acceptable workaround for me, since having an extra copy of the images is not bad. Let me know if I can help the original issue by providing any info or playing with the hardware.

polakovic, 2024-12-08:
The problem is, that for some "PTP properties" are created "INDIGO properties", but not for all (in this case controlled by the symbol SHOW_ADVANCED_GROUP). The crash happened because driver tried to update non-existent INDIGO property.

Pls. can you test if this 44eeeeefdcf76c73a261e59cfe9eccc1a7fb26a8 fixed the issue?

polakovic, 2024-12-08:
You right. It can't be fixed as easily I supposed, so instead I forced this property to be defined as INDIGO property as well. It doesn't crash now, but at least on my X-T1 doesn't work if "Delete downloaded image" is off.

evanmayer, 2024-12-08:
With 27900148bdcf5ec27fd86e9fbfb128c9666167d2, I confirmed that the shutter fires with "Delete downloaded image" on or off, and the image is saved or not according to the option. It looks like that feature is working as intended on the X-T5.

polakovic, 2024-12-09:
CardSave is just a "phantom" INDIGO property created just to avoid crash (together with Control Priority). Whatever you will set on it, it will be overridden during exposure anyway.

So it looks like it is a new feature working on X-T5, but not on X-T1. Now I have to find which is the oldest model supporting it :(

evanmayer, 2025-02-26:
I'm fine with closing this issue. Feel free to leave open if you need to, though. Thanks!

polakovic, 2025-02-26:
ok

closed 2025-02-26 by polakovic