indigo-astronomy/indigo #585: indigo server crash during image acquisition using --remote-server

Ghir8, 2025-01-05:
Hello, 
as per title, we discover an issue with multiple servers setup.

In our observatory we have a two-server setup:
- Server A (primary) with the mount and GPS connected
- Server B (secondary) with cameras, focuser, and filter wheel connected

Server A connects to Server B using the `--remote-server` option and can see its devices.
**Clients connect to Server A** and, through it, are able to see all connected devices, including those from Server B.

We have noticed that when connecting the devices from Server B to the agents on Server A, the following crashes occur on Server A:
- A camera from Server B connected to the guider agent on Server A, and we make a preview request immediately after downloading the image
- A camera from Server B connected to the imager agent on Server A, and we perform autofocus immediately after downloading the first image

In the latter case, this sometimes also causes AIN to crash.

However, when a camera from Server B is connected to the imager agent on Server A and we perform a preview or a batch of photos, no crashes occur.

The logs do not show any trace or information about the crash. Even when analyzing the logs in trace mode for both servers, Server B does not show any anomalies, while Server A simply restarts without any message.

We have noticed this incorrect behavior since version 2.0-296 and it persists in the current version 2.0-308.

For now, we have resolved the issue by bypassing Server A and connecting AIN to Server B for the imaging and guiding agents.

polakovic, 2025-01-06:
We'll check it, but it would be very helpful if you can make trace logs and also to run the servers with valgrind (start the servers with a command line like "valgrind indigo_server -- ...the rest of your command line...".

It is possible that you simply run out of memory, what is the hardware of these servers?

Nevertheless, just for my curiosity, why do you use this setup (I mean server chaining and remote devices)? INDIGO is designed to use agents as close to the devices as possible, so for the best performance always run imager agent on the server where the main camera is connected and guider agent on the server where the guiding camera is connected. Running imager agent on one server and camera driver on another degrades performance to something just slightly better than legacy INDI. It is also better to keep related agents on the same machine, so if you want to use dithering, keep imager and guider agents on the same server.

rumengb, 2025-01-06:
We managed to reproduce it. We are working on it...

Paolo97Gll, 2025-01-06:
> We'll check it, but it would be very helpful if you can make trace logs and also to run the servers with valgrind (start the servers with a command line like "valgrind indigo_server -- ...the rest of your command line...".

Sure! I will send everything to you in the next days.

> It is possible that you simply run out of memory, what is the hardware of these servers?

I don't think, the main server is a virtual machine with 10 cores and 16GB of RAM, while the secondary server is a Raspberry Pi 5 with 4GB of RAM. Also, by checking during the utilization, we have plenty of RAM free.

> Nevertheless, just for my curiosity, why do you use this setup (I mean server chaining and remote devices)? INDIGO is designed to use agents as close to the devices as possible, so for the best [...]

The remote server functionality is one of the reasons why we chose INDIGO.

The Raspberry Pi is placed on the OTA inside the dome and it connects via USB to all the camera-related equipment (cameras, wheel, focuser). The virtual machine runs on a server in the server room inside the observatory main building and is the main control server: it connects via LAN to the equipment that has a built-in LAN connection (like the mount or the dome) and to the Raspberry Pi INDIGO server.

The main reason for this setup choice is that we need the system to be as stable as possible since it's used in a completely remote context (i.e., in the best case, we are one hour by car from the observatory, but sometimes the system is also used from other countries). Until two days ago, we used a Raspberry Pi 4 with a microSD card, which is not comparable in terms of stability with a virtual machine running in a server room with hardware redundancy and scheduled backups (now we use a Raspberry Pi 5 with NMVe SSD, which is much more stable, but again not comparable). Furthermore, the main server has to handle all the accessory services needed by a completely remote system (like the control dashboard, the power-on and shutdown procedures, the emergency handling in the case, for example, of internet or AC failure, ...). As you may understand, a Raspberry Pi (or in general a mini PC) placed on the OTA or inside the dome and directly connected to the instruments is simply not adequate.

We decided to use the Raspberry Pi as a sort of "USB passthrough" for all the camera-related equipment, while everything else is connected on the observatory LAN directly to the main server. A Raspberry Pi is both powerful enough to guarantee optimum performance and cheap enough to be replaced in case of failure at a low cost. Also, its configuration is very simple, to guarantee a minimum downtime due to a reconfiguration in the event of a replacement. This is important for us since the environmental conditions inside a dome are not so adequate for IT equipment.

All these stability concerns are also the reason why, for example, the mount is connected to the main server and not to the Raspberry Pi (controlling a moving part like the mount is much more critical than controlling a camera).

Just to give you more context about our choice, the dome was built almost 30 years ago as a separate structure from the main observatory building and it wasn't originally intended for remote utilization (so no conditioning system and auxiliary rooms were built in or near the dome). When we decided to convert the dome for remote utilization, we placed the control server in the server room since is the only place with a controlled environment in which run everything we need.

Finally, regarding keeping related agents on the same machine, we had always used only the agents on the main server for controlling everything (imager, guider, mount, astrometry, scripting), we never used the RPi agents. We switched to the RPi agents only now as a workaround, since it's the only way to make all the equipment work with the new indigo versions.

> We managed to reproduce it. We are working on it...

Thank you!

polakovic, 2025-01-07:
I think we  reproduced the problem and is hopefully fixed here: 05587ec78f17a8d87ee7d721caf756db8057ed16

> Just to give you more context about our choice, the dome was built almost 30 years ago as a separate structure from the main observatory building and it wasn't originally intended for remote [...]

I understand, just keep in mind, that keeping agent and its devices on the same machine is a performance advantage. So even if it is strictly possible to use local agent and remote devices, it is fragile setup, among other it depends on stable network connection. With remote agent running with its local devices, you can start process and forget (or even close the client).

rumengb, 2025-01-08:
Hi, we just released Indigo 2.0-310 with the fixes. Please test it and report if it is ok now. It should be available for update from our PPA in 10-15 minutes.

Paolo97Gll, 2025-01-08:
Hi, thank you! I will let you know before the end of the day.

Paolo97Gll, 2025-01-08:
Everything seems to work! We will let you know if we encounter any other related problems.

rumengb, 2025-01-08:
Good :)
Out of curiosity, what equipment do you control with indigo?

Ghir8, 2025-01-10:
Sorry for the delay!!

We are an amateur observatory with three domes, used for photography, science communication, and research in astrometric and spectrophotometric fields. We have currently used Indigo only in the photographic dome, where we have a Celestron C14, primarily for astrophotography and comet photometry. We plan to also use it in the main dome with a 51cm telescope, once we rebuild the entire motion control system.

We essentially have three components:

Photographic Equipment:
* Atik 16200
* Primaluce Focuser
* SX Filter Wheel
* ASI 220mini Guider
All connected to a Raspberry Pi mounted on the telescope, using INDIGO to expose everything to the main server.

Main Control Server:
* GPS
* EQ8 mount via custom EQMod-Ethernet converter
The server exposes all equipment via INDIGO server pass and its agents to the clients.

Dome:
Custom ESP32-based hardware and REST-based control
Electrical panels and safety UPS, also REST-based
It is controlled by the main server through some server-side services.
The dome sync with the telescope was initially implemented with a custom INDIGO driver, but we noticed an issue: the control agent provided the position in real-time during slewing, and since we have a german equatorial mount, the dome kept trying to track the position moment by moment, resulting in erratic dome movements during object pointing.
We then decided to switch to an external Python script, getting the final position of the telescope from the ALPACA agent, and pointing the dome only after slewing was complete.
The main server also exposes a custom dashboard to the user, allowing them to turn systems on, monitor weather and sky conditions through an all-sky camera positioned at the observatory, as well as a couple of custom functions we’ve implemented over time (such as automatic anti-dew bands, safety shutdowns, custom pointing limits that shut down the mount in case of dangerous positions, etc.).

rumengb, 2025-01-10:
Thank you for getting back to me. I see the problem. I will see what I can
do to make the dome track the mount while mount is tracking and move
directly to the target position when goto is issued. Do you think this will
work?

Rumen

Ghir8, 2025-01-10:
Yes i think it would be great! 
we never open an issue for that problem because we solved with the custom script that also can slew the sun during winter days to defrost the shutter 😅

But integrating the basic dome controls into Indigo will be fantastic!

rumengb, 2025-01-10:
We have dome Salving mode where the dome follows the mount did you use it?
[image: image.png]