indigo-astronomy/indigo #588: Indigo Control Panel, Imager... Not populating in windows

lucaNobile, 2025-01-18:
good evening to all. It is nowmail several months that I am 
struggling to get some astrophotography systems working, as either one thing is not working for me or another is not working for me.I have now come to the INDIGO platform, but it too is not working for me.
The first thing I have to mention is that, given the money I have spent, I would like to stay on windows (having primaluce), or at most since I own a Linux laptop (Endeveuors). I have tried installing INDIGO Control Panel, Viewer, Image.
But to my dismay at startup the window of what it could load does not populate.I tried entering in the box avaiable service:
indigo-server@INDIGO-server:7624
indigosky@indigosky:7624
but restarting windows still found the list of services empty.
I must say that on Windows I also have Kstars and idriver ASCOM.
I am quite distressed iin that now nothing has worked for me for about 3 months and being in the high mountains I would like to take advantage of this situation.
Thanking everyone I greet

Translated with DeepL.com (free version)

rumengb, 2025-01-18:
Hi, please make sure indigonserver and indigo client are in the same local
network. You can also try to connect to the server via the ip address of
the server by adding it in File->manage services menu.

lucaNobile, 2025-01-18:
Thanks for the quick reply.There are some things I don't understand, 
I don't know anything about networking, so I hope you can help me initially but I don't see the windows Indigo server and/or Indigo client screen.As you told me I entered an ip address which should be from my local network(localhost?)

closed 2026-08-31 by rumengb